ImpulseAdventure/GUIslice #253: No transparency for C graphics using ILI9341_t3

MVogtDD, 2020-09-01:
It acutally might be a problem with the ILI9341_t3 (Teensy) library and not of the GUIslice builder:

When importing a C graphics file as a graphics button, the transparancy colour (i.e. magenta) is correctly displayed transparent in the builder. However on the ILI9341 display it is still displayed as magenta. I haven't looked into the ILI9341_t3 library in detail yet to figure out if transparancy is supported there but I thought you might know.

Best regards,
Marcus

ImpulseAdventure, 2020-09-04:
Hi @MVogtDD --

I have created a test branch for you to try out: https://github.com/ImpulseAdventure/GUIslice/tree/WIP253-BMP_Trans_RAM

Please try downloading this version of the library and let me know if it operates any differently with the transparency rendering. Note that I have not had an opportunity to test it out on my hardware yet, but thought I would push the code to the test branch in the meantime.

thanks

ImpulseAdventure, 2020-09-14:
Hi @MVogtDD — I will close this for now, but if you have an opportunity to try out the patch later, let me know and I will reopen. Thanks

ASMotionLab, 2020-10-11:
I was searching for transparency from flash, and came across this issue.
I have slightly different hardware, but I think it's close enough - apologies for adding noise if it's not! (I'm on ESP32, TFT_eSPI, ILI9341)
Decided to test out the new branch WIP253-BMP_Trans_RAM.
Results are as op's original comment:
- when icon is added to GUIslice builder, by default it has the magenta colour where transparency is.
- when transparent is toggled to true, GUIbuilder renders it perfectly.
- after flashing, the result is with magenta pixels:
[image]
I'm happy to test on behalf of op if my hardware is suitable.

ImpulseAdventure, 2020-10-11:
Hi @ASMotionLab — the test branch referred to above only modified the GUIslice_drv_adagfx.cpp as the OP was using a Teensy. I presume you are using TFT_eSPI, which means that a similar modification should be carried out in GUIslice_drv_tft_espi.cpp.

Assuming you will be able to try out my test branches, I will reopen this issue and proceed to make the enhancement for TFT_eSPI as well. I will also check to see if there are any additional APIs available in TFT_eSPI that we can utilize.

ASMotionLab, 2020-10-11:
@ImpulseAdventure - I missed that op was using adagfx, apologies. My noobness is showing. Appreciate you looking into it though!

Pconti31, 2020-10-11:
@ASMotion You should be aware that this turns off the high speed painting.   You get high speed or transparent pixels.    

At least that's the implementation Calvin put into this branch so far for Adafruit GFX.  If it turns out to be the same for TFT_eSPI expect all your C array image graphics could be 15 times slower not just your transparent C array images.   

A new API that lets people select on a image by image basis to have transparency or not might be a better way to go but of course that's a much bigger job.

IMHO a faster implementation of transparent images would not be possible given the TFT Displays we Hobbyist use.

Paul--

Pconti31, 2020-10-11:
@ImpulseAdventure As I thought about TFT_eSPI's API maybe the sprite interface might provide a faster solution than the current Adafruit's GFX version you have in this branch.  

I do think though a new API allowing user to select on an image by image basis to enable transparency or not would be helpful
so as not to slow down all C array image painting because of the lack of BLIT.
Paul--

ImpulseAdventure, 2020-10-11:
Thanks Paul & ASMotionLab —

Thankfully TFT_eSPI provides an optimized pushImage with transparency that seems like it could still be relatively quick. The existing pixel-wise emulation here would incur moderate overhead / slowdown.

I agree that this might be a good opportunity to provide an alternate API so that we can choose when to take advantage of these optimized BLIT modes vs transparency rather than a global config option.

Let me look into what options might be available and propose something. In the meantime, I plan to update this branch to leverage the optimized TFT_eSPI pushImage (with transparency if enabled in config).

Pconti31, 2020-10-12:
@ImpulseAdventure 
@ASMotionLab   Well, I hope pixel-wise emulation here would incur moderate overhead / slowdown.  

However I suspect its quite a bit slower.   Be interesting to do some timings.  Bodmer's code is optimized compared to doing one pixel write at a time but not much better.  Basically his code devolves to one pixel write for each transparent pixel.  

Non-transparent pixels on one row line are stored in a line buffer until either the end of the scan line or another transparent pixel is encountered.  Then the line buffer is batch written to the display(with the new transparent pixel)  and the buffer flushed. 

If new transparent pixel now found then it and the now empty line buffer is written to the display.

So if you have two or more transparent pixels in a row like most transparent images its still just one pixel at a time write for much of the display.

Seems like if you used his code as an example it would be a simple matter to not write the transparent pixel and just place it inside the line buffer and do writes only when the buffer is full.   This then would be as fast as what the non-transparent images are displayed.   Unlike Adafruit's BLIIT routine his is always doing one row at a time BLITs anyway but since the hardware is so much faster it isn't noticeable.

Paul--

ImpulseAdventure, 2020-10-12:
From Bodmer's code, I had figured his implementation of `pushImage()` with transparency should be quite a bit faster than the pixel-wise "Adafruit" default we have now. My thinking was based on the assumption that a) the processor's pixel-wise RAM transparency comparison delays are dominated by the SPI transaction delays (`addrWindow()` and the SPI transaction itself) and b) like the OP's gear icon, a given raster line within a transparent "bitmap" generally has 3 "runs" (transparent run, solid run followed by a transparent run), so we would only see a single buffer push (and it's associated addrWindow call) per line. Thus, my feeling was that the time spent traversing transparent pixels (on a modest MCU) would be far smaller than cumulative PA/CA set overhead we'd see with `drawPixel()`.

As you point out, most useful of all would be to see how the actual timing compares, as one can only infer so much about performance from looking at code :)

Paul -- in your last paragraph was your suggestion that we consider implementing our own line buffer and stream these out? (ie. leverage some of Bodmer's optimization and apply it across all of the GUIslice "drivers"?)

Thanks for the great feedback

Pconti31, 2020-10-12:
@ImpulseAdventure 
 Yes I'm suggesting you could do better for Adafruit's GFX and some of the other drivers where we do pixel by pixel transfers by taking a closer look at Bodmer's routine.  I'm not suggesting we replace or try and improve TFT_eSPI.

farfarfromhome, 2020-11-02:
I am seeing the same behavior with TFT_eSPI and IL9488 driver.  Transparency works in the Builder using a 24bit BMP file.  However it doesn't work on the actual hardware from PROGMEM.  If you guys come up with a potential solution I am more than happy to join the test team!

I did notice that the only combination that correctly draws the image is 24bpp parameter GSLC_IMGREF_FMT_BMP24 with 16bpp c-file per example 28.   GSLC_IMGREF_FMT_BMP16 with 16bpp c-file and GSLC_IMGREF_FMT_BMP24 with my guess for a 24bpp c-file do not work.

ImpulseAdventure, 2020-11-02:
Hi --

The previous updates that were in the WIP253 branch were only intended for the Adafruit/Teensy display modes.

@ASMotionLab & @farfarfromhome -- I have now added support for TFT_eSPI. We should see very fast transparency with memory-based images (eg. via `gslc_DrvDrawBmp24FromMem`). I have run the example from @MVogtDD ,, and only needed to change `ElemSetFillEn` from `true` to `false` in `Transparent_GSLC.h`: This update has been attached as "Transparent.zip".

[image]

[Transparent.zip](https://github.com/ImpulseAdventure/GUIslice/files/5472746/Transparent.zip)

I have tested this both in the Adafruit/Arduino display drivers as well as TFT_eSPI. Both are showing transparency (with magenta designated as the transparency color, ie. 0xFF,0x00,0xFF).

The added benefit of TFT_eSPI mode is the accelerated transparency support.

If someone can confirm that it looks to be working for them, I will merge these changes.

Thanks,
Cal

farfarfromhome, 2020-11-02:
Cal

I downloaded the WIP253 branch and it works per your instructions.  Transparency is working with ElemSetFillEn set to false,  The transparent color is converted to white when set to true.  I attach a picture to show.  The upside down icons are a problem with my convert process for the material design icons files!

[image]

ImpulseAdventure, 2020-11-02:
Thanks @farfarfromhome for the confirmation!

Yes, if FillEn is true, then the element's "Fill" color will be used prior to rendering the image. Generally, when the image uses transparency, we will want to set FillEn=false.

The changes have now been merged into the main repo.

closed 2020-11-02 by ImpulseAdventure

farfarfromhome, 2020-11-02:
@ImpulseAdventure  Thanks for the amazingly fast overnight new feature!!

I am not sure if I am running the latest version of the Builder but selecting transparency in the Builder sets FillEn=true - or maybe I am doing something wrong?

ImpulseAdventure, 2020-11-03:
Regarding the fill enable... this was indeed a bug fix that was recently corrected in the latest version of the Builder.

From Paul’s release notes for Builder v0.16:
_Bug No. 98 Text element "Fill Enabled=false" doesn't render with transparency_