ImpulseAdventure/GUIslice #254: Support for ILI9225

ImpulseAdventure, 2020-09-04:
From @pierbout:
> Wonderful tool! I love it.
> Is there any chance I could adapt it to a 2.2' 176x220 TFT screen (ILI9225 library)?
> 
> Thanks in advance for your help, and sorry if I didn't post in the best discussion...

_Originally posted by @pierbout in https://github.com/ImpulseAdventure/GUIslice/issues/79#issuecomment-686589400_

Hi @pierbout -- thanks!

Yes, I believe GUIslice should already support your display.
Please try downloading the "mcufriend_kbv" library and confirm that the provided example works correctly first.
If that works OK, then we can create a new GUIslice config for you.

pierbout, 2020-09-04:
Thank you for your answer.
I'm going to try this or see if I can adapt the ILI9341 GUIslice config to ILI9225 (with a bit of luck there might not be too many variables to modify...)

ImpulseAdventure, 2020-09-06:
Pierre -- for reference, I have uploaded a new config to the GUIslice repository for your ILI9225 display. Once you have confirmed that the **mcufriend_kbv** example works fine on your display, please update GUIslice and uncomment the following line in `/src/GUIslice_config.h`:

`ard-adagfx-ili9225-notouch.h`

pierbout, 2020-09-07:
Thank you Calvin.

mcufriend_kbv does not seem to be fully compatible with ILI9225 yet. ILI9225 is listed in some of its files but not others. Maybe in a few months...

As an example, "diagnose_TFT_support" doesn't recognise the screen ID:
     _tft.readID() finds: ID = 0x0
     MCUFRIEND_kbv version: 2.9.9
    This ID is not supported_

I'm working on adapting GUIslice directly to ILI9225 directly, without using mcufriend_kbv. It compiles but some adaptations are still needed. I'll upload a suggestion in 2-3 days.

pierbout, 2020-09-07:
Hello Calvin. I updated a few files in the ILI9225 branch. Can you see it?

They compile for ILI9225 but are not usable, as ILI9225 does not have enough accessors to update all your functionnalities. So I added ILI9225-due configuration, which is far more powerfull (many many more accessors, and very similar to ILI9341-due accessors.

pierbout, 2020-09-07:
I won't modify anymore the ILI9225 adaptation: it lacks too many accessors to adapt to GUIslice. I think the best solution is to wait for MCUFRIEND to manage ILI9225.

I still have some issues with ILI9225_due, but I don't know yet what is due to the configuration ans what is due to the builder:
-Text: OK
-Text color: OK
-Text size OK
-Progress bar: OK
- Text button: OK
-Text alignment: always align on the left -->I'll try to correct it
-Toggle button: Compiles, but only the middle button displays -->I'll try to correct it
-Change police other than default: KO (compiles but does not display) --> How can I load in the builder's GUI the list of available polices (and only them?)
-Text imput and number input: not enough memory on an Arduino Nano (requieres GSLC_FEATURE_COMPOUND = 1). I won't work on it in a near future. One day maybe (on STM32)

ImpulseAdventure, 2020-09-07:
Hi Pierre —

Thanks for taking the time to submit some proposed updates.

I’d like to see if mcufriend_kbv might work if we force the ID. With the mcufriend_kbv example graphicstest_kbv.ino can you try replacing the tft.begin(ID) with the following and let me know if that works any differently?

tft.begin(0x9225);

pierbout, 2020-09-08:
Hello Calvin.

I tried to force the "begin", as suggested: it compiles, but the display remains black.

Nevertheless, it might be due to the fact that I didn't find the appropriate variable to define the clock input pin (CLK = SCK):
#define LCD_CS 10 // Chip Select
#define LCD_CD  9 // Command/Data
#define LCD_WR 11 // LCD Write
#define LCD_RD    // LCD Read (not requiered for ILI9225)
#define LCD_RESET 12 // Can alternately just connect to Arduino's reset pin
//#define TFT_CLK 13  // SCK
//#define TFT_LED 0   // 0 if wired to +5V or +3.3v directly

I also tried to force the begin in diagnose_TFT_support:
    //ID = 0x9225;
    ID = (uint16_t)0x9225;
    Serial.println(F(""));
    Serial.print(F("Forced ID: ID = 0x"));
    Serial.println(ID, HEX);
    Serial.println(F(""));

I get:
tft.readID() finds: ID = 0x0
MCUFRIEND_kbv version: 2.9.9
Forced ID: ID = 0x9225
This ID is not supported

In my opinion, it confirms thar ILI9225 is not supported yet "as is", despite it is listed among the supported libraries. It can certainly be adapted/corrected, but I'm affraid I won't have time to investigate a lot more, unless if you have suggestions

ImpulseAdventure, 2020-09-08:
If you haven’t already, I would recommend posting a message on the Arduino Display subforum: https://forum.arduino.cc/index.php?board=7.0

David Prentice is extremely helpful in identifying and adapting displays and may be able to give you a more concrete idea whether your display can be supported. I would prefer that we confirm the status of mcufriend_kbv support before we integrate additional drivers. If he indicates it can’t be supported I will take a look at the alternate ILI9225 libraries you referenced (thanks for the pointers!)

Thanks

pierbout, 2020-09-08:
Thank you Calvin. I'm going to do.

pierbout, 2020-09-09:
Hello Calvin,

David is indeed very quick and helpfull! Here is his answer:
https://forum.arduino.cc/index.php?topic=704322.0

ImpulseAdventure, 2020-09-12:
Thank you Pierre! Yes, I should have taken more note of your comment "*I didn't find the appropriate variable to define the clock input pin*" to indicate that you had a SPI display, for which the mcufriend_kbv doesn't support (as you correctly noted).

In that case it looks like the best path forward is for me to look at adding the **ILI9225_due** support. I presume you are using a Due to drive your display.

Note that I might not be able to start with your ILI9225_due PR #259 as it appears to be based on a much older version of GUIslice. As a result, the changes for adding `DRV_DISP_ADAGFX_ILI9225_DUE` are somewhat difficult to review due to the large number of differences. If you are able to clone the latest version of GUIslice and re-apply the smallest changes necessary to just add `DRV_DISP_ADAGFX_ILI9225_DUE` into a new PR, then I may be able to integrate them for you directly.

thanks!

pierbout, 2020-09-12:
Hello Calvin,

The GUIslice version I used is the one provided by default with Linux Mint (which is often not the last one).

The wiser solution, if I find time, is that I try ti do the modifications, as I have the device to test it and that I know what I changes and why (with an updated version of GUI slice).

We keep in touch.

pierbout, 2020-09-13:
Hello Calvin

Shall I use the master of GUIslice, or V0.15.0? I don't know if they are
identical or not...

ImpulseAdventure, 2020-09-14:
Hi @pierbout -- the changes since v0.15.0 (from the Library Manager) are relatively minor. Nonetheless, I would always recommend taking a fresh clone of the GUIslice repository (master) before making your changes and submitting a PR.

Thanks!

pierbout, 2020-09-14:
Ok, Calvin. I'm going to do that.

Thanks.

pierbout, 2020-09-14:
Hello Clavin,

Sorry to bother you, but I'm not used to git (I usually use SVN). How can I create a pull request with 4 files, and not 4 distinct PR?

See attached files

[GUIslice-#254-pierbout-Add_ILI9225_due.zip](https://github.com/ImpulseAdventure/GUIslice/files/5219342/GUIslice-.254-pierbout-Add_ILI9225_due.zip)

ImpulseAdventure, 2020-09-18:
Hi @pierbout --

Thanks again for putting together an enhancement for **ILI9225_DUE**.

I have started to create a new branch [WIP254-ILI9225](https://github.com/ImpulseAdventure/GUIslice/tree/WIP254-ILI9225) which should incorporate the major changes you had done within your various repository patches. It was a bit complicated as you had created 11 patch branches in parallel :) Git can definitely be tricky at first! Nonetheless, I think I have integrated the majority of the changes you needed.

Please try the updated branch and let me know if it works or if additional changes are required:
- Download the branch here: [WIP254-ILI9225](https://github.com/ImpulseAdventure/GUIslice/tree/WIP254-ILI9225)
- Modify `/src/GUIslice_config.h` to uncomment `due-adagfx-ili9225-notouch.h` (note that I renamed your file)
- As the ILI9225 is a smaller display, try an example like **ex40_ard_ctrls_small**
- If your testing shows that it works OK, I will proceed to create a PR for these changes

Referring to your comments in https://github.com/ImpulseAdventure/GUIslice/issues/254#issuecomment-688447647:
- Can you clarify what you meant by *Change police other than default: KO (compiles but does not display)*? I assume "police" might have been a mis-translation
- Regarding the KeyPad: we have work in progress in #255 that should greatly reduce the memory consumption of the XKeyPad. That upgraded component should be available soon.

Thanks again for your contributions!

pierbout, 2020-09-18:
Hello Calvin,

Sorry again for the mess with the patches. I did my best but as I said I'm not very familiar with Git.

I'm going to test the branch. Thanks.

-Change police other than default: KO (compiles but does not display): police = font = typeface. In the GUI, if I select one of the listed polices instead of the one by default, the compilation works but the text is not displayed on the screen (if I leave the default police, I see my text on the TFT screen). I did not have time to investigate more. It's probably a configuration problem (the listed fonts are not the ones from ILI9225* libraries).

pierbout, 2020-09-18:
I tested the WIP254-ILI9225 branch with the following programs:
- ex40_ard_ctrls_small: Not enough space : 35848 octets requiered (116%) 
   --> I cutted InitGUI to display everything I can (by commenting 3/4 of the code each time) and it seemed to work. At least every method compiled.
-ex01_ard_basic : OK (after modification of the rectangle size to fit the screen)
- ex02_ard_btn_txt: OK (after modification of the rectangle size to fit the screen)
- ex03_ard_btn_img: GSLC_SD_EN required so I could not test it yet
- ex04_ard_ctrls: Not enough space :37502 octets requiered (122%)
- ex05_ard_pages: Not enough space  2079 octets requiered (101%)

Anything you would like me to test?

I also noticed an issue withe the default font on the following letters: i, l, L, 1: they are displayed with 2 vertical bars. For example, "1" is displayed "1l". I'm going to test other fonts.

ImpulseAdventure, 2020-09-18:
Hi Pierre — great time hear that the branch appears to be working for your display!

No problem at all about the patches! GitHub has a nice “Network Graph” that made it easier to figure out what all of the patches were doing.

As you are using a device with very limited memory capacity, you may need to try the FLASH-based examples instead. Have a look in the “arduino_min” folder. Not all of them may work, but it could help on certain sketches. In the Builder you would want to select “use FLASH API” wherever possible.

I haven’t looked at the “custom” font support yet. We may need to investigate that further.

As for your observation about the “I /l” font rendering: this is actually a bug in ILI9225_due that occurs when the font has a trailing blank column. I may see if I can submit a fix for Martin to integrate.

ImpulseAdventure, 2020-09-19:
Pierre -- regarding the font glitch, I think I have a potential fix ready to try out, though I haven't tested it on any ILI9225 hardware yet. Please try the following:
- Download the attached sketch ([ILI9225_due-font-glitch.zip](https://github.com/ImpulseAdventure/GUIslice/files/5249122/ILI9225_due-font-glitch.zip)) and modify the pinout to match your device
- Run the sketch and let me know if you see any font glitches like what you observed
- If you do see the glitch, then try unzipping and copying the following patch into your /Arduino/libraries/ILI9225_Due directory:
[ILI9225_due-patch1.zip](https://github.com/ImpulseAdventure/GUIslice/files/5249126/ILI9225_due-patch1.zip)
- Rerun the ILI9225_due-font-glitch sketch; does it look any different?

Please note that I have not tried running any of the above on an ILI9225, so it is quite possible that they won't work as-is. If it does work, then I will submit this to Martin as a pull request.

thx

pierbout, 2020-09-21:
Hello Calvin,

I don't see any difference before and after patch. On the other hand, the problem only occurs in transparent mode. Interesting...

Before patch:
Trouble on 1, i, l, I in transparent mode. Solid mode seems OK.
[image]

After patch:
[image]

ImpulseAdventure, 2020-09-21:
Thank you Pierre for testing it out and sending a photo.

This is good news, because it confirms:
- The ILI9225_Due has the exact same issue as I had seen in ILI9341_Due, which I had already fixed.
- The first attempt at a patch still rendered text ok, so the basic code is working.

I will need to re-examines the code again to see what additional edits on the patch may be required, or may ask that you run a debug version for me.

Thanks

ImpulseAdventure, 2020-09-22:
Hi @pierbout --

The updates for GUIslice (that you saw in the development branch) have now been integrated into the latest GUIslice library.

That said, there are still two open issues that we can followup on:
- ILI9225_Due: attempt to develop a suitable fix for the library. I have created a separate issue in Martin's ILI9225_Due repository to track this. Perhaps I might be able to refine the fix and try again later for you to test
- GUIslice support for custom fonts on ILI9225_Due. Can you try the attached sketch (ex02_ard_btn_txt-font) and let me know if it renders the Arial font within the button?
  - [ex02_ard_btn_txt-font.zip](https://github.com/ImpulseAdventure/GUIslice/files/5258970/ex02_ard_btn_txt-font.zip)

thanks

ImpulseAdventure, 2020-10-20:
As the main driver support for ILI9225 was added in #270 , I will close this issue for the moment.

If you find an opportunity to test out the custom font example in https://github.com/ImpulseAdventure/GUIslice/issues/254#issuecomment-696482242 and discover that it doesn’t work, let me know and we can track that in a new issue.

As for the ILI9225_due font glitch, I had posted an issue / pull request for Martin to review a month back but it looks like he might not be currently maintaining the library.

closed 2020-10-20 by ImpulseAdventure