ImpulseAdventure/GUIslice #255: Support for Alphanumeric keypad

helax4411, 2020-09-04:
### Feature Request or Enhancement
Many times you need to input a combination of text and numbers, for example if you want to introduce the SSID if a WIFI network, most of the times those SSID names are the combination of letters and numbers. Guislice has only keypad for numbers or keypad text for letters, you cannot write for example a SSID name like RAP34$NOW , is not possible, or for example what about when you want to introduce and IPV4 address?: the floating point allow you just to use ones the point and the number of characters are limited to 11, same for letters, and alphanumeric keypad will allow you to introduce combination of letters and numbers and/or complex numbers like IPV4 or V6. Is not a nice to have I think is  a must to have.

Helax4411

### Additional context
*Add any other context or screenshots about the feature request here.*

Pconti31, 2020-09-04:
@helax4411 I submitted a new keyboard implementation a long time ago but I guess it had some issue so it never was accepted.  Nevertheless I use it and it works fine for me. By the way I wrote the original version that Calvin modified for production.
Anyway, if you want to experiment with it you can pull it out from my clone of GUIslice in mycontrols branch.
[https://github.com/Pconti31/GUIslice/tree/my_elements](url) 

Now my branch hasn't kept up with all of the bug fixes calvin has made to the API so it would be best
to just take the files inside src/elem/XKey*.*

Upgrade to Keypads. First you no longer need to use compound elements saving about 1500kb of Ram. The Number keypad now allows you set on a per field basis to be either float or integer number
(Currently you must choose on the whole keypad during configuration now you can do per field).
The Alpha keypad is now smaller and has buttons to switch to either a number or special symbols keypad. No code changes are required to use these keypads.

Now if you like it say so and Calvin and I maybe can work out a version to include in the API.

Paul--

helax4411, 2020-09-04:
Hello Paul
That will be fantastic, if you and Calvin can have a look on that will be really appreciated

Pconti31, 2020-09-04:
Sorry, but I just tried my version with the latest version 0.15.0 api and it doesn't work.
It only works with an older GUIslice API.
I'll update it and post a zip file for you here once I get it working again.
You will need to try it and say if its better or not.

Paul--

helax4411, 2020-09-04:
Hello Paul, works fantastic, my platform here is ESP32 DEvKit with tft 4 inch SPI touch screen, better impossible.
Thank you

helax4411, 2020-09-04:
Hello Paul
I forgot to mention; all letters are in Capital, for passwords must exist the way to change from normal letters to Capital letters (CAPS), is almost perfect!!

helax4411, 2020-09-06:
Hello Paul
I found another issue, it accepts only 11 characters, no more, to introduce a IPV4 need at least 15 characters and for a DNS name more, let say 25 characters.
Alex

Pconti31, 2020-09-06:
Well as for limit on text size a little imagination gets you around that doesn't it.
IPV4 breaks nicely into four fields each three long with you adding the "." between them.
DNS can also be broken up into more than one field.

This doesn't mean I'm not looking at a better keyboard but it could be a while before I have a solution. 
So plan accordingly.
Paul--

helax4411, 2020-09-06:
Hello Paul
You are right, I was thinking on that way for IPV4 and DNS but regarding the CAPS letters I have no solution.
Alex

ImpulseAdventure, 2020-09-06:
Hi @helax4411 --

As mentioned, I am in the process of incorporating some of Paul's enhancement to the XKeyPad (which will reduce memory consumption considerably). At the same time, I am exploring a further enhancement that may enable more flexibility, including extra keyboard operations (such as case swap). I can't promise that this extra enhancement will be available right away, but nonetheless will try to see if I can create a suitable implementation. thx.

helax4411, 2020-09-08:
Hello Calvin
Thank you for your support, just notify me here when you have some code for testing.
Alex

Pconti31, 2020-09-14:
@helax4411 Calvin and I have completed an experimental version of XKeyPad that should address your requirements.   Calvin cleaned up the code quite a bit, making it easier to maintain and added a lowercase set of characters.  These changes allowed me to add some new features

So what is different in this version.  Lowercase has been added as I stated above.  
What about larger field support?  I now use the field you supply as input to determine max field size for the keypad. 
Example you define this field in code:
```
  // Create E_TXT_VAL1 modifiable text label
  static char m_strtxt5[16] = "";
  pElemRef = gslc_ElemCreateTxt(&m_gui, E_TXT_VAL1, E_PG_MAIN, (gslc_tsRect) { 10, 100, 150, 30 },
    (char*)m_strtxt5, sizeof(m_strtxt5), E_FONT_TXT1);
```
The keyboard or keypad will set the max size to 16. 15 characters with a trailing nul '\0'
Internally there is a max size we will allow for any input field.  It's set in XKeyPad.h as:
```
#define XKEYPAD_MAX_STR      26   // Maximum Input Text Field size
```
This can of course be changed by you if you need larger fields.

Now as I used this I realized that typing in a long field and making a mistake at an earlier character position was a real pain to have to backspace and then retype the whole string over again so I added what I call an embedded cursor. 

It's set to '^' and will show up when you type in your first character.  The nice thing is that I also added two scroll buttons "<" left scroll and ">" right scroll.  This allows you to place the cursor anywhere within your input string and either insert additional characters or backspace from that point to delete characters.  Also, if you input field is larger than the display width of the keypad this will also automatically scroll.  

Oh, backspace on the old keypad was '<' and because of the conflict with scroll buttons has been renamed to "BS".

We would appreciate any feedback you or anyone else cares to post.
First though remember that this is an experimental version and that the one Calvin includes in GUIslice API may be quite different.  We will however continue working at maintaining backward compatibility.  

Paul--
[XKeyPad-cursor-v3.zip](https://github.com/ImpulseAdventure/GUIslice/files/5220133/XKeyPad-cursor-v3.zip)

helax4411, 2020-09-16:
Hello Paul

Thank you to you and Calvin for the fast development, I will test within the next days and report my findings here. 

Alex

helax4411, 2020-09-16:
A quick test show me a compile error:

C:\Users\HB9DRI\AppData\Local\Temp\arduino_build_715568\sketch\Alpha_01.ino.cpp.o:(.literal._Z16InitGUIslice_genv+0xd80): undefined reference to `gslc_ElemXKeyPadCfgSetRoundEn'
C:\Users\HB9DRI\AppData\Local\Temp\arduino_build_715568\sketch\Alpha_01.ino.cpp.o: In function `InitGUIslice_gen()':
C:\Users\HB9DRI\Documents\Arduino\AnTrackPro\Alpha_01/Alpha_01.ino:1024: undefined reference to `gslc_ElemXKeyPadCfgSetRoundEn'
C:\Users\HB9DRI\Documents\Arduino\AnTrackPro\Alpha_01/Alpha_01.ino:1024: undefined reference to `gslc_ElemXKeyPadCfgSetRoundEn'
collect2.exe: error: ld returned 1 exit status
Using library GUIslice at version 0.15.0 in folder: C:\Users\HB9DRI\Documents\Arduino\libraries\GUIslice 
Using library TFT_eSPI at version 2.2.14 in folder: C:\Users\HB9DRI\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 1.0 in folder: C:\Users\HB9DRI\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
Using library FS at version 1.0 in folder: C:\Users\HB9DRI\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
Using library SPIFFS at version 1.0 in folder: C:\Users\HB9DRI\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS 
exit status 1
Error compiling for board ESP32 Dev Module.

ImpulseAdventure, 2020-09-16:
Thanks... yes, we just yesterday noticed a typo in this particular patch that had accidentally omitted the ElemXKeyPadCfgSetRoundEn function. For now, please comment out that line in your sketch (assume line 1024). Thx

helax4411, 2020-09-16:
Just follow your recommendation and after comment that line compile and works, the cursor option to do correction without delete and re-type works great, I will do more extensive test and back later.
Alex

ImpulseAdventure, 2020-10-17:
For reference, we have completed a number of enhancements to the XKeyPad and I will be pushing these changes to a new branch shortly.

Quick summary:
- Greatly reduced memory consumption (no need for compound element)
- Support for multiple pages within the alphanumeric keypad (e.g uppercase, lowercase, symbols)
- User-customization of the keypads (key labels, layout, colors, etc.) via *-setup.h files
- Cursor navigation within the keypad, to allow insert/delete anywhere
- Dynamic button styling (color, visibility, content, etc.)
- Text input scrolling
- Staggered key layouts

Example image: (photo of TFT)
[image]

ImpulseAdventure, 2020-10-30:
Closing as the new version of the XKeyPad has been merged into the repository (v0.15.1 which will be released as v0.16). Please note that some API changes in this version could require minor edits from those previously using the older version of the KeyPad.

A new version of the Builder (v0.16) will be released shortly in combination with **GUIslice release v0.16**.
- Information on the new keypad will be captured later at the following page: [Custom KeyPads](https://github.com/ImpulseAdventure/GUIslice/wiki/Custom-KeyPads)

In the meantime, one can try running the examples in v0.15.1, eg. `ex27_ard_alpha` or `ex26_ard_calc`.

closed 2020-10-30 by ImpulseAdventure