ImpulseAdventure/GUIslice #265: Problems getting GUIslice working with STM32F411 + TFT_eSPI

ASMotionLab, 2020-09-15:
Hi.

Thank you for your efforts in creating this fantastic tool!

I have an STM32F411CE (BlackPill) working with TFT_eSPI and ILI9341 display.

I'd like to retrofit GUIslice into the project, but the stm32-adagfxas-ili9341_xxx.h and stm32-adagfx-mcufriend-xxx.h setups are requiring dependencies which are incompatible with the STM32F4 series (looking at the MCUFRIEND and Roger Clarke's dependencies in the setups, GUIslice is assuming STM32F1 chip, as there isn't an adagfx version for the F4.)

TFT_eSPI standalone works however, so how can I point my GUIslice to the already existing and working TFT_eSPI setup for the F4?
I believe TFT_eSPI library rolls it's own adagfx library which has very wide platform compatibility and hence why it works with my F4.

Thanks!

ASMotionLab, 2020-09-16:
I think I answered my own question:

I was able to use the esp-tftespi-default-xxx.h config with my STM32F411 straight out of the box, because my TFT_eSPI is already set up for my F4 so the bulk of the config is taken from that directly.

closed 2020-09-16 by ASMotionLab

ImpulseAdventure, 2020-09-16:
Excellent... glad to hear it works for you on your STM32F4 using the TFT_eSPI driver base — I will note this as an alternate option going forward.

ASMotionLab, 2020-09-16:
Yes it works great!
Assuming TFT_eSPI keeps up to date with ST's official Arduino core, going through TFT_eSPI could be a great way to ensure all those devices are supported by GUIslice, instead of the current Roger Clarke specific adagfx drivers. (Nothing against the guy - he's done incredible work porting so many libraries to the ST chips, but most of them are now supported by the official core, and lots more hardware).