ImpulseAdventure/GUIslice #266: Toggle button rounded corners not rendering correctly

ASMotionLab, 2020-09-16:
Hi,

The toggle's corners look perfect in the builder, but don't render correctly on the display.
Rectangular radio button looks fine.

I'm running STM32F411CE (BlackPill) with TFT_eSPI and ILI9341 display. Using the TFT_eSPI drivers.

[image]

[image]

Pconti31, 2020-09-16:
@ASMotionLab Please zip your entire project folder and attach to this issue.

Paul--

Pconti31, 2020-09-16:
@ASMotionLab Sorry, no need to supply any additional information. I was able to reproduce the error with TFT_eSPI.  Doesn't happen with Adafruit_GFX library.

The work-around for  the moment is to go to the property view of the toggel button and set Circular?=false for box version.
 Paul--

ASMotionLab, 2020-09-17:
@Pconti31 thanks for finding the error.
I'm using the square radios for now.
From a ux point of view they feel more like sliders than radio buttons that's the only issue, but I appreciate this is a small issue for a small portion of users, so you have better things to do 👍

Pconti31, 2020-09-17:
@ASMotionLab Actually here is the fix. A one line change to XTogglebtn.c to clear the area.  Not sure why Adafruit GFX clears the area but never mind that.
Not a slider since you can't slide the button like our scrollbars.  Touch only works.
Attached is a zip of the modified XTogglebtn.c I'll issue a pull request later today for Calvin to add to GUIslice API.
Paul--
[XTogglebtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5238953/XTogglebtn.zip)

ASMotionLab, 2020-09-17:
@Pconti31 fantastic! Thank you so much 👍
Sorry I mixed up my words, I meant toggle button.
I meant if someone was looking at the rectangular toggle button - they might think it's a slider and try to slide it (I'm thinking my mum/non-tech users).
Toggle buttons' distinct round shape is what really tells people it's a binary toggle, from a glance.

Pconti31, 2020-09-17:
@ASMotionLab I understand. I do need you to confirm if the fix works for you or not.  No rush. 
Paul--

ASMotionLab, 2020-09-17:
@Pconti31 Unfortunately it's still not working 😢 

[image]

I've attached the project and the library too so you can see my config.
[TestProj2.zip](https://github.com/ImpulseAdventure/GUIslice/files/5239428/TestProj2.zip)
[GUIslice.zip](https://github.com/ImpulseAdventure/GUIslice/files/5239429/GUIslice.zip)

Pconti31, 2020-09-17:
@ASMotionLab Well, I had reproduced your problem with TFT_eSPI and Adafruit's ESP32 Feather with 3.5 display using HX8357D and my fix to XTogglebtn worked fine.  I just ran your TestProg2 and its also correct.

So I just tried it with my Adafruit ESP32 with 2.4 display using ILI9341 and I can once again reproduce your error screen.  Ugh!  Anyway, back to the drawing board for me.   I'll post a fix when I figure out what's going on.
Paul--

Pconti31, 2020-09-17:
@ASMotionLab Once again I think I fixed your problem, at least your sample TestProg2  worked for me on a ILI9341 display.  Again, let me know if it works or not for you.  
Paul--
[XTogglebtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5241047/XTogglebtn.zip)

ASMotionLab, 2020-09-17:
Success! Thank you @Pconti31, it's fixed and my toggle button is great again 👍

Pconti31, 2020-09-17:
Super, I'll issue a pull request to Calvin and that will close this issue.
Paul--

closed 2020-09-18 by ImpulseAdventure