ImpulseAdventure/GUIslice #268: Update text in gslc_ElemCreateTxt_P

Avpi18, 2020-09-20:
Good day

I have noticed that it is difficult to update a text in a text field.

I have not managed to overwrite a text field in my case.

I have a .h file that the Builder made for me. In it I have a code: `gslc_ElemCreateTxt_P(&m_gui,E_ELEM_TEXT11,WAIT_PG,105,50,110,24,"Please wait...",&m_asFont[E_FREE_SANS9], GSLC_COL_BLACK,GSLC_COL_WHITE,GSLC_COL_WHITE,GSLC_ALIGN_MID_LEFT,false,true);`

In the .ino file, I want to change the displayed text ("Please wait...") constantly.
This does not work with `gslc_ElemSetTxtStr()`.

I am a beginner and apologize for the probably easy question.
Can you help me to solve this problem?

ImpulseAdventure, 2020-09-21:
Hi @Avpi18 --

Can you try setting `External Storage Size` in the Builder's element properties to the maximum length of the string you expect it to have after any updates?

DETAILS:
You have set the Builder to create an element with `Use FLASH API?` set to true (to put the text element in PROGMEM). However, if you would like a read-write field, we also need to specify a non-zero "External storage size". Setting "External storage size" to 0 makes it read-only. I think we might want to update the Builder User Guide to make this a little clearer (it currently refers to "runtime access").

Pconti31, 2020-09-21:
I'll update text external storage comments in the user guide for the GUIsliceBuider Release 0.15.b006
Paul--

Avpi18, 2020-09-21:
@ImpulseAdventure it works ! Thank you for answering :-)

I also think that it would be more understandable if there's a short explination in the users guide. Thanks for updating.

You can close the Issue.

Pconti31, 2020-09-21:
I should also mention a very good explanation was already in FAQ.pdf
Paul--

closed 2020-09-22 by ImpulseAdventure