ImpulseAdventure/GUIslice #269: 'struct gslc_tsGui' has no member named 'nSwapXY'

SunboX, 2020-09-21:
I'm trying to build GUIslice simple example with PlatformIO and getting the above message.
This is the project:

[DisplayTest (3).zip](https://github.com/ImpulseAdventure/GUIslice/files/5258323/DisplayTest.3.zip)

Pconti31, 2020-09-21:
@SunboX I think 
```
#warning No config selected in GUIslice_config.h - resorting to defaults.
```
is a pretty good hint.
Edit GUIslice/src/GUIslice_config.h and uncomment the driver you are using.
Have you checked  the wiki for how to set things up?
[https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino](url)
Paul--

SunboX, 2020-09-21:
Thx! Works.

Yes, I did read that page but did not understand that I have to uncomment somth in GUIslice_config.h

I included the correct driver in my main.cpp and thought that it will work like this.

And because of all those warnings in the console, I did not see this warnung you postet.

But thanks a lot! It works now. :)

closed 2020-09-21 by SunboX

ImpulseAdventure, 2020-09-22:
Good to see it is working. I plan on removing GUIslice_config_ard.h in the next release so this should help avoid a potential source of confusion.

Thanks for the followup