ImpulseAdventure/GUIslice #271: Flip toggle button

kinafu, 2020-09-24:
### Enhancement
Currently the toggle button is implemented the other way around, than expected by me and probably most of the users.

When the button is checked, the slider/thumb should be on the right side.
The unchecked position should be on the left.

Why?
[Here you can check](https://docs.microsoft.com/en-us/xamarin/xamarin-forms/user-interface/switch), that Android and iOS handle the buttons in the same way like suggested in this issue.
Maybe the often used iOS and Android implementation is the reason why the currently implemented GUISlice button does not feel right for me.

Below you can see the current implementation on the left and my idea on the right.

[image]

ImpulseAdventure, 2020-09-26:
Hi @kinafu — it seems like a reasonable request to adopt the more conventional representation. @Pconti31 as you contributed the toggle button originally, let me know if you have any concerns with me proceeding to make this change.

Pconti31, 2020-09-26:
@kinafu 
@ImpulseAdventure No Problem At my end.

ImpulseAdventure, 2020-09-27:
Visuals swapped now in latest version on GitHub.
@kinafu - please let me know if the update works as requested.

kinafu, 2020-09-28:
I just tried it out.
The change works for me as expected ✔️

closed 2020-09-28 by ImpulseAdventure