ImpulseAdventure/GUIslice #272: Pressure not reading correctly [at all?] from XPT2046

ASMotionLab, 2020-09-24:
Hi guys.

I have an ESP32 with an ILI9341 SPI display, using the TFT_eSPI driver.
All works, apart from touch (XPT2046) having a lot of contact noise.

Before I software debounce, I wanted to see if I can increase the min z pressure.

But after checking it via the calibrator, it's reading a solid 501.

[image]

Whereas with the XPT2046_Touchscreen library from Paul Stoffregen, pressure is being read correctly and very precisely from about 500 upwards (the blue line in the graph).

Am I doing something wrong in my setup, or is z not implemented in the XPT2046 that ships with GUIslice?

I've attached my GUIslice_config.h and esp-tftespi-default-xpt2046_int.h which I use.
[GUIslice_config.h_and_esp-tftespi-default-xpt2046_int.h.zip](https://github.com/ImpulseAdventure/GUIslice/files/5279779/GUIslice_config.h_and_esp-tftespi-default-xpt2046_int.h.zip)

Thanks!

ImpulseAdventure, 2020-09-25:
Hi @ASMotionLab -- thanks for providing some helpful details and the graph.

As you are aware, there are a couple options for using the XPT2046 touch controller.
- **esp_tftespi-default-xpt2046_int** (TFT_eSPI integrated touch controller)
- **esp_tftespi-default_xpt2046** (PaulStoffregen/XPT2046 touch controller, also called xpt2046_ps)

I'd like to confirm a couple additional details:
- When using **xpt2046_int** and run the diag_ard_touch_calib sketch, does each of the 5 corner crosshairs get detected OK? When it then displays the red dots at each press, does it appear to track ok?
- When using **xpt2046_ps** and run the diag_ard_touch_calib sketch, does each of the 5 corner crosshairs get detected OK? When it then displays the red dots at each press, does it appear to track ok?
- In each of the above, if the touch pressure is inconsistent, then you will likely see it advance to the next touch orientation randomly (as it does this after detecting a touch release)

Background details:
- The **TFT_eSPI** display library `getTouch()` doesn't return a Z / pressure value; just the count of valid events
- The GUIslice **xpt2046_int** mode implements the same TFT_eSPI filtering by default (you can disable it by commenting out `DRV_TOUCH_TFT_ESPI_FILTER` in GUIslice_drv_tft_espi.cpp).
- As the **xpt2046_int** mode reuses the same TFT_eSPI filtering logic (which just returns an active touch or not), the pressure reading given to GUIslice is simply `ADATOUCH_PRESS_MIN + 1` which explains the 501 you see.
- When you are using the **xpt2046_ps** driver, a range of pressure values should be detected.

Thanks!

ASMotionLab, 2020-09-26:
Very thorough reply, thank you for taking your time!

I didn't understand that PS library was a possibility, so I used the TFT_eSPI implementation (arguably it also makes for a cleaner setup in terms of config files - when using TFT_eSPI for many touch projects it helps to let it handle touch too).

When using **xpt2046_int** and running the diag_ard_touch_calib sketch, corner crosshairs get detected fine, but can be challenging to make a clean click. It's over sensitive - sometimes it advances to the next screen on it's own. Sometimes one touch will make 2 advances and have to start the calibration over. Always reporting 501 pressure (it's because like you say, my min pressure set to 500 and it's +1). When complete, dots are drawn succesfully but they are very far between. Seems like touch polling frequency is slow.

When using **xpt2046_ps** and running the diag_ard_touch_calib sketch, corner crosshairs are detected fine, fewer/no false clicks. Pressure data is is visible. When complete, dots are drawn much more rapidly, seems like polling is 10* faster, you can also see the text fields refreshing a lot faster. It's much more responsive.

Thank you for explaining the background detail.
I feel like I've wasted your time a little here, as I should have used the PS library to begin with - perhaps I was too eager to get going I missed the guidance in the docs to use this library?

I've now switched to esp-tftespi-default-xpt2046.h with  **#define DRV_TOUCH_XPT2046_PS** and commented out **#define TOUCH_CS 21** in my TFT_eSPI setup header file and everything is working great.

Thank you, your project rocks! I wish there were more larger SPI displays to make more use of the great GUI capabilities (speed not that important!)

ImpulseAdventure, 2020-09-26:
Excellent — thank you for providing the detailed test result and confirming that XPT2046_PS mode works well.

closed 2020-09-26 by ImpulseAdventure