ImpulseAdventure/GUIslice #273: Text background not transparent. ILI9341/TFT_eSPI

ASMotionLab, 2020-09-26:
Hi,

I hope you can help me once again.

I'm not sure if this is a bug or user error, so I didn't mark this as a bug.

Text background is not rendering transparent, there is a black box behind the text when fill is set to **false**
The code is correct: gslc_ElemSetFillEn(&m_gui,pElemRef,false); In my case it's generated from GUIslice.

The image should explain the issue clearly:
[image]

Is this another ILI9341 bug? Or are there some settings i'm missing.

Thanks a huge bunch!

ImpulseAdventure, 2020-09-26:
I will take a look to determine why the transparency doesn’t appear to be working in your example.

There could very well be an issue in the library for a particular display driver or font mode. Could you attach the following for me so I can narrow it down:
- Your sketch file
- Your user config (eg. esp-tftespi-default-xpt2046.h)
- If TFT_eSPI used, then your TFT_eSPI setup file referenced by User_Setup_Select.h
- What display model is being used (Ideally a link to the listing if you have it), and what MCU (eg. ESP32?)

Thanks!

ASMotionLab, 2020-09-26:
Thank you, all attached!
[GitIssueTextBg.zip](https://github.com/ImpulseAdventure/GUIslice/files/5286822/GitIssueTextBg.zip)

ImpulseAdventure, 2020-09-27:
Thanks @ASMotionLab -- in your TFT_eSPI User_Setup I see that `SMOOTH_FONT` has been enabled.
In this particular mode, it looks like there is an issue with transparent text output that needs fixing.

I haven't tested it yet, but I wonder if the following line added into GUIslice.cpp (additional line highlighted in blue) might help you to use SMOOTH_FONT mode: (alternately, commenting out SMOOTH_FONT in your user setup may work in the short term)
[image]

I would be interested to hear if the above modification works for you. Nonetheless, I will continue on to test this mode later and report back.
thanks

ASMotionLab, 2020-09-27:
Thanks @ImpulseAdventure 

With SMOOTH_FONT enabled:
Adding the colBg = pElem->colElemText; into GUIslice.c indeed works!
[image]

With SMOOTH_FONT disabled:
Background rendering works correctly by default like you suggested.
And works correct
[image]

I think I only enabled SMOOTH_FONT because that enabled SPIFFS for some example sketches I was trying.

Is there a performance penalty for having colBg = pElem->colElemText; in GUIslice.c by default?

ASMotionLab, 2020-09-27:
Sorry, I used a different example to illustratoe my findings, Here's the original sketch:
[image]

ImpulseAdventure, 2020-09-27:
Great -- thank you for the tests! I have integrated the fix into the latest GUIslice.
There will be no additional performance impact with this change above TFT_eSPI's transparent text rendering.

closed 2020-09-27 by ImpulseAdventure