ImpulseAdventure/GUIslice #275: Toggle Button Frame?

DonpK, 2020-10-06:
A toggle button frame can be enabled and a color specified using GUISlice/Builder, however, I don't see a frame on the display. It looks like the GUISlice code doesn't have a frame parameter. Perhaps I'm missing something.

Pconti31, 2020-10-06:
@DonpK Yes, I had to remove it recently. See issue `#266` to see what frames look like on many displays.
Paul--

DonpK, 2020-10-06:
Thanks, Paul. Sorry I missed the #266 discussion.

I installed the most recent version of the XTogglebtn.c fix posted in #266, however, I don't see frames on the toggle buttons. I am using [Adafruit 3.5" TFT-LCD ](https://www.adafruit.com/product/2050) with the HXD8357D controller. I've attached the files I'm using.
  Thanks, Don

[XTogglebtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5335335/XTogglebtn.zip)
[Toggle_Button.zip](https://github.com/ImpulseAdventure/GUIslice/files/5335336/Toggle_Button.zip)

Pconti31, 2020-10-06:
Ok, I'll take a look and see what I can come up with.
Paul.

Pconti31, 2020-10-06:
@DonpK Here is my fix.  Let us know if it works for you. Tested with HX8357 and ILI9341 controllers using both Adafruit_HX8753 lib, Adafruit_ILI9341 lib and TFT_eSPI's version of these two libraries.

[XTogglebtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5335885/XTogglebtn.zip)

NOTE: You must enable frames on the buttons to see them. 
I suggest the following change to your test_toggle.ino
```
  // ------------------------------------------------
  // Create graphic elements
  // ------------------------------------------------
  InitGUIslice_gen();

  gslc_ElemSetFrameEn(&m_gui,m_pElemToggle1, true);
  gslc_ElemSetFrameEn(&m_gui,m_pElemToggle1_3, true);
  gslc_ElemSetFrameEn(&m_gui,m_pElemToggle1_4, true);
```
Seems to be a bug in the Builder's code gen not outputting frame details.  I'll fix that in the next release of the Builder.

DonpK, 2020-10-07:
I replaced XTogglebtn.c  with the file in XTogglebtn_1006.zip

I'm afraid I don't see any frames on the Toggle Buttons. I've attached the latest .ino and .prj files with a screenshot and the XTogglebtn.c file from src/elem which should be the one in your XTogglebtn_1006.zip.

Don

[XTogglebtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5337831/XTogglebtn.zip)
[Toggle_Button.zip](https://github.com/ImpulseAdventure/GUIslice/files/5337815/Toggle_Button.zip)
[Toggle_Button_Screenshot.zip](https://github.com/ImpulseAdventure/GUIslice/files/5337816/Toggle_Button_Screenshot.zip)

Pconti31, 2020-10-07:
@DonpK Sorry, but I can't reproduce any problem using your uploaded files.  There was a bug but its been fixed.
I can't really get a clean shot of the HX8357 display I own to show you.  
Maybe try setting the frame to black will show something else.
```
  // Create toggle button E_ELEM_TOGGLE3
  pElemRef = gslc_ElemXTogglebtnCreate(&m_gui,E_ELEM_TOGGLE3,E_PG_MAIN,&m_asXToggle3,
    (gslc_tsRect){68,106,50,20},GSLC_COL_BLUE_LT1,GSLC_COL_RED,GSLC_COL_GRAY_LT1,
    true,false,&CbBtnCommon);
  gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_BLACK,GSLC_COL_GRAY_LT1,GSLC_COL_GRAY_LT1);
  m_pElemToggle1_3 = pElemRef;
``'
Be sure to make certain you are in fact using the new routine.   Otherwise, Calvin will need to look into it further.
Paul--

DonpK, 2020-10-07:
@Paul, I maybe misunderstanding the  issue. I replaced XTogglebtn.c with the file in XTogglebtn_1006.zip.  Using Builder to generate code, the following Toggle Button code is generated:

// Create toggle button E_ELEM_TOGGLE3
  pElemRef = gslc_ElemXTogglebtnCreate(&m_gui,E_ELEM_TOGGLE3,E_PG_MAIN,&m_asXToggle3,
    (gslc_tsRect){68,106,50,20},GSLC_COL_BLUE_LT1,GSLC_COL_RED,GSLC_COL_GRAY_DK1,
    true,true,&CbBtnCommon);
  m_pElemToggle1_3 = pElemRef;

No frame is visible. However, if I hand add the the gslc_ElemSetCol() function in the Toggle_Button_GSLC.h file tab of the sketch as shown below, the frames look okay. Maybe that's what you were suggesting all along.

// Create toggle button E_ELEM_TOGGLE3
  pElemRef = gslc_ElemXTogglebtnCreate(&m_gui,E_ELEM_TOGGLE3,E_PG_MAIN,&m_asXToggle3,
    (gslc_tsRect){68,106,50,20},GSLC_COL_BLUE_LT1,GSLC_COL_RED,GSLC_COL_GRAY_DK1,
    true,true,&CbBtnCommon);
    gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_YELLOW,GSLC_COL_GRAY_LT1,GSLC_COL_GRAY_LT1);
  m_pElemToggle1_3 = pElemRef;

Thanks,
   Don

Pconti31, 2020-10-08:
@DonpK Yes.  When I enlarged the picture you sent me of your buttons it looked like the frame was there but hard to detect with the default colors.  Going darker or in your case lighter I thought might show up correctly.  As I stated above I still need to fix the Builder since it's  ignoring turning frame on/off and setting its color.  That I'll fix in the next Builder release.

I have issued a pull request to add the fix of the toggle button to GUIslice API.
Paul--

DonpK, 2020-10-08:
Thanks, Paul. And thanks to you and Cal for the enormous amount of work you've done on this great project. These displays would be much more difficult to work with without GUISlice/Builder.
  Don