ImpulseAdventure/GUIslice #278: alternative keypad?

kellers76, 2020-10-18:
Hello,
I've been using your gui for a short time and I'm excited so far!
I have a pool project in which I would like to map a timer and my idea would be to create a numeric field where an alternative keypad pops up.
So far I haven't found how to refer to an alternative keypad.
i.e. I would like to define in the Builder that instead of a normal keypad, a keypad with hours from 0 a.m.-11 p.m. should be selected and a keypad with 5 minute steps should open in another field.
Can you help me?

best regards 

stefan

ImpulseAdventure, 2020-10-19:
Hi stefan --

With the new KeyPad component from #255 that is about to be released, it should be possible to create your own user variant keypads for arbitrary input (like time). I have put together a draft example of a **XKeyPad_Time** variant that lets the user select the hour, then minute and AM/PM from a single KeyPad instance targeting a single destination field/string. However, if you prefer to have two separate fields (one for hour and another for minutes) and use two separate keypads, that would also be possible.

Once the XKeyPad update has been integrated, I will post a followup here so you can try out the example I created.

cal

kellers76, 2020-10-28:
Hello,
I just searched for the described versions 0.16 of the API and the builder but couldn't find it.
I'm sorry for my stupid question, but how do I get the updates?

ImpulseAdventure, 2020-10-29:
Hi Stefan — there were a number of last minute changes required so the update took longer than anticipated. Paul and I also spent some time to update the Builder to work with the new APIs.

I hope to have a draft version posted by tomorrow for you to try out. Thanks!

ImpulseAdventure, 2020-10-30:
@kellers76  -- the changes for the new XKeyPad have been rolled into the GUIslice core library now. Please download the [latest from GitHub](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) (currently showing v0.15.1, which will be officially released later as v0.16, along with the Builder).

I am also attaching the example **XKeyPad_Time** widget I created for you to demonstrate how one can customize the new KeyPad to address requests such as your time control widget.

As for custom keypads -- if one simply wants to change the keys that appear within one of the existing KeyPad variants (eg. XKeyPad_Alpha), then it is a simple matter to update `/src/elem/XKeyPad_Alpha-setup.c` as needed. The Builder codegen should still work properly.

However, if you intend to create new keypad variants (such as your time entry widget), note that some customization of the code within the variant will be required -- the Builder won't be able to do this automatically for you. :)

After installing the latest GUIslice, try adding the attached **XKeyPad_Time** widget and try the example:
- [WIP278-KeyPadTime.zip](https://github.com/ImpulseAdventure/GUIslice/files/5463254/WIP278-KeyPadTime.zip)
- Copy `XKeyPad_Time*` into `/src/elem/`
- Run example `Int278-KeyPadTime1`

Some additional information about KeyPad configuration will be posted here: [Custom KeyPads](https://github.com/ImpulseAdventure/GUIslice/wiki/Custom-KeyPads)

thanks!

ImpulseAdventure, 2020-11-02:
Closing as the extended keypad functionality has been integrated. @kellers76 -- if you have any difficulties with using the new API to support your time controller after seeing the "KeyPad_Time" example, please feel free to reply and I'll reopen it.

closed 2020-11-02 by ImpulseAdventure

kellers76, 2021-01-24:
hi,
unfortunately I have a problem with my keypads.
I could not integrate the keypad Time 1:1 but only make it work by converting the keypad Alpha to the keypad Time.

In my current project I would like to have at least 4 different keypads, Num and Alpha ( standard ), plus my own Time and Speed.
The Speed keypad is a reduced Num keypad with filtered digits from 1-3 (I use it to make a motor speed).
I copied the element Num for the Speed Keypad and renamed all variables from Num to Speed.
In my sketch I copied all the definitions from Num and added them as Speed.

Unfortunately the whole thing doesn't work, when I start the call my program does a reset and restarts.
On the other hand, if I modify the Num keypad, then the call works.

Am I doing something wrong with the call? Can I use more than 2 keypads?

Translated with www.DeepL.com/Translator (free version)