ImpulseAdventure/GUIslice #282: Problems in GUIslice touch with WaveShare TFT 4” 480x320 + TFT_eSPI + TTGO T8

1ecu, 2020-10-26:
Hi,

I’m looking for a way to get touch functionality in GUIslice working with ESP32 TTGO T8 + TFT_eSPI + WaveShare TFT 4” 480x320 (ILI9486, XPT2046 / SKU 13587 / https://www.waveshare.com/wiki/4inch_TFT_Touch_Shield). Display’s connections to TTGO are: MISO 12, MOSI 13, SCLK 14, CS 15, DC 19, touch CS 4 and RST to RST (-1).

As background, TFT_eSPI and its examples (both Bodmer and ImpulseAdventures’ fork) work nicely with #define RPI_DISPLAY_TYPE, #define RPI_ILI9486_DRIVER and #define SPI_FREQUENCY  20000000. Touch sketches behave ok as well with calibration and so on.

GUIslice’s box example without touch is also ok. But after calibration sketch (with TFT_eSPI’s internal XPT driver) is started touch is very unresponsive on first page. A normal click gets you nowhere but somewhat “violent swipe” is needed. Then on corner calibration page picking and holding the points seem to work normally (steady values are shown) but sometimes it takes ages before the program moves to next point. When not touched the screen shows x = 0,  y = 4095, and z value appears to be ADATOUCH_PRESS_MIN +1. 

Calibration runs give an error message “ERROR: Missing corner segment (status=4)”. Status can sometimes be also 12. Program proceeds to the end but touch points are heavily off. Below are a typical recorded coordinates (with no remapping, swapping or flipping). 

CALIB: Config defaults: XMin=3900 XMax=200 YMin=3880 YMax=280 RemapYX=0
Capture TL: X=1380 Y=4005 Cnt=858, Z=(11..11)
Capture TR: X=201 Y=3959 Cnt=776, Z=(11..11)
Capture BL: X=2830 Y=1188 Cnt=339, Z=(11..11)
Capture BR: X=241 Y=1404 Cnt=238, Z=(11..11)

I’ve tried to calibrate also with combinations of points remap, swap and flip but outcome is the same albeit recorded coordinates can be different. I’ve also experienced with values from 1 to “n” for GSLC_TOUCH_MAX_EVT with no major effect.

On the other hand, if an external XPT2046 driver (P.Stoffregen or ImpulseAdventure’s fork) is defined for GUIslice nothing works i.e. screen is also blank. When trying XPT2046 library’s example TouchTest.ino (without interrupts) it didn’t work until I placed SPI.begin(14, 12, 13) before ts.begin().

Sorry for a bit lengthy story but it there any ideas about how to get the touch working in my setup?

ASMotionLab, 2020-10-26:
I had a similar experience with very similar hardware, I can't remember the exact details but check out this issue thread https://github.com/ImpulseAdventure/GUIslice/issues/272
 
I switched to DRV_TOUCH_XPT2046_PS driver and it all started working great. So for you it must just be a case of defining the correct pins in the xpt2046_ps settings

1ecu, 2020-10-27:
UPDATE:

When looking into GUIslice configs a bit more closely it turned out that the  display in question is actually directly supported... oops! So when selecting appropriate driver and adjusting the settings sketches started to work ok. Thanks anyway to ASMotionLab for commenting the initial issue which in this case remained unsolved - not that it should be solved in the first place!

closed 2020-10-27 by 1ecu

ImpulseAdventure, 2020-10-27:
Hi @1ecu --

As @ASMotionLab mentioned, we have often seen inconsistent touch response from the internal TFT_eSPI touch driver. In general, we recommend using the XPT2046_PS touch driver instead.

For future reference, can you let me know what config file you ended up using and what changes were required to the config for your Waveshare + TTGO T8?

Thank you for posting a followup, and glad to hear that it is working for you!

1ecu, 2020-10-28:
Hi @ImpulseAdventure,

when experimenting with TFT_eSPI I used the following libraries:

GUIslice v. 0.15.0
TFT_eSPI v. 2.2.23
XPT2046_Touchscreen-master (PS) v. 1.3.0

I've enclosed the config files. I'm sorry I wasn't that systematic so I didn't have all older config versions but reverted some changes to show settings' earlier states. That's why file save dates can be newer than my first posting. 

As per pin configuration, TFT_eSPI's User_Setup.h shows also my *present* SPI settings (HSPI-->VSPI) which I actually needed in order to get GUIslice running at all with XPT2046_PS (#define USE_HSPI_PORT didn't help). But still touch wasn't working. Anyway at that time I discovered the "ready-made solution" so I didn't bother trying SPI with initial pins anymore. And just to note there's currently nothing else installed or connected to the TTGO.

Well, one combination still could perhaps be TFT_eSPI +  ImpulseAdventure XPT2046_Touchscreen. I've however understood this library fork is meant particularly to address compatibility with ESP8266. Or does it have something to do with ESP32?

And talking about the combinations, is there any general opinion about what would be the most future proof graphical solution if utmost performance is not a top priority? Adafruit GFX is common but old and they themselves say it's becoming more or less a mess. So which path to take?

[Config.zip](https://github.com/ImpulseAdventure/GUIslice/files/5450431/Config.zip)