ImpulseAdventure/GUIslice #283: always get error Config: GSLC_SD_EN required despite already enabled

mcarosh, 2020-10-26:
### Describe the bug
Config: GSLC_SD_EN required for this program but not enabled. Please update GUIslice/config/

### Device hardware
- *MCU*: ??? ATmega2560
- *ard-shld-generic1_35_touch (https://www.amazon.de/Anzeigemodul-TFT-LCD-Display-Modul-480x320-Arduino-Vorstand/dp/B07Y6179LG/ref=sr_1_22?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2G0O6WUT2E2Z0&dchild=1&keywords=arduino%2Btft%2Btouchscreen&qid=1603752162&sprefix=arduino%2Btft%2Caps%2C234&sr=8-22&th=1)
- ard-shld-generic1_35_touch

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
Everything works fine 

### Expected behavior
Arduino should compile

### Initialization messages
Config: GSLC_SD_EN required for this program but not enabled. Please update GUIslice/config/
### Additional info
I really do not know what is my fault??? Already tried other setups but always get this error despite already enabled in GUIslice_config_ard.h  Is there a further setting which should be done?
Thanks for help
mcarosh

Pconti31, 2020-10-27:
@mcarosh I suggest you start by reading the wiki pages.  In your case look at:
[Arduino Mega](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-Arduino)
It will explain things better than I can.  Basically you are editing the wrong file.

Look down the wiki page for your TFT display driver and touch driver, say it's ILI9341 with stme610 touch (adfruit 2.8 display) in that case it calls for `ard-adagfx-ili9341-stmpe610.h`  you go into GUIslice/config and find this file and edit it to turn on the SD card support.  

Now as the guy who wrote the Builder I would like to be able to output a better warning I'll change it in the next release to point people back to the wiki.
Paul--

ImpulseAdventure, 2020-10-27:
@mcarosh -- as Paul noted, the `/src/GUIslice_config_ard.h` file is not used anymore and has been replaced by the various user configuration files in `/config/`. You should find a `GSLC_SD_EN` setting within Section 6 of your config; this will need to be set to 1. I plan on removing GUIslice_config_ard.h in the next version to avoid any confusion -- it was there to enable the library to compile out-of-the-box without any modifications (for a particular display setup), but as users generally need to make some changes to their config files, it probably makes more sense to remove it.

thanks

mcarosh, 2020-10-27:
Thx a lot
I managed Arduino to compile code :)
But display will not initialize???Is there still the problem with sharing pins between LCD-data and touch pins at the mcufriend library??? I can run successfully every example code from the mcufriend library but generated code from GUIslice will not initialize the display???
And again thanks for your help 
mcarosh

mcarosh, 2020-10-27:
Ha
Solved it :) Forgot to change the SD_CS-pin to 53. Now everything is working.
And again, really big thx to this great program.
mcarosh

closed 2020-10-27 by mcarosh