ImpulseAdventure/GUIslice #285: creating a graph

mcarosh, 2020-11-01:
Sorry for this silly question, but I´m not able to integrate graph
Trying to show graph from temp but do not know where to put the "link to temp" part in code :/ Already studied the graph example but this is more confusing to me as it is linked to a slider. I think it is probably easy but I m always stuck.
Maybe someone have a better example or hint for me.
Ah...and used material is a Arduino Mega Mini with ILI9486-Display. Everything is working fine so far (Touch, showing Temp,etc)
Thx in advance
mcarosh

Pconti31, 2020-11-02:
@mcarosh check issue [241](https://github.com/ImpulseAdventure/GUIslice/issues/241)

I attached my sample using XGraph no slider involved.
I hope it helps.

[ex_bld_graph_sin.zip](https://github.com/ImpulseAdventure/GUIslice/files/5477127/ex_bld_graph_sin.zip)

Paul--

mcarosh, 2020-11-04:
thx a lot Paul :)
Now I made it :) Just one more question :/ If I draw the graphbox bigger than 250 pt in x-direction the graph starts in the middle of the box??? is the box less than 250 pt everything works fine. Where is my fault? :/
But so far I say really thanks, 
You made my day :)
mcarosh

Pconti31, 2020-11-04:
@mcarosh I doubt its your problem.  Mostly likely a bug inside the XGraph.  If you post a zip file of your project that has this problem I'll see if I can debug it.  Heplful if you comment out any sensor reading and supply dummy data.
Paul--

mcarosh, 2020-11-05:
Hello Pconti
Thx for reply. As i wrote I use a Arduino mega with a Ili9486-TFT. The attached sketch (zip) is exactly the size I would like to use but there it always shows only the first (estimated) 20 pixels in x-direction. If I limit the width of the graph to 265 pixels it works perfectly. Maybe you can figure out the problem ;)
So far thanks again and stay brave
greez
mcarosh
[bld_graph_sin.zip](https://github.com/ImpulseAdventure/GUIslice/files/5495841/bld_graph_sin.zip)

Pconti31, 2020-11-05:
@mcarosh Caused by bug inside XGraph that was using uint8_t instead of uint16_t in a few places.
Attached is a new XGraph.c zipped.
Let me know if it works for you.  If so I'll make a pull request.
Paul--
[XGraph.zip](https://github.com/ImpulseAdventure/GUIslice/files/5497206/XGraph.zip)

mcarosh, 2020-11-05:
Hi Pconti31
Unfortunately it does not work :( There is is still some space empty and now the graph is not shown correct. Here is a picture:
[image]
Furthermore I should mention that the gap of the graph is switching. Sometimes it´s on the left side and then switches to the right. Sorry for the bad news and maybe there are some settings wrong on my side. But you have my sketch.
Nevertheless thanks for your efforts.
mcarosh

Pconti31, 2020-11-06:
@mcarosh Sorry, I thought your problem was with width of the graph which is what I fixed.  Your gaps are caused by a math issue.  I thought your mod to my example was to simply demo the width problem. I didn't realize you didn't understand the math.

First to graph anything you need to understand the data set.   You created a XGraph 400x90 this means you only have 90 points for y axis.  Now you are using my test program and cut the y from 200 to 90 which wasn't the best try.  Why?  The program is graphing sin() and what is the range of sin()? -1 to 1 as a floating point number.   Now the XGraph can't graph floating point numbers so I used a mapping function to map the values from -100 and 100 to range 50 to 150 after I multiplied sin(x) by 100. This gives me an integer that represents a floating point number to two digits  which on a y of 200 looks fine.
Ok, lets change yours Graph from 400,90 to 400,100.  This will look a little better but still create gaps due to converting floating point to integers.   Now lets see what you did with your map function:
```
  int16_t nY =  map((int16_t)(y*100), -50, 50, 25, 60);
```
Your -50 to 50 is cutting out all sin() values from -0.51 to -1.0 and 0.5 to 1.0 
I hope you can see why you have a gap.
Now if you instead do
```
  int16_t nY =  map((int16_t)(y*100), -50, 50, 25, 85);
```
The graph will look a lot better but not perfect due to the size of Y.
So that ends the lesson.
Paul--

mcarosh, 2020-11-06:
Hi Paul
Yes the problem is with the width of the graph and I hope my understanding of the map-function is good enough. With the map I will only modify the amplitude of the sin-function. For the period I need to modify the x-value. Here are some pictures of what I meant with gap. I should have better wrote "offset" instead. And this happens only with your new Graph.c-file.
First picture shows the graph (in this case after 1 period), The second shows the graph after ca. 3 periods and the third one is ca after 8 periods. As you can see it´s not about the small gap between linked lines. Furthermore there is still some space on the very right sight. Hope I could clarify the real problem I encounter.

Nevertheless thanks for your efforts.
mcarosh

mcarosh, 2020-11-06:
Here are the pictures:

[image]
[image]
[image]

mcarosh, 2020-11-06:
I think it´s a problem with the addressing of the pixels???

Pconti31, 2020-11-06:
@mcarosh Sorry, but my version doesn't do that.  I attached my current mods to the one you posted.
basically simply changing the map function to 
```
  int16_t nY =  map((int16_t)(y*100), -100, 100, 15, 85);
```
and the number of data points to match the width of the x axis you set which is 400.

[bld_graph_sin.zip](https://github.com/ImpulseAdventure/GUIslice/files/5501495/bld_graph_sin.zip)

I'm not sure what you are expecting here.  The first reply I made pointed you to closed issue 241 where I stated:

_I agree this control needs some major work to be useful, like being able to **input or scale the X position**.
Having labels and axis, an array of array of points to show multiple line graphs and so on...
By the way, only dots and fill exist. Line is not implemented.
Anyway, attached is a simpler example showing a sin wave. I hope it helps._

And Calvin (ImpulseAdventure) responded:
_Hi guys -- yes, the XGraph could definitely do with some rework. It was only meant as a simple example of what could be done,_

Frankly, I need to get back to getting the Builder ready for the next release so I'm not volunteering to rewrite this control or debug any deeper.  If you are careful my modified XGraph control should work for your temperature application but its not going to work for plotting equations or any app where X needs to be input unless you simplify them greatly.
Paul--

mcarosh, 2020-11-06:
Sorry
It was not my intention to bother you. Just wanted to give some feedback. If you say you do not encounter such behavior with your mods, the fault need to be on my side. I really appreciate your great work. 
In this sense thx again for your efforts.
mcarosh

closed 2020-11-06 by mcarosh

ImpulseAdventure, 2020-11-07:
Hi @mcarosh -- sorry for the delay in responding here...

I agree with you that the existing XGraph lacks a number of features that would make your application easier to implement, as it was just meant as a simple demonstration, not a functional control.

That said, I have created a new issue to capture a request for discussion on what features would be desired in a proper plot control. Please feel free to comment in the thread if you feel that there are things that might make it easier to achieve what you are looking for.

Thanks for your feedback!