ImpulseAdventure/GUIslice #287: M5Stack core can't build with arduino ide

huang443765159, 2020-11-05:
NOTE: For issues with the GUIslice Builder, please file a report instead at:
https://github.com/ImpulseAdventure/GUIslice-Builder-source/issues
Hello, I am a new Arduino.
I am use M5Stack core with arduino, but when I build examples I always get err like this:
[image]
Am I missing something that I have to install？
Thanks.

Pconti31, 2020-11-05:
@huang443765159 The error indicates GUIslice thinks you are configured for Arduino not M5Stack.
Did you read the wiki?
[Configure GUIslice for M5stack](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-M5stack)
Paul--

ImpulseAdventure, 2020-11-05:
Hi @huang443765159 -- as Paul mentioned, it is likely that the configuration selected in `GUIslice_config.h` needs to be changed.

Please try selecting **esp-shld-m5stack.h** in `GUIslice_config.h` by uncommenting that line (and ensuring all other config lines are commented out):
[image]

Does that correct the issue for you?
thanks

huang443765159, 2020-11-06:
Thanks for your reply. 
Sorry, I may have missed reading the document.I'll try to modify and build again.

closed 2020-11-06 by huang443765159