ImpulseAdventure/GUIslice #289: TOP/BOT alignment of text within Text Button when frame is selected

farfarfromhome, 2020-11-09:
### Describe the bug
When creating gslc_ElemCreateBtnTxt(), with gslc_ElemSetTxtAlign() and GSLC_ALIGN_BOT_MID or GSLC_ALIGN_TOP_MID, with gslc_ElemSetFrameEn() the text overwrites the frame.

### Versions
GUIslice: Version: 16.0
Platform: tft_espi
Touch Driver: esp-tftespi-default-xpt2046.h
TFT Driver: ILI9488

### Expected behavior
 I am not sure whether this is intended behavior or not but it would be nice to offset the text so as not to erase the border.  This could be done by either a permanent offset when the border is selected or a parameter for the offset.   My preference to keep it simple is a permanent offset.

ImpulseAdventure, 2020-11-10:
Thank you for noticing this! (it is easy to miss on the small displays)

I have now pushed a fix that will inset the text region when a frame is enabled.
Please let me know if this corrects the overlap that you observed.

Note that you also have the option of manually specifying a margin for the text within an element.
For example, the following will add a 2 pixel margin horizontally and 5 pixel vertically:
```c++
gslc_ElemSetTxtMarginXY(&m_gui,pElemRef,2,5);
```

farfarfromhome, 2020-11-10:
Thanks for the super quick response again!!  I tested and the change looks just perfect.

Thanks for pointing out the gslc_ElemSetTxtMarginXY() function.  However since I am anticipating 10+ pages in my project I am trying to stay within the bounds of the builder generated code.  To this end I did put a request into Paul to add the capability of selecting the TOP/BOT text alignment.  Its not so important that it works on the builder text field, just that the alignment fields get entered into the xxx_GSLC.h file to minimize my editing on each pass.

ImpulseAdventure, 2020-11-12:
Thanks @farfarfromhome for the followup.... good to hear that it is working.

I understand from Paul’s post in the Builder repo that he may have addressed your second request in the v0.16 release.

closed 2020-11-12 by ImpulseAdventure

farfarfromhome, 2020-11-12:
Hi Cal

 

Yes, I downloaded this evening and everything is working on the builder.

 

I am making great progress and am very enthusiastic with GUIslice and the Builder.  I made sure I starred Paul today, he sounds a bit down!