ImpulseAdventure/GUIslice #291: XKeyPad Compile Error in Ver. 0.16.0 But Not 0.15.0

DonpK, 2020-11-12:
### Describe the bug
I'm getting the Arduino IDE compile error shown below involving the XKeyPad element. No changes were made to the sketch. The sketch compiles fine with GUISlice Ver. 0.15.0 but generates the error when using Ver. 0.16.0. I can upload the sketch if the error message doesn't provide enough information.

Kitchen_Thermometer_V1.2_GSLC.h:99:1: error: 'gslc_tsXKeyPad_Num' does not name a type
 gslc_tsXKeyPad_Num              m_sKeyPad;
sketch\Kitchen_Thermometer_V1.2_GSLC.h: In function 'void InitGUIslice_gen()':
Kitchen_Thermometer_V1.2_GSLC.h:226:56: error: conversion from 'gslc_tsXKeyPadCfg_Num' to non-scalar type 'gslc_tsXKeyPadCfg' requested
   gslc_tsXKeyPadCfg sCfg = gslc_ElemXKeyPadCfgInit_Num();
Kitchen_Thermometer_V1.2_GSLC.h:227:44: error: 'gslc_ElemXKeyPadCfgSetFloatEn' was not declared in this scope
   gslc_ElemXKeyPadCfgSetFloatEn(&sCfg, true);
Kitchen_Thermometer_V1.2_GSLC.h:228:43: error: 'gslc_ElemXKeyPadCfgSetSignEn' was not declared in this scope
   gslc_ElemXKeyPadCfgSetSignEn(&sCfg, true);
Kitchen_Thermometer_V1.2_GSLC.h:232:6: error: 'm_sKeyPad' was not declared in this scope
     &m_sKeyPad, 10, 100, E_AO_10X16PT7B, &sCfg);
exit status 1
'gslc_tsXKeyPad_Num' does not name a type

ImpulseAdventure, 2020-11-12:
Hi Don —

Normally, the Builder should be able to upgrade your sketch from 0.15 to 0.16. Please try opening your project in the latest Builder (0.16), regenerate your code and let me know if that helps.

More details on the migration for the XKeyPad were described in the release notes:
https://github.com/ImpulseAdventure/GUIslice/wiki/Migration-Notes-(XKeyPad)

Thanks

DonpK, 2020-11-12:
Hi Cal,

I didn't realize Builder 0.16 was released. I installed it and the sketch compiles fine. 

Thanks, as always, for your great work and help.

Don

closed 2020-11-12 by DonpK