ImpulseAdventure/GUIslice #292: Button callback waits before refreshing the UI?

koichirose, 2020-11-17:
I have a button which should make a HTTP request and update a text in the UI.
I'd like to set the text to `Loading`, then run the request and then update it again when it finishes:

```
bool CbBtnCommon(void* pvGui, void *pvElemRef, gslc_teTouch eTouch, int16_t nX, int16_t nY)
{
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui, pElemRef);
  char            acTxt[MAX_STR];

  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BRITA:
        make_request("MY URL");
        break;
//<Button Enums !End!>
      default:
        break;
    }
  }
  return true;
}

String make_request(char* url)
{
  set_response_text("Loading");
  //request() is in another file
  String response_text = request(url);
  set_response_text(response_text);

  return response_text;
}

void set_response_text(String text)
{
  gslc_ElemSetTxtStr(&m_gui, m_pElemOutTxtResponse, text.c_str());
  gslc_ElemSetTxtAlign(&m_gui, pElemResponse, GSLC_ALIGN_MID_MID);
}

```

With this, it only shows the HTTP response text and not the `Loading` text.
It seems that the UI will only update when the callback returns. The button itself is like in an active state while performing the HTTP request.

Is there a way to do it? Sorry if this has been asked before or was documented, I could not find it.

Pconti31, 2020-11-17:
@koichirose I dunno, but I would try:
```
String make_request(char* url)
{
  set_response_text("Loading");
  // refresh display
  gslc_Update(&m_gui);
  //request() is in another file
  String response_text = request(url);
  set_response_text(response_text);

  return response_text;
}
```
Otherwise break this routine up into two pieces, one to send request and another to wait for reply.  Create a global state variable, set inside the make_request, set it say WAITINGREPLY when you send request and inside your loop() check this state variable for WAITINGREPLY  to see if state requires you to look for a response from the server then you get it set the state variable to NOREQUEST and set your text variable.  Of course, this works best if you don't actually wait, or pend for the reply inside your new routine.
  
Paul--

ImpulseAdventure, 2020-11-18:
Hi Sergio — I agree with Paul’s suggestion here. Fundamentally, your request() function is blocking, so that typically means that all other sketch operations on the micro (including the GUI) will be paused during that time.

Calling gslc_Update() will allow a final redraw to be done (clearing the “active” appearance) before you enter your blocking request() call.

As Paul alluded to, writing your request() function to be non-blocking would be far better, as one could still enable the GUI (and other operations) to continue while you are awaiting a response.

koichirose, 2020-11-18:
Thank you both for your answers.
I'm not sure how to make the request non-blocking. I'm using this HTTP client: https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPClient/src
It seems that the `send_request` function there is blocking.

The `gslc_Update` you suggested works, but the text looks weird when it is set.
I edited my callback like this:
```
  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    lastTouch = (unsigned)time(NULL);
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BRITA:
        make_request("URL");
        break;
      case E_BTN_BLACK:
          gslc_SetPageCur(&m_gui, E_PG_BRITA);
        break;
//<Button Enums !End!>
      default:
        break;
    }
  }
```

and in my `loop`:
```
void loop()
{
  gslc_Update(&m_gui);

  //switch to a black page if no touch is detected for 5 seconds - is there a better way? TODO
  if ((unsigned)time(NULL) - lastTouch > 5) {
    gslc_SetPageCur(&m_gui, E_PG_BLACK);
  };
}
```

And modified the set_response_text function like you suggested:
```
void set_response_text(String text)
{
  gslc_ElemSetTxtStr(&m_gui, m_pElemOutTxtResponse, text.c_str());
  gslc_ElemSetTxtAlign(&m_gui, m_pElemOutTxtResponse, GSLC_ALIGN_MID_MID);
  gslc_Update(&m_gui);
}
```

Here's a [short video](https://imgur.com/a/ZPaKXwO) with the whole process. The server responds with `Done` and the first time that is set as the response text it looks a little garbled. Then the screen goes "off" (because of the timeout) and when I touch it to turn it on the `Done` text is now looking fine, don't know why.
I tried tinkering a bit (changing fonts, sizes, triggering further updates manually) but couldn't fix it.

Sorry for the long post and the not great code, I'm just getting started with this.

Pconti31, 2020-11-19:
@koichirose Without seeing your whole program its hard to say what is broken although it looks like the text field refresh is broken using the update this way.   Try, commenting out the actual calls to http and just put in a delay after the update command and then  just stuff in the "Done" string to see what shows up on screen.  If that is also broken post a zip file of this smaller project and include all of the files.  This will give  @ImpulseAdventure aka Calvin something to debug.
As for the HTTPclient you can't make it non-blocking.   You can however use a non-blocking library like:
[asynchronous HTTP for ESP](https://github.com/boblemaire/asyncHTTPrequest)
Paul--

koichirose, 2020-11-19:
Here's a stripped down project, it has the same UI as my original project, but doesn't actually connect to WiFi nor make HTTP requests. Just a simple button that sets a text, waits a bit, then sets another text.
In this case, the `Done` text is a little less garbled than the one in the original project.

@Pconti31 Thanks for your HTTP library suggestion, I'll definitely try it.
[TFTTest.zip](https://github.com/ImpulseAdventure/GUIslice/files/5567313/TFTTest.zip)

Pconti31, 2020-11-20:
@koichirose  I looked your app over and I think you only need to increase the 'E_ELEM_RESPONSE' text element from a height of 13 to a height of 15.
Paul--

ImpulseAdventure, 2020-11-20:
Agreed. I believe the only issue is the slight overdraw from the previous text update. Does it look like the Builder’s text max dimensioning calc might be slightly off for the FreeMono24’s ascenders / descenders?

koichirose, 2020-11-20:
That was it! Not sure why the second time it got displayed correctly though.
Thanks, I'm closing this.

closed 2020-11-20 by koichirose