ImpulseAdventure/GUIslice #293: Guru Meditation Error: Core  1 when calling gslc_ElemSetTxtStr

notascooby, 2020-11-19:
Get a Guru meditation error with version 16.0.
I have traced this to a call to gslc_ElemSetTxtStr from my sketch, I perform the call as follows.
      snprintf(acTxt, MAX_STR, "%s", myString);             //update acTxt with variable 
      gslc_ElemSetTxtStr(&m_gui, m_pElemWTime[i], acTxt);   //Update element ready to send to screen
(MAX_STR = 100 and number of characters within myString is around 22)
The code ran perfectly with GUIslice version 15.0.
Looking at your GUIslice.c code I see that the gslc_ElemSetTxtStr routine has changed.
You have moved:
  if (pElem->nStrBufMax == 0) {
    // String was read-only, so abort now
    return;
  }
to a new routine gslc_StrCopy called within gslc_ElemSetTxtStr , so it seems to me that if the above expression is true the remaining code in gslc_ElemSetTxtStr will still run.
I added   
    if (pElem->nStrBufMax == 0) {
    // String was read-only, so abort now
    return;
  }
to gslc_ElemSetTxtStr and all my existing code runs correctly.

Hopefully I have supplied enough information.

### Device hardware
- *MCU*: ??? ESP32
- *Display with link*: ??? ILI9486 3.5" RPI
- *GUIslice config selected*: ??? `esp-tftespi-default-xpt2046_int

ImpulseAdventure, 2020-11-22:
Thanks for reporting this and identifying the difference; much appreciated!

I’ll fix this up and likely roll it into a v0.16.1 release given its impact.

ImpulseAdventure, 2020-11-23:
Hi @notascooby -- looking at the code again, I think the problem may have been triggered by the bad range in `strncmp()` around the code that you had identified.

I have pushed an update that should prevent this from occurring.
Please give the latest a try and let me know if it changes anything for you.

If it still causes an error, please attach your sketch so that I can ensure I am testing the same condition as you.

thanks

notascooby, 2020-11-23:
Yes, the update works correctly - thanks

closed 2020-11-23 by notascooby