ImpulseAdventure/GUIslice #296: Three exclamation marks in a row

mcarosh, 2020-11-23:
### Describe the bug
Sketch will not upload (properly) if I create Text with three exclamation marks in a row

### Device hardware
- *MCU:  ATmega2560
- *Display: Generic  ILI9486 3.5" 

Hello
I´m not really sure if this is a bug (or a feature ;) but when I try to create Text with three exclamation marks in a row I am (or better Arduino IDE) not able to upload the sketch. When I put a space between the exclamation marks or only write two of them everything works fine. Tried different font styles but that doesn´t matter. So to me it´s really strange but not a big problem. Just wanted to inform you that there is a little devil in the detail ;)
Put an example-sketch to the attachment which I am not able to upload because of the three exclamation marks. It is in the E_ELEM_TEXT2.

[Longpress-Btn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5586372/Longpress-Btn.zip)

best regards

Pconti31, 2020-11-24:
@mcarosh Your code gen was using Builder 0.15.b005.  I just opened your example with 0.16.b001 Builder and added your
" !!!" to E_ELEM_TEXT2, did a save and code gen without errors.  Using Adruino IDE 1.8.12 and GUIslice API 0.16.0 it compiled and ran fine using a Arduino Mega.   
It might help us help you if you gave some details about what you mean by not upload properly, what MCU you are using and rev of Arduino IDE.
Paul--

mcarosh, 2020-11-25:
Updated to latest GUIslice API 0.16.0 and 0.16.b001 Builder. But still encounter this phenomenon.
I am using Adruino IDE 1.8.13 on a ATMEGA2560 16U-HT MCU with a CH340G USB to Serial (TTL) Module. (Arduino MEGA Mini). And by "not upload properly" I mean Arduino compiles without error and start the upload like normal but cannot finish. I will not get the "upload done" message from Arduino and the program itself is also not uploaded in a correct way. After resetting the board the program will not start (white LCD). But this only occurs when I write "!!!". As I already wrote this is no problem to me but it is really strange, isn´t it?
Regards
mcarosh

Pconti31, 2020-11-25:
@mcarosh Ok, so I took your zip file again, unzipped it again, made no modifications and simply compiled and uploaded using Arduino IDE 1.8.13 instead of 1.8.12 and everything worked fine.   Now I'm about to lose my system to a Microsoft update.  The last time took a few days to get going again. So not sure when I'll be back...  Anyway, in the IDE goto file->preferences and check  **Show verbose output during:  Upload** [checked]
This is what I get:
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\Paul\AppData\Local\Temp\arduino_build_14061/Longpress-Btn.ino.hex"
avrdude: writing flash (43826 bytes):

Writing | ################################################## | 100% 7.04s

avrdude: 43826 bytes of flash written
avrdude: verifying flash memory against C:\Users\Paul\AppData\Local\Temp\arduino_build_14061/Longpress-Btn.ino.hex:
avrdude: load data flash data from input file C:\Users\Paul\AppData\Local\Temp\arduino_build_14061/Longpress-Btn.ino.hex:
avrdude: input file C:\Users\Paul\AppData\Local\Temp\arduino_build_14061/Longpress-Btn.ino.hex contains 43826 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 5.32s

avrdude: verifying ...
avrdude: 43826 bytes of flash verified

avrdude done.  Thank you.
```

mcarosh, 2020-11-25:
Hi again
Here is the output:

`avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM16
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\mcarosh\AppData\Local\Temp\arduino_build_177160/Longpress-Btn.ino.hex"
avrdude: writing flash (47044 bytes):

Writing | #################################################avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
`

And this happens only with this !!!
With two of them or a space between everything works well. But as I mentioned before this is not a big deal. If this happens only to me we can close this issue because a space between looks more pretty ;) The only thing is, that this is really strange.
And thanks a lot for your efforts
best regards
mcarosh

ImpulseAdventure, 2020-11-25:
What you are observing is not actually to do with GUIslice but more to do with a "feature" of the upload process.

The three exclamation points `!!!` is apparently an escape character for the ATmega2560 to enter monitor mode, which will interfere with the upload. There are more details [here](http://avr-developers.com/bootloaderdocs/index.html) and forum posts such as [this](https://forum.arduino.cc/index.php?topic=132595.0).

mcarosh, 2020-11-25:
Ha :)
Thx a lot for the clarification.  This need to be the reason. 
And if I find time and inspiration I am going to look for a "fixed" bootloader version. 
But for now I am happy that this "feature???" is not because for lack of my coding-skills :)
So thanks again for not let me going crazy. This was really strange.
greez
mcarosh

closed 2020-11-25 by mcarosh