ImpulseAdventure/GUIslice #297: Limited Touch functionality on RM68090 board (MCUfriends copy)

michaelwilkinson1992, 2020-11-25:
### Describe the bug
*unable to calibrate simple 4 wire resistive touch *

### Device hardware
- *MCU*: ??? ESP32 UNO board (Pins have been changed according to the info in the ESPiTFT library)
- *Display with link*: ??? 2.8" resistive MCU copy (https://www.banggood.com/Geekcreit-UNO-R3-Improved-Version-+-2_8TFT-LCD-Touch-Screen-+-2_4TFT-Touch-Screen-Display-Module-Kit-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-1428291.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=DE&utm_content=minha&utm_campaign=minha-de-en-pc&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=frank-ssc-de-all-en-0608feed&ad_id=440288844860&cur_warehouse=CN)
- *GUIslice config selected*: ??? `ard-adagfx-MCUFriends or ESPi_TFT`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.* all walkthroughs have been followed and extensive googleing has been carried out to no avail....

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* ??? Display functions perfectly in all manner (both display AND touch) and can be successfully calibrated using examples from MCU Friends library.
- *Touch works correctly with standalone touch library examples without GUIslice:* ???
Touch works using Adafruit Touchscreen demo (raw data from all areas of the screen can be observed via seriell monitor.
Touch pins are as follows: XP: 14, XM: 4, YP: 15, YM: 27 (confirmed via calibration example in MCUFriends library)
All other pins are standard UNO shield layout/ESPuno layout.

### Expected behavior
*A. I should be able to select MCUFriends 4Wire config file, define touch pins, and screen should function with touch.

or

B. I should be able to configure the ESPi_TFT library, select and configure esp-tftespi-default-simple and display should function with touch*

### Actual behavior
*only a small area of the touch screen is active, and calibration does not work.*

### Additional info
*In order for ard-shld-mcufriend_4wire to function, the ADA_simple_touch workarounds need to be copied from GUIslice_drv_tft_espi.cpp and added to GUIslice_drv_adagfx.cpp, otherwise code is unable to compile for the ESP32 due to port names? and pin definitions?
Once completed the display functions, however touch cannot be calibrated. Its as though touch only works on the lower right corner. The best way I can explain it is the touch screen has shrunk, you can select any part of the display, but only from a small area of the touch screen.

The display will also function if GUIslice_drv_tft_espi.h is used, however  DRV_DISP_ADAGFX and DRV_DISP_ADAGFX_MCUFRIEND need to be defined in esp-tftespi-default-simple.h, otherwise display shows only a white screen. (also unable to funtion externally using TFT_eSPI library. (I suspect forcing the display ID could be a workaround for this, but I havent found a way to configure it yet).

Using either configure files with default settings results in a black screen thats pixelated in different colors (I suspect this is due to sharing pins between display and touch, as outlined in the ADA_Simple_touch workarounds). However, ard-shld-mcufriend.h works. *

michaelwilkinson1992, 2020-11-26:
### Problem identified
1. *analogread was not set to 10 bit*
2. *RM68090 not supported by TFT_eSPI library*

## Solution
*ESP32 workarounds need to be added to GUIslice_drv_adagfx.cpp (including the following!!)*  
`#if defined(FIX_4WIRE_ADC_10)
      analogReadResolution(10);
    #endif`
*(its located much further down on the page, so i missed it the first time when adding the workarounds)*

*This will allow the code to compile for the ESP32 using DRV_DISP_ADAGFX, DRV_DISP_ADAGFX_MCUFRIEND,  and DRV_TOUCH_ADA_SIMPLE drivers. (target device set to Arduino)*

### Please implement this to a new config file eg. ESP32_RM68090 
*I have spoken with the guys from the TFT_eSPI, they have no interest in finding workarounds for this diplay, so I reccomend a config file for this perticular display in conjunction with ESP32 that doesnt rely on their drivers, using the workaround outlined here*

ImpulseAdventure, 2021-03-11:
@MickTheMechanic -- my apologies as I had overlooked this issue. Not sure if it is still useful for you, but I have now integrated a change into the master branch that auto-detects ESP32 and enables the 10b ADC workaround for the Adafruit 4-wire touch mode.

You did mention that you'd like to see a specific config file for the ESP32 RM68090. If this is still helpful, please attach the version you are working to this ticket so that I can integrate it or see how it differs from existing config files. If further changes are needed to the core GUIslice library for this, please let me know and I'll reopen the issue. thanks!

closed 2021-03-11 by ImpulseAdventure

MickTheMechanic, 2021-03-11:
Hi, yeah its not super important to me anymore, but generally somewhat important nonetheless. 

These partictular screens are sold on Banggood and are quite common, but there are no functioning drivers in the TFT_ESPI library, making using them with an ESP32 a bit tricky. 

The first problem was regarding touch, but it looks like you have that covered. 

The second problem was a custom config was needed. The workaround was to use ard_mcufriends config file, but this means you have to set the target device to Arduino in the builder, and SPIFFS will not be supported. 

Im not sure if you want to pursue this further, perhaps a note in the config file outlining the workaround and its limitations would be enough.

kekanath, 2024-08-04:
HI I am now to this,
I am using ESP32 nodemcu with RP6809 for image display from the SD card could you please help me here