ImpulseAdventure/GUIslice #299: VGA driver

bil-ash, 2020-11-27:
### Feature Request or Enhancement
I would like GUIslice to be able to output 640*480@60/50Hz directly to a VGA monitor using a VGA connector on moderately powerful boards like arduino due, esp32 or a more powerful one like linkit smart

### Additional context
I would like the driver to implemented in only c/c++(no assembly) and be generic in nature so that the user can provide the required timer,pwm and gpio board-specific functions,this will ensure portability in future. There exists a demo of the same at https://github.com/abelykh0/VGA-demo-on-bluepill but it is STM specific.

ImpulseAdventure, 2020-11-30:
Hi @bil-ash --

The VGA output demo on the STM32 is certainly very interesting!

Due to limited time availability, I wouldn't have the opportunity to write a generic port of the STM32 display driver as you have requested.

However, if you (or someone else) did create one with an API similar to what appears in [vga.h](https://github.com/abelykh0/VGA-demo-on-bluepill/blob/master/src/vga.h), I believe it should be theoretically possible for one to also enable it to work with GUIslice. This would involve creating a simple display layer (eg. **GUIslice_drv_vga**) much like **GUIslice_drv_adagfx**. I could help give you some guidance on this step in case you wanted to try it. Note that the vga.h API is very limited, so some features like text justification would not be available.

bil-ash, 2020-11-30:
I found another embedded GUI library which is kind of abandoned but seems to have a simpler porting API as compared to GUIslice. Since I have to implement the porting myself(as you are busy), I will first have a shot at porting that(most probably next week onwards). If I can't ,I shall obviously get back to you seeking guidance for porting GUIslice. So closing this issue for now.

closed 2020-11-30 by bil-ash