ImpulseAdventure/GUIslice #302: Support for Round Displays

bsom, 2020-12-01:
### Feature Request or Enhancement
Support for round displays like this one:
http://www.icstation.com/128inch-display-module-round-240240-gc9a01-driver-interface-240x240-resolution-p-14910.html

ImpulseAdventure, 2020-12-02:
Hi @bsom —

Have you identified a suitable driver library for this round display?

For example, does this library work?
https://github.com/moononournation/Arduino_GFX

Note that the PDQgraphicstest example from that library will probably require you to select Arduino_GC9A01 instead of the default Arduino_ILI9341.

bsom, 2020-12-04:
That's the library I use and it works well.  I'm using it with a Teensy 4.1 and SWSPI.

ImpulseAdventure, 2020-12-23:
Hi @bsom --

Sorry for the delay in responding as I have been very busy with work at the moment.

I would normally be keen to add support for **Arduino_GFX**. Having a quick look through the API, it seems like the actual display driver support should be straightforward to accommodate in GUIslice. However, my impression is that the configuration could be quite a challenge to integrate. Looking at [PDQgraphicstest.ino](https://github.com/moononournation/Arduino_GFX/blob/master/examples/PDQgraphicstest/PDQgraphicstest.ino) (for example), one will notice the large number of display instantiation variants and settings that are possible. Normally, I would aim to provide a means of configuring each graphics library from within the GUIslice config file mechanism. It isn't clear to me what a good approach would be for Arduino_GFX.

Aside: For an upcoming major version of GUIslice, I have been considering the idea of enabling users to pass their own existing sketch-configured display drivers into GUIslice, rather than having GUIslice attempt to instantiate and configure them. If I successfully implemented this, adding support for libraries like Arduino_GFX (and others like it) would be much easier.

At this time I don't think I'll have time to work through this, but will leave this tagged as "open for future enhancement", in case I determine a suitable method. Another option would be for you to make a custom fork of GUIslice that adds just the round display mode (ie. **GC9A01**). I could assist in this if it's a direction you'd like to pursue, but note that it may take me some time before I can get on it due to other work. thx

closed 2020-12-23 by ImpulseAdventure