ImpulseAdventure/GUIslice #303: Simple tutorials....

bostonmacosx, 2020-12-01:
### Feature Request or Enhancement

I'm wondering if there are intro tutorials to someone without C experience. I've created the interface but doing something as easy as changing text for a clock alludes me.I'm coming from a phthon/PGP/Swift/Kotlin......

### Additional context

Been trying to do simplle text updates to a default text area E_TEXT_1 and no idea where to start...

Pconti31, 2020-12-01:
@bostonmacosx You could begin by reading the FAQ.pdf available at:
[Builder Releases](https://github.com/ImpulseAdventure/GUIslice-Builder/releases)
Then looking over the examples, ex04_bld_ctrls does what you want.
Paul--

DonpK, 2020-12-01:
There is a good amount of documentation for GUISlice/Builder although it tends to be spread around on various platforms. 

- For starters, here is a [Builder Walkthrough](https://github.com/ImpulseAdventure/GUIslice/wiki/Builder-_-Walkthrough) with examples. 
- The Builder User Guide on [this page](https://github.com/ImpulseAdventure/GUIslice-Builder/releases) is quite detailed with several code examples.
- There's a GUISlice User Guide on[ this page](https://github.com/ImpulseAdventure/GUIslice/wiki/User-Guide). 
- There are several documentation links on[ this page](https://github.com/ImpulseAdventure/GUIslice/tree/v0.16.0/docs). The GUISlice API  documentation is included. It's extremely detailed. There is an online version which might be somewhat more up-to-date, but I tend to prefer the pdf version because it has a much better search capability.

Hope this helps,
  Don

bostonmacosx, 2020-12-20:
thanks for all the info

closed 2020-12-20 by bostonmacosx