ImpulseAdventure/GUIslice #304: Unable to update Text on Simple layout....

bostonmacosx, 2020-12-01:
### Describe the bug
I'm unable to update the text in a simple text box on screen. I've been trying to follow other code but just not able to get it to change. 

### Device hardware
- *MCU*: ?ESP32
- *Display with link*: https://www.amazon.com/gp/product/B07KPD4DHD/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
- *GUIslice config selected*: `  #include "../configs/esp-tftespi-default-notouch.h"

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:*  Turns on and displays layout...can't make changes to the text....

- *Touch works correctly with standalone touch library examples without GUIslice:* Touch is not hookked up at this time..

### Expected behavior
I expect text to change when I update it...
### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.* WHERE? I don't see iit in the config file.....

### Additional info
Creation code....
`pElemRef = gslc_ElemCreateTxt(&m_gui,E_ELEM_TEXT1,E_PG_MAIN,(gslc_tsRect){3,8,132,26},
    (char*)"9:12:12",0,E_FREESANSBOLD18);
   
  gslc_ElemSetTxtMargin(&m_gui,pElemRef,10);
  gslc_ElemSetCol(&m_gui,pElemRef,((gslc_tsColor){150,150,150}),GSLC_COL_BLACK,GSLC_COL_BLACK);
  gslc_ElemSetTxtEnc(&m_gui,pElemRef,GSLC_TXT_ENC_UTF8);
`
ESSP32 loop code..

`void loop()
{
  
  // ------------------------------------------------
  // Update GUI Elements
  // ------------------------------------------------
  char mytime[] = "00:00:00";
  //char carray[8] = "       ";
  delay(4000);
  gslc_tsElemRef*  pElem  = gslc_PageFindElemById(&m_gui,E_PG_MAIN,E_ELEM_TEXT1);
  gslc_ElemSetTxtStr(&m_gui, pElem, "0");
   
  delay(6000);
  // ------------------------------------------------
  // Periodically call GUIslice update function
  // ------------------------------------------------
  gslc_Update(&m_gui);
    
}`

Pconti31, 2020-12-01:
@bostonmacosx You need to begin by assigning storage to the text field you want to modify at runtime.
Otherwise you are trying to store a new string into the flash const.
Inside the builder for E_ELEM_TEXT1 set the property "External Storage Size
" equal to the largest string you want to use.  The Builder will assign one more than youy ask for to account for the needed trailing null character '\0'.  You should use the very lastest builder 0.16.b002 since I just fixed a sizing bug in this area.
Paul--

ImpulseAdventure, 2020-12-08:
@bostonmacosx — I will presume that the above setting works for you. However, if you still encounter a problem, please feel free to update this issue and I will reopen it for you. Thanks

closed 2020-12-08 by ImpulseAdventure