ImpulseAdventure/GUIslice #305: Flash/Blink Text

mcarosh, 2020-12-03:
I am really sorry for asking this newbie question :/ 
but is there a command for flashing/blinking Text on the LCD or do I need to do the "millis"-workaround?
I ´ve searched the examples for solution but had no luck.

Regards 
mcarosh

Pconti31, 2020-12-04:
@mcarosh IMHO timers are always involved in any animation even if hidden in layers of software.  There are I suppose many ways to do this from using timer libraries that are based upon interrupts to just checking against current time.  I put together a simple demo just using miilis();  I hope it helps.
Paul--
[Flashing-Text.zip](https://github.com/ImpulseAdventure/GUIslice/files/5643425/Flashing-Text.zip)

ASMotionLab, 2020-12-04:
There are many ways that people like doing timers eg - blocking delay(), there's vdelay on esp32 if you run different code on separate cores it's non blocking, there's what Paul suggested which is the best and most 'traditional' way in Arduino, but my absolute favourite is the TimerObject.h library https://github.com/aron-bordin/ArduinoTimerObject.
You define a delay as an 'object' and operate on the object by pausing it, resuming it, restarting it, changint it's interval, etc. And when it elapses, you have a 'callback' function which is run when the timer finishes.
I think it's the most human-readable way to do timers in Arduino, personally!

mcarosh, 2020-12-04:
A big thanks @ Pconti31 and ASMotionLab. 
This solved my Blink-problem ;) I was just a little bit afraid of fighting with millis and in the end there exist already a command in GUIslice. 
Thanks again for your efforts and support.
My project is nearly finished :) If it´s done I´ll let you know ;)
best regards
mcarosh

closed 2020-12-04 by mcarosh