ImpulseAdventure/GUIslice #307: Unable to write to SD (ESP32)

MickTheMechanic, 2020-12-06:
First of all, sorry for even posting here, this is a real niche, there arent alot of people with experience using this library over on the Arduino Forums.

Here goes...

## Unable to write to SD card when using the generated wire frame GUI
I have tried various methods, including simply copying and pasting simple SD card examples that work on their own.
I can get the SD card to initialize, but I cant get it to write.

##Hardware
ESP32 Uno
2.8 TFT display with intergrated SD slot
32gb sd card formatted on FAT32

##Checklist
I can read and write when using an example by itself
When I copy an example over into my wire frame I get the initialization successful message via serial monitor, afterwards I always get the unable to write to xyz file or so on.
I also use the SD card to store my Icons, So I can confirm SD card and communication are working.

##Other info

I have a feeling it may be caused by the icons being read from the SD card, there is however no way I could find to do a non blocking write to SD on the ESP32 (SD lib for ESP32 does not support availableForWrite(); )

I have also attached a .zip with my current project.
Please note, the sd card functions ect. are quite scattered, I was having no luck so i just started going nuts trying everything.
[tiretool.zip](https://github.com/ImpulseAdventure/GUIslice/files/5649525/tiretool.zip)

Pconti31, 2020-12-06:
@MickTheMechanic You haven't included the bmp files so I can't run a test.  Besides it too messed up to follow the logic.  I strongly suggest you try and put together a simple example that reproduces your problem using the fewest UI elements with your write to SD card and include just one img button with any bmp files inside a sub folder named res. Included everything including the *.prj file for this smaller sample in your zip.
Paul--

MickTheMechanic, 2020-12-06:
Here is my project with all bmp images and accosiated files.

Attached is also a very basic example;
-There is an image button (the default "x" supplied with GUISlice(exit_n24)) and a text field displaying 12345678
-If the button is pushed, the contents of the text field should be saved to the SD card in a file named datalog.txt.
[SD_TEST.zip](https://github.com/ImpulseAdventure/GUIslice/files/5649684/SD_TEST.zip)
[tiretool.zip](https://github.com/ImpulseAdventure/GUIslice/files/5649685/tiretool.zip)

ImpulseAdventure, 2020-12-07:
Hi @MickTheMechanic --

I haven't taken a good look through your "tiretool" project yet, but looking at the SD_TEST project, I see what looks like a problem with the way that you are reading the text values.

It appears that you might be trying to read the "Val_1" field at the top of the program, where we are initializing all the variables:
```c++
const int chipSelect = 10;
String dataString = gslc_ElemGetTxtStr(&m_gui, Val_1);
File dataFile;
```

At this point in the program, GUIslice hasn't even initialized yet, so nothing can be read from the "Val_1" field. Instead, you should move the read down into the part of the program where you are triggering the SD card write when pressing the button. For example:

Change the instantiation of dataString to:
```c++
String dataString;
```

Then move the read of the "Val_1" to here:
```c++
    switch (pElem->nId) {
      //<Button Enums !Start!>
      case SAVE_BTN:
        dataString = gslc_ElemGetTxtStr(&m_gui, Val_1);
        dataFile = SD.open("datalog.txt", FILE_WRITE);
```

Note that I would strongly advise against using "String" variables whenever possible. You will likely have fewer problems in the long run if you use character arrays instead of "String" variables.

Hope that helps!
Cal

Pconti31, 2020-12-07:
@MickTheMechanic 
@ImpulseAdventure Attached is a working copy of your SD_Test project.
You had a couple of problems here.
The one Calvin pointed out.  
Doing the SD.begin() when its already done inside GUIslice to load the BMPs.
And finally not looking closely at what I output in the Builder for loading a BMP
```
//<Resources !Start!>
#define IMG_EXIT_N24              "/exit_n24.bmp"
```
Note the '/'.
Changing:
```
//        dataFile = SD.open("datalog.txt", FILE_WRITE);
        dataFile = SD.open("/datalog.txt", FILE_WRITE); // Paul: You need a path to your file
        if (dataFile) {
```
And Yes I know no examples show using a full path but I have rarely gotten SD cards to work without a path added.   One other point.  You could also put your BMP files in the SPIFFS file system.  Just a thought.
Paul--

Pconti31, 2020-12-07:
Oops! Forgot to attach the repaired project :(
Paul--
[SD_TEST.zip](https://github.com/ImpulseAdventure/GUIslice/files/5653353/SD_TEST.zip)

MickTheMechanic, 2020-12-09:
Thanks, I currently have this part of my data logger working. 
I am now unable to recall the .txt.
I would like to use  gslc_ElemSetTxtStr () to write the values in the txt file back to some GUI elements.

In my project I am using 12 elements to temporarily store 12 values, that are then written to the SD card:

 (thanks for the help with that! 💯  )

`int counter (-1);
char   logTemp[5];
char   nullVal[5];
char   logName[MAX_STR + 1];
static char nullnullval[7] = "---.--";

bool CbBtnCommon(void* pvGui, void *pvElemRef, gslc_teTouch eTouch, int16_t nX, int16_t nY)
{
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui, pElemRef);
  gslc_tsElemRef* textfield[] = {
    DATA_TEXT_1,
    DATA_TEXT_2,
    DATA_TEXT_3,
    DATA_TEXT_4,
    DATA_TEXT_5,
    DATA_TEXT_6,
    DATA_TEXT_7,
    DATA_TEXT_8,
    DATA_TEXT_9,
    DATA_TEXT_10,
    DATA_TEXT_11,
    DATA_TEXT_12,
  };
  TS_NTC_103 sensor (0.11, 10);
  float Temp (sensor.getTemp(analogRead(36)));
  float readLog;
  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
      //<Button Enums !Start!>
case NEXT_DATA_BTN:
        counter++;
        if (counter > 11) {
          counter = 0;
        }
        snprintf(logTemp, MAX_STR, "%.2f", Temp);
        gslc_ElemSetTxtStr(&m_gui, (textfield[counter]) , logTemp);
        break;
      case PREV_DATA_BTN:
        if (counter >= 0)
        {
          snprintf(nullVal, MAX_STR, "%.2f", nullnullval);
          gslc_ElemSetTxtStr(&m_gui, (textfield[counter]) , nullnullval);
          counter --;
        }
        break;
      case RESET_DATA_BTN:
        gslc_SetPageCur(&m_gui, Log_Delete);
        break;
      case SAVE_LOG_BTN:
        deleteFile(SD, logName);
        writeFile (SD, logName, gslc_ElemGetTxtStr(&m_gui, (textfield[0])));
        appendFile(SD, logName, "\n");
        for (int i = 1; i < 12; i++) {
          appendFile(SD, logName, gslc_ElemGetTxtStr(&m_gui, (textfield[i])));
          Serial.print (gslc_ElemGetTxtStr(&m_gui, (textfield[i])));
          appendFile(SD, logName, "\n");
          delay(20);
        }
        Serial.print ("file saved");
        break;
//<Button Enums !End!>
      default:
        break;
    }
  }
  return true;
}

void writeFile(fs::FS &fs, const char * path, const char * message) {

  File file = fs.open(path, FILE_WRITE);
  file.print(message);
  file.close();
}
void appendFile(fs::FS &fs, const char * path, const char * message) {

  File file = fs.open(path, FILE_APPEND);
  file.print(message);
  file.close();
}
void deleteFile(fs::FS &fs, const char * path) {

  fs.remove(path);
}
void readFile(fs::FS &fs, const char * path) {
  Serial.printf("Reading file: %s\n", path);

  File file = fs.open(path);
  if (!file) {
    Serial.println("Failed to open file for reading");
    return;
  }`

Everything is working well, I have an output of 12 lines everytime with the values that are stored in the elements.

Unfortunately the problem arises when trying to take the values from the .txt file and re-apply them to the elements.
Here is my code so far:

`char   logRead[MAX_STR + 1];

//...//
      case OPEN_LOG_BTN:
        for (int i = 0; i < 12; i++) {
          gslc_ElemSetTxtStr(&m_gui, (textfield[i]) , logRead);
        }
        break;
//...//

void readFileLine(const char * path, const int lineNum, String list) {
  File file = SD.open(path);
  int  RecNum = 0;
  while (file.available()) {
    list = file.readStringUntil('\n');
    Serial.println (list);
    if (RecNum = lineNum) {
      for (int i = 0; i < 12; i++) {
        if (RecNum == (i)) {
          Serial.println ("Done");
          file.close();
        }
      }
    }
    RecNum++;
  }
}`

No matter what I try I alway just end up with the text elemets being blank.

MickTheMechanic, 2020-12-09:
Here is a simple project containing the bug

[SDReadTest.zip](https://github.com/ImpulseAdventure/GUIslice/files/5667874/SDReadTest.zip)

Pconti31, 2020-12-09:
@MickTheMechanic I'm sorry but I can't run your program because you haven't posted the testLog.txt file.
Maybe you can explain the following code:
```
      for (int i = 0; i < numberLogLines; i++) {
          readFileLine("/testLog.txt", i, openLog);
          gslc_ElemSetTxtStr(&m_gui, (textfield[i]) , openLog);
        }
```
which is opening the same file 3 times in a row inside readFileLine() which then reads the line 1 the first time, lines 1 and 2 the second time, and lines 1,2, and 3 the third time.  Not a very good way to program.
If you really need this maybe you should google the seek() command and make fixed size records or modify readFileLine() to read all lines and do the text assignments as they are read.

When you exit readFileLine() you are assigning openLog to each text field which by the way is never set and most likely has garbage.   If you cleanup the example code and post the data file I'll take another look.

One more point if you only have 12 data fields why are you using an SD card?
Take a look at storing your values in flash.
[ESP32 Flash Memory – Store Permanent Data (Write and Read)](https://randomnerdtutorials.com/esp32-flash-memory/)
Paul--

MickTheMechanic, 2020-12-09:
-The log file is located in the .zip, in the res folder.

-The expected behavior is the file will be read until \n, printed to the first element in the array, then read untill the next \n, print to the next element, and so on.
Im sorry, thats the best I can do, I also have a thread on the Arduino forum, and they suggested I do it like this. If its terrible as you say, could you help me improve it?

-I dont think seek() will work, because the values can be anywhere between -40.00 to 150.00. So it wouldnt be a fixed number of characters, it will vary from 4-6 characters.

-I am using the SD card so that log files can later transferred to PC, added to excell, and so on. So basically just for removeable storage.

MickTheMechanic, 2020-12-09:
Figured it out.
I got rid of all that garbage and wrote this:

```case E_ELEM_BTN1: {
          File file = SD.open("/testLog.txt");
          int counter = 0;
          while (file.available()) {
            for (int i = 0; i < numberLogLines; i++) {
              counter ++;
              String list = file.readStringUntil('\n');
              const char* listConv = list.c_str();
              Serial.println (list);
              gslc_ElemSetTxtStr(&m_gui, (textfield[i]) , listConv);
            }
            if (counter == numberLogLines) {
              Serial.println ("Done");
              file.close();
            }
          }
        }
        break;```

It appears to be working now. Does this seem like a viable option to you?

MickTheMechanic, 2020-12-09:
If there are no major flaws in my code then please feel free to close the issue.
Thankyou also for your support

Pconti31, 2020-12-10:
@MickTheMechanic You're new code looks fine to me.  Good Job!
Paul--

MickTheMechanic, 2020-12-10:
@Pconti31, Thanks! And thanks for the help!
Im curious, did you see my other [issue?](https://github.com/ImpulseAdventure/GUIslice/issues/310) 
Maybe you could take a quick look? (you seem very knowledgeable, maybe you know whats going wrong there)

I will close this issue as it has been resolved. 
Thanks again for the help and support, and most of all the patience!

closed 2020-12-10 by MickTheMechanic