ImpulseAdventure/GUIslice #310: gslc_DrvDrawFillRect() not working

MickTheMechanic, 2020-12-09:
### Unable to compile code when using gslc_DrvDrawFillRect() in conjunction with AMG8833 lib
*I am trying to set up a page to display information from an AMG8833 8x8 IR array. Because I am using GUISlice, I am unable to use the ```Tft.fillRect()``` function, so I have replaced it with ```gslc_DrvDrawFillRect()``` , which should function in the same way. Unfortunately code does not compile. Error message is as follows:* ``` \Documents\Arduino\IRCAM\IRCAM.ino: In function 'void loop()':
IRCAM:189:55: error: could not convert '((double)(int)displayPixelHeight * floor((double)(i / 8)))' from 'double' to 'gslc_tsRect'
       gslc_DrvDrawFillRect(&m_gui, displayPixelHeight * floor(i / 8), displayPixelWidth * (i % 8),
```

### Device hardware
- *MCU*: ESP32
- *Display with link* 2.8" MCUFriends copy
- *GUIslice config selected*: ??? `esp-tftespi-default-simple.h"` (using DRV_DISP_ADAGFX driver)

### Expected behavior
*I should be able to draw and color a rectangle inside the GUISlice in a similar manner to how its done using other display libraries, or how its done in the Adafruit AMG8833 example,*

### Additional info
*[Adafruit AMG8833 library](https://github.com/adafruit/Adafruit_AMG88xx/blob/master/examples/thermal_cam/thermal_cam.ino) *

[IRCAM.zip](https://github.com/ImpulseAdventure/GUIslice/files/5668719/IRCAM.zip)

Pconti31, 2020-12-10:
@MickTheMechanic This is getting to be a full time job and its not even my repository!
I can't test this but give this a try
```
      // create our rectangle position
      gslc_tsRect rCamRect = {displayPixelHeight * floor(i / 8), 
                           displayPixelWidth * (i % 8),
                           displayPixelHeight,
                           displayPixelWidth};
      // now convert the color to something GUIslice will recognize
      uint16_t     nCamColor_rgb565 = camColors[colorIndex];
      gslc_tsColor nCamCol_rgb888;
      nCamCol_rgb888.r = (nCamColor_rgb565  )>> 11  ; 
      nCamCol_rgb888.g = (nCamColor_rgb565 & 0x7e0) >> 5  ;
      nCamCol_rgb888.b = (nCamColor_rgb565 & 0x1F);
      //draw the pixels!
//      gslc_DrvDrawFillRect(&m_gui, displayPixelHeight * floor(i / 8), displayPixelWidth * (i % 8),
//                           displayPixelHeight, displayPixelWidth, camColors[colorIndex]);
      gslc_DrvDrawFillRect(&m_gui, rCamRect, nCamCol_rgb888);
    }
```
Paul--

MickTheMechanic, 2020-12-10:
You sir, are a true gentleman. Thankyou very much.
Two small issues arise using this code.
-The colored rectangle remains on screen when I return to the home screen. (How can i flush the rectangle?)
-The colors are very dull (something with the conversion?)

Pconti31, 2020-12-10:
@MickTheMechanic Hmmm, I'm in my seventies so my memory isn't what it used to be.
The formula I used I think is for 32 bit color r,g,b try this instead:
```
      // now convert the color to something GUIslice will recognize
      uint16_t     nCamColor_rgb565 = camColors[colorIndex];
      gslc_tsColor nCamCol_rgb888;
      gslc_tsColor nCamCol_rgb;
      nCamCol_rgb.r = ((nCamColor_rgb565 >> 11) & 0x1F);
      nCamCol_rgb.g = ((nCamColor_rgb565 >> 5) & 0x3F);
      nCamCol_rgb.b = (nCamColor_rgb565 & 0x1F);

      //draw the pixels!
      gslc_DrvDrawFillRect(&m_gui, rCamRect, nCamCol_rgb);
```
As to why you are still have the drawing I suspect its a screen optimized drawing only outputing the defined size of your home page.  Try defining a BOX element the size of your screen using your background color.
Drag it up to the top of your home page elements using the TreeView so your other elements on your home page show up.  If that doesn't solve the redraw then you will need @ImpulseAdventure Calvin's help.
Paul--

MickTheMechanic, 2020-12-10:
Both solutions work well, thanks alot.

If you're up for it I have another problem I cant figure out (last one, i swear!)

Im trying to interpolate the IR array now, but something is going wrong. The code compiles, but im getting some error messages in the serial monitor.

[Here is the example for interpolation ](https://github.com/adafruit/Adafruit_AMG88xx/blob/master/examples/thermal_cam_interpolate/thermal_cam_interpolate.ino)

here is the error message I get:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f0c0a  PS      : 0x00060730  A0      : 0x800d20fc  A1      : 0x3ffb15e0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb1650  A5      : 0x3ffb1652  
A6      : 0x3ffb165c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000005  A12     : 0x00000000  A13     : 0x3ffc0068  
A14     : 0x00000000  A15     : 0x00000030  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400f0c0a:0x3ffb15e0 0x400d20f9:0x3ffb1600 0x400d22ca:0x3ffb1630 0x400d118f:0x3ffb1690 0x400d67dd:0x3ffb1fb0 0x40086275:0x3ffb1fd0

```

Here is the output from the exception decoder:

```
PC: 0x400f0c0a: gslc_DrvSetClipRect(gslc_tsGui*, gslc_tsRect*) at C:\Users\MickW\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp line 828
EXCVADDR: 0x00000004

Decoding stack results
0x400f0c0a: gslc_DrvSetClipRect(gslc_tsGui*, gslc_tsRect*) at C:\Users\MickW\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp line 828
0x400d20f9: gslc_PageRedrawGo at C:\Users\MickW\Documents\Arduino\libraries\GUIslice\src\GUIslice.c line 4673
0x400d22ca: gslc_Update at C:\Users\MickW\Documents\Arduino\libraries\GUIslice\src\GUIslice.c line 706
0x400d118f: loop() at C:\Users\MickW\Documents\Arduino\IRCAM/IRCAM.ino line 203
0x400d67dd: loopTask(void*) at C:\Users\MickW\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 23
0x40086275: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

If i add some serial prints I can see that the code doesnt make it past gslc_DrvDrawFillRect(&m_gui,rCamRect,nCamCol_rgb888);

here is a zip with the project.

I really do appreciate the help.

[IRCAM.zip](https://github.com/ImpulseAdventure/GUIslice/files/5675660/IRCAM.zip)

Pconti31, 2020-12-10:
@MickTheMechanic Not really sure what you want to do here.  It looks like you are trying to reduce the image from 8x8 to 6x6.   I'm not sure if your algorithm is actually doing this.  When I have done this in the past  I use a well known Color Quantization algorithm like Nearest Neighbor.  In anycase, i suspect your Rect is off.  maybe larger than screen or negative coordinates.  This is what you should trace.
Paul--

MickTheMechanic, 2020-12-10:
I was just trying to throw something together using the avaliable code from the adafruit library.
I will tak a look at the rect, maybe I can find out whats going on.
Do you think something like Nearest Neighbor would be eay to implement?

Pconti31, 2020-12-11:
@MickTheMechanic Easy or Hard is subjective.  Depends upon your background.
Again, I'm not really sure what you want to accomplish.  If it is really an image reduction then maybe this link will help you out.
[bitmap resize algorithm](http://www.davdata.nl/math/bmresize.html)
Paul--

MickTheMechanic, 2020-12-11:
I am trying to interpolate to a higher resolution (from 8x8 to 16x16 or 32x32 for example).
I am unsure what resolution this example is aiming for, There was no mention in the documentation in the repo. Perhaps it is 6x resolution (48x48)?

Pconti31, 2020-12-11:
@MickTheMechanic Well, I think we are way off topic for this issue list so this will be my last set of comments.
The link I gave you was a "simple" way to reduce an image in size.  Now I see you instead really wanted to increase the size.   

The algorithms that I'm aware of include:
- Nearest Neighbor easiest to understand but can give a ragged image
- Blinear which maybe your copied code is trying to do
- Bicube which can be tuned for sharper or smoother images

There are many others.   I would suggest asking on other forums and using google to explore your options.
This source will give you more information than you likely will want to know.

[Image Processing in C](https://homepages.inf.ed.ac.uk/rbf/BOOKS/PHILLIPS/cips2ed.pdf)

You should also set your expectations low.   Taken a high quality image with thousands or even millions of pixels and reducing or expanding the image works great.  Starting with 8x8 doesn't give you much real information and will likely lead to pixelation quite early.   Think checkerboard with different colors for each square.

You could also save your image to an SD card as a BMP file and use one of the many tools already available online to rework your images.  I mention this since you already have code in place to create the rectangle image with pixels colors already.  You mostly need to write the header.  see this link for C code examples:

[Writing BMP image in pure c/c++ without other libraries](https://stackoverflow.com/questions/2654480/writing-bmp-image-in-pure-c-c-without-other-libraries/2654860#2654860)

Paul--

MickTheMechanic, 2020-12-11:
No problem, thankyou for your help. 

I will take a deeper look into interpolation.

It’s been a pleasure, keep up the good work.

closed 2020-12-11 by MickTheMechanic