ImpulseAdventure/GUIslice #311: buttons on page 3 not initiating callback routines.

rvxfahim, 2020-12-11:
### Describe the bug
There are 3 buttons for changing pages. E_Elem_BTN6 is in main page which takes to E_PG2, E_Elem_BTN7 is in page 2 which takes to E_PG3. Here is the **buggy** button E_ELEM_BTN12 which is in E_PG3 that doesn't work. It is supposed to take to E_PG_MAIN. Infact no other buttons on page 3 initiate the callback routines.

### Device hardware
- *MCU*:  Arduino Mega 2560
- *Display with link*: Some chinese version of 2.4inch tft
- *GUIslice config selected*: mcu friend 4 wire

[Project_file.zip](https://github.com/ImpulseAdventure/GUIslice/files/5680034/Project_file.zip)

Pconti31, 2020-12-11:
@rvxfahim Sorry, but your program has too many external dependences to allow me to run it for debugging.

Furthermore nothing you posted matches.  The project file sequence_controller_2.prj only has two pages not three, the sequence_controller_2.ino file was created with 0.16.b002 and the sequence_controller_2_gslc.h file doesn't exit but you included a sequence_controller_2_#21_gslc.h file created with 0.15.b005. 

 Unless you can post a smaller version that matches and with no external dependences you are on your own. 
 
I suggest you make use of the Arduino IDE serial monitor and start by editing your config file and making sure you have 
```
  #define DEBUG_ERR               2   // 1,2 to enable, 0 to disable
```
and turn on deeper debug logging by uncommenting:
```
#define DBG_LOG           // Enable debugging log output
```
You can add tracing to GUIslice.c if you still need to go deeper.
Paul--

Pconti31, 2020-12-11:
@rvxfahim By the way, I think the easiest way to get a sample would be to backup your files, delete the *.ino and *_Gslc.h files and have the builder recreate the GUI.  The page jumping should still work and you if it does you can add your code back in a little at a time.
Paul--

rvxfahim, 2020-12-11:
Sorry for the wrong attachment, classic human error.
So I figured out that the page jump works fine if I just move the problematic case statemets BTN12,BTN13,BTN14 and BTN15 at the beginning of the button enum section rather than the end.
nothing else in the code changed, and now the page jump works. I simply shuffled cases in the case structure. The code compiled before and compiles after too.
Pardon me for any silly mistakes, I am not an expert in C/C++.
I am including only the buttom enum snippets here.

So this code works :
[works.txt](https://github.com/ImpulseAdventure/GUIslice/files/5682046/works.txt)

but this code doesn't call the button case statements:
[doesn't work.txt](https://github.com/ImpulseAdventure/GUIslice/files/5682047/doesn.t.work.txt)

both compiles fine.

without any code & not doing the shuffling of case statements and using only the generated code from guislice also seems to work.
Am I missing anything?

Pconti31, 2020-12-12:
@rvxfahim While not strictly wrong E_ELEM_BTN11 having two break statements could be confusing the gcc code generation.  Try moviing the code to a subroutine.  Which is a good practice for any lengthy bit of code inside a case statement or if statement for that matter. 
```
      case E_ELEM_BTN7:
        gslc_SetPageCur(&m_gui, E_PG3);
        break;
      case E_ELEM_BTN11:
        processRelay();
        break;
//<Button Enums !End!>
  default:
        break;
    }
  }
void processRelay() 
{
    gslc_ElemXTextboxReset(&m_gui,m_pElemTextbox1);
  //concat_relay="\0";
  skip_relay = gslc_ElemXSpinnerGetCounter(&m_gui, &m_sXSpinner5);
etc...
}
```
Paul--

rvxfahim, 2020-12-12:
Thank you very much for the advice.
Since this isn't an issue with the library or GUISlice, I am closing it.

closed 2020-12-12 by rvxfahim