ImpulseAdventure/GUIslice #312: Support for png

awareness2015, 2020-12-15:
### Feature Request or Enhancement
Is it possible to add support for png to the library?

ImpulseAdventure, 2021-01-04:
Hi — sorry for the delay as I although I had written a reply a while back it looks like I must have missed submitting it.

Unfortunately PNG decoding can be fairly involved and usually involves maintaining a sizable memory buffer that makes it unrealistic for many “Arduino” platforms.

If possible, I suggest converting your images from PNG to BMP as this will minimize the memory overhead and also speed up the rendering considerably.

I will close this for now but mark for future enhancement. Hopefully the offline BMP conversion will address your needs.

thanks

closed 2021-01-04 by ImpulseAdventure

awareness2015, 2021-02-04:
Hi, GUIslice is very excellent.Thank you very much!