ImpulseAdventure/GUIslice #313: Radio button states get cleared automatically.

rvxfahim, 2020-12-15:
There are 4 radiobuttons, two on page 1 and the other two are on page 3.
If I set the state by gslc_ElemXCheckboxSetState of any radio button of the front page, then the state of other two radio buttons on page 3 will be cleared. Same thing happens if I set the state of any of the radio buttons on page 3, then radio buttons of page 1 gets cleared. 

there are no other codes in the loop except gslc_ElemXCheckboxSetState in the case statements

if ( eTouch == GSLC_TOUCH_UP_IN ) {
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BTN1:
        break;
      case E_ELEM_BTN4:
        gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio1, true);
        gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio2, false);
        break;
      case E_ELEM_BTN5:
      gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio1, false);
        gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio2, true);
        break;
      case E_ELEM_BTN6:
        gslc_SetPageCur(&m_gui, E_PG2);
        break;
      case E_ELEM_BTN7:
        gslc_SetPageCur(&m_gui, E_PG3);
        break;
      case E_ELEM_BTN11:
        break;
      case E_ELEM_BTN12:
        gslc_SetPageCur(&m_gui, E_PG_MAIN);
        break;
      case E_ELEM_BTN13:
      gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio3, true);
      gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio4, false);
        break;
      case E_ELEM_BTN14:
      gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio3, false);
      gslc_ElemXCheckboxSetState(&m_gui, m_ElemRadio4, true);
        break;
      case E_ELEM_BTN15:
        break;

//<Button Enums !End!>
      default:
        break;
    }

https://www.youtube.com/watch?v=Y3kJJn8_MHo

Pconti31, 2020-12-15:
@rvxfahim Please post a zip file of your complete sample project *.prj, *.ino, *.h, if you want us to look at it.
Paul--

rvxfahim, 2020-12-15:
I hope I am not attaching wrong files again 
[radio_button_bug_project.zip](https://github.com/ImpulseAdventure/GUIslice/files/5696881/radio_button_bug_project.zip)

Pconti31, 2020-12-15:
@rvxfahim I think you have a misunderstanding of the radio buttons.  

You don't need to manage them yourself.

Simply group them together and only one will be set and the other one will automatically be turned off.

I modified your sample and attached it.  
- What I did was first to comment out your code inside the button calllback.  
- Then Inside the builder I selected both your main page radio buttons using mouse right click while holding down control key.  
- Then used the Group command icon that is way over on the right hand-side over the Misc segment.  This asssigned them to group1. 
-  I also enabled callback = true on these two buttons.  
- Did the same on page 3 with the two other radio buttons this then was assigned automatically to group2.  
- I then generated the code and edited the radio button callbacks to simply Serial.println their status.

I admit Calvin hasn't made this easy by making the callbacks different than other buttons and furthmore requiring users check nSelId for radio buttons instead of pElem->nId like the checkboxes but its what we got.

Let me know if this is a satisfactory solution for you or not.
Paul--

[sequence_controller_2.zip](https://github.com/ImpulseAdventure/GUIslice/files/5697631/sequence_controller_2.zip)

rvxfahim, 2020-12-15:
Well I never intended the radiobox to be user interactable. I am actually trying to simulate an LED with green and red checked marked icons of the radio button. The buttons Stop, Start, On and OFF dictate which LED (radio button)  will be lit.
Should I just avoid the radio buttons and simply replace them with four "Box" elements and set their color green and red ?
If so is there any command like "GSLC set box element 1 fill color" ?

Pconti31, 2020-12-15:
@rvxfahim Yes, Box would be the way to go.
you can use gslc_ElemSetCol() and simply set the glow to your fill color.
```
///
/// Update the common color selection for an Element
///
/// \param[in]  pGui:        Pointer to GUI
/// \param[in]  pElemRef:    Pointer to Element reference
/// \param[in]  colFrame:    Color for the frame
/// \param[in]  colFill:     Color for the fill
/// \param[in]  colFillGlow: Color for the fill when glowing
///
/// \return none
///
void gslc_ElemSetCol(gslc_tsGui* pGui,gslc_tsElemRef* pElemRef,gslc_tsColor 
colFrame,gslc_tsColor colFill,gslc_tsColor colFillGlow);
```

Paul--

rvxfahim, 2020-12-16:
Thanks Paul, everything is solved now.
My project is complete finally. Very happy to use GUISlice.

closed 2020-12-16 by rvxfahim