ImpulseAdventure/GUIslice #315: New Utilities for GUIslice users

Pconti31, 2020-12-17:
I created a couple of Utilities that users of GUIslice API might be interested in.

GUIslice_Image2c
The GUIslice image2c is a standalone Java desktop application that is designed to convert images to C Array memory based images for GUIslice.

Input images can be in BMP, PNG, JPG, or GIF formats.

Note that the utility can also convert various formats to BMP files for use with SD Cards and GUIslice.

The generated output code (.c) will contain the imported image as a C Array that can be included in your projects. 

The Repository is [GUIslice_Image2C](https://github.com/Pconti31/GUIslice_Image2C)

TTF2GFX

Release 0.16.b004 of GUIsliceBuilder now supports a new Dialog called CharacterMap.  By right clicking on
a text property field you can easily add non ascii characters to your projects.

TTF2GFX is a standalone Java desktop application to allow conversion of TrueType fonts to Adafruit's GFX Font format.

The main advantage of the ttf2gfx utility is that it supports a simple way for users to pluck individual glyphs out of a larger font and assign them slots between 0 and 255.

The Repository is [TTF2GFX](https://github.com/Pconti31/TTF2GFX)

Paul--

ImpulseAdventure, 2020-12-23:
@Pconti31 -- these are some very useful utilities!

I am thinking we should update the wiki / Builder user guide to reference these tools (if not already).
In particular, when users choose "image from memory", the GUIslice_Image2C should avoid the hand-editing that may be required currently.

Where do you suggest we may require documentation updates for this?

thanks again for the initiative on this!
cal

Pconti31, 2020-12-23:
@ImpulseAdventure 
For GUIslice_Image2C you might create a pointer in your wiki page:
[Image in Flash](https://github.com/ImpulseAdventure/GUIslice/wiki/Display-Images-from-FLASH)
I also can't find a wiki link to this page except from your examples.  Maybe I just misssed it.

You could mention the TTF2GFX utility could be list in your wiki tips & tricks under font handling:
[tips and tricks](https://github.com/ImpulseAdventure/GUIslice/wiki/Tips)

I'm open to anything else.
Paul--

closed 2021-01-04 by Pconti31

ImpulseAdventure, 2021-01-07:
For reference, I have updated the associated wiki pages to include references to the new utility repositories and created a discussion note. thanks!