ImpulseAdventure/GUIslice #319: White square appears where image is expected...

bostonmacosx, 2020-12-20:
### Describe the bug
*Please describe what you observe on the display or any error messages*

### Device hardware
-  ESP32
- *Display with link*: ILI9341 Generic 3.2"
- *GUIslice config selected*: ??? `esp-tftespi-default-notouch.h`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:*  YES
- *Touch works correctly with standalone touch library examples without GUIslice:* YYES

### Expected behavior
I expected to see the image where I placed it in the Builder window...

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.* NO ERRORS

### Additional info

So i've attempted to do the following steps but end up with a GREY square on the display where the iimage is supposed to be....

In the builder I add an image.....
I've added 12 images to the SPIFF area of my ESP32. 
I can print out the file list...
`20:51:46.945 -> File: /partnight_sm.jpg
20:51:46.945 -> File: /downpour_sm.jpg
20:51:46.945 -> File: /thunderstorm_sm.jpg
20:51:46.945 -> File: /clearnight_sm.jpg
20:51:46.945 -> File: /snow_sm.jpg
20:51:46.945 -> File: /snowshowers_sm.jpg
20:51:46.945 -> File: /Clouds_sm.jpg
20:51:46.945 -> File: /partsun_sm.jpg
20:51:46.945 -> File: /dangcold_sm.jpg
20:51:46.945 -> File: /verycloud_sm.jpg
20:51:46.945 -> File: /thunder_sm.jpg
20:51:46.981 -> File: /windy_sm.jpg
20:51:46.981 -> File: /Sunny_sm.jpg
20:51:46.981 -> File: /danghot_sm.jpg
`

I set the image to one in the file list..... uaing the builder....

`#define IMG_CLEARNIGHT_SM         "/clearnight_sm.jpg"` 

` pElemRef = gslc_ElemCreateImg(&m_gui,E_ELEM_IMAGE1,E_PG_MAIN,(gslc_tsRect){215,17,70,70},
    gslc_GetImageFromFile(IMG_CLEARNIGHT_SM,GSLC_IMGREF_FMT_JPG));
`

the above appears in the _GSLC.h file. 

I also made sure that in the GUIslice config file I changed 

`  #define GSLC_SPIFFS_EN               1 `
 in the 

esp-tftespi-default-notouch.h

Still end up with the grey square thoug.....am I missing  some stepss...

Pconti31, 2020-12-20:
@bostonmacosx You don't seem to have missed any steps.  
Please  post a zip file of complete sample project including *.prj, *.ino, *.h and the data sub folder with its image otherwise we have no idea what's going on.   Try and redo the sample with just one image for us to debug.  
Paul--

Pconti31, 2020-12-20:
@bostonmacosx I just created a new project and stored image button and image in SPIFFs and it worked fine.
When you pack up your zip file for debugging include you're TFT_eSPI/User_Setup.h and esp-tftespi-default-notouch.h
Paul--

bostonmacosx, 2020-12-20:
Thanks Paul. Here are the requested files....
[GUISLICE_GIT.zip](https://github.com/ImpulseAdventure/GUIslice/files/5721606/GUISLICE_GIT.zip)
[Otherconfigs.zip](https://github.com/ImpulseAdventure/GUIslice/files/5721607/Otherconfigs.zip)
@Pconti31

Pconti31, 2020-12-21:
@bostonmacosx Turns out you are actrually getting an error from TFT_eFEX that isn't being reported by GUIslice.
`Jpeg file format not supported!`
According to my Adobe tools you have ICC Profile: GIMP Builtin sRGB color
with format option:  progressive with 3 scan lines.

I changed format option to Baseline: Standard for clearnight_sm.jpg and it loaded fine.
Attached is the new file.
[clearnight_sm.zip](https://github.com/ImpulseAdventure/GUIslice/files/5721750/clearnight_sm.zip)

If you need me to convert the others It will have to wait until tomorrow.  You might want to report this as an issue for the JPeg converter.
Paul--

bostonmacosx, 2020-12-21:
SUPER! Should I rerport it as (should be reported in GUISlice as well)

bostonmacosx, 2020-12-21:
@Pconti31 bodmer replied will never be support for progressive images.

Pconti31, 2020-12-21:
@bostonmacosx Well, at least he answered you...
Close the issue if it works otherwise post a zip images that fails.  As to reporting to GUIslice, Ah.. isn't that where we are?   The whole idea of this issue list is so @ImpulseAdventure Calvin will notice and fix problems.

Note that the error reporting is a little unfair in that TFT_eFEX's routines are void with no error return so GUIslice thinks everything is working.   Still a blank block is a pretty good hint that something is wrong.
Paul--

ImpulseAdventure, 2020-12-23:
I have now forwarded the error return value from **TFT_eFEX** `drawBmp()` / `drawJpgFile()` / `drawJpeg()`, which should then be trapped in `ElemDrawByRef()`. If `DEBUG_ERR=2` is set in the config (full debug messaging), it might now report an error within GUIslice.

@Pconti31 -- Note that I haven't tested this specific code yet, but I was thinking the progressive image decode error would be exposed by a `false` return from drawJpgFile()?

I had the fun of writing a JPEG decoder from scratch a while back ( [JPEGsnoop](https://github.com/ImpulseAdventure/JPEGsnoop) ) and similarly figured that progressive images weren't worth the effort, especially with broadband access being pretty common now.

Pconti31, 2020-12-23:
@ImpulseAdventure A little impulsive there, Cal.  Get it! :)

Anyway, you now have a problem compiling:
```
C:\Users\Paul\Documents\Arduino\libraries\GUIslice-0.16.8\src\GUIslice_drv_tft_espi.cpp:1052:45: error: void value not ignored as it ought to be

   return fex.drawBmp(pStrFname, nDstX, nDstY);
```
With this fixed I tested spiffs with JPG's and it correctly loads non progressive jpegs.
However, no error is reported with progressive jpegs.
I would suggest something like:
```
bool gslc_DrvDrawJpegFromFile(gslc_tsGui* pGui,int16_t nDstX,int16_t nDstY,gslc_tsImgRef sImgRef)
{
  const char* pStrFname = sImgRef.pFname;

  // Load JPEG image from file system
#if defined(ESP32)
  // use optimized ESP32 native decoder  if(!fex.drawJpgFile(SPIFFS, pStrFname, nDstX, nDstY)) {
   // Unsupported source
    GSLC_DEBUG_PRINT("DBG: DrvDrawJpegFromFile() %s unsupported jpeg format\n", pStrFname);
  }
#else 
  // use library decoder
  if(!fex.drawJpeg(pStrFname, nDstX, nDstY)) {
    // Unsupported source
    GSLC_DEBUG_PRINT("DBG: DrvDrawJpegFromFile() %s unsupported jpeg format\n", pStrFname);
  }
#endif
return true;
}
```
and this successfully reported:
```
INFO:   Add elem to collection: ElemRef=6/6, ElemRAM=6/6, ElemId=4 (RAM)
DBG: DrvDrawJpegFromFile() /danghot_sm.jpg unsupported jpeg format
```
Paul--

ImpulseAdventure, 2020-12-23:
Oops, yes, maybe a bit impulsive :) I did run my changes through the regression environment and didn't encounter any errors. But I see now that the `GSLC_SPIFFS_EN` code wasn't being tested -- so I'll have to add that to my checks as well.

Looks like I was thrown off by the fact that that both `drawJpg()` and `drawJpgFile()` had return values, but not `drawBmp()` or `drawJpeg()`, despite all being able to detect an error. For the same reason, I'm guessing that your last proposed line with `drawJpeg()` might not actually operate as intended.

Thankfully, for the specific problem at hand, we can still detect the error on ESP32 platforms.

I have pushed a change for the above... hopefully this one reports the error as intended! (Though I don't have the test setup ready on hand at the moment, so have only confirmed compilation, not on physical hardware)

ImpulseAdventure, 2021-01-01:
Closing for now as I believe it has been addressed.
If there are any outstanding issues with the latest changes, please reply and I'll re-open.

thx

closed 2021-01-01 by ImpulseAdventure