ImpulseAdventure/GUIslice #322: SdFatSoftSpi usage with GSLC_SD_EN 2

rvxfahim, 2020-12-26:
### Describe the bug
I had a brief discussion about this issue over email with @ImpulseAdventure, this is regarding usage of SD library with GSLC_SD_EN 2

Current State:
The project does not utilize any SD function through guislice so GSLC SD EN is also set to 0, and in this state I can declare things like:

```
File timerfile;  
File runstatus;
```
also these are declared in the beginning:
```
#include <SdFat.h>
SdFatSoftSpi<12, 11, 13> SD;
```

In this way everything works fine and I can access txt files in SD and do stuff in the main loop.

**The Problem:**
I want to show images from SD through the guislice library/builder, so I enable GSLC_SD_EN 2 and also make this change in the SdFat library as mentioned in GUIslice_drv_adagfx.cpp   (// -     #define ENABLE_SOFTWARE_SPI_CLASS 1 // Change default from 0 to 1)
After doing these and as adviced by Calvin I **remove** the following from the beginning since they are being declared by the guislice library itself:
```
#include <SdFat.h>
SdFatSoftSpi<12, 11, 13> SD;
```
and this from void setup

```
if (!SD.begin(10)) {
    Serial.println("initialization failed!");
 return;
  }
  Serial.println("initialization done.");
```
Now if I compile, I get errors like these 

```
sequence_controller_2_SD:24: error: 'File' does not name a type

 File timerFile;

 ^~~~

sequence_controller_2_SD:25: error: 'File' does not name a type

 File runstatus;

 ^~~~

sequence_controller_2_SD:26: error: 'File' does not name a type

 File relaynumber;
```
and many more errors in other places where SD is used in the code previously.
### Device hardware
- *MCU*: ATmega2560
- *Display: using mcufriend kbv 4 wire for a 2.4inch display

ImpulseAdventure, 2020-12-27:
Hi @rvxfahim --

Could you first try downgrading **SdFat** to version 1.1.4 via the Arduino Library Manager (and again set **ENABLE_SOFTWARE_SPI_CLASS 1** in the `SdFatConfig.h`)?

Some changes were made in SdFat v2 that may not be compatible at the moment.

If that still doesn't work, could you attach your sketch for me to test?

thanks

rvxfahim, 2020-12-29:
I did a couple of testings with the new Guislice repo

1. Can't use SdFat v2 because SdFat v2 no longer has the "File" class which was code compatible with the "File" class of builtin arduino SD.h. It is now replaced with "File32". Therefore, I am not sure if replacing my existing code keywords from "File" to "File32" will work. I need some time to learn how to use the SdFat v2 library before making any changes to existing code.

2.  I was already using SdFat 1.1.4, and just to avoid any doubt I reinstalled the SdFat library and set ENABLE_SOFTWARE_SPI_CLASS 1. 
    With the new guislice library there is one problem in GUIslice_drv_adagfx.cpp. 
`if (!SD.begin(SD_CONFIG))` this is in line 748 and SD_CONFIG is defined in line 199 which is for SdFat v2. So compiling it gives error about SD_CONFIG not defined. So I replaced SD_CONFIG with 10 (SD CS Pin) in line 748 : `if (!SD.begin(10)`

At this point my main goal is to use GSLC SD EN 2 and show images from SD and also use SdFat library to read and write into some text files.

So here is the change that **made everything work**
I had to include <SdFat.h> again in the main .ino file; not redeclaring = compile error
Also this line: 
`SdFatSoftSpi<12, 11, 13> SD1;` Guislice uses SD and I am using SD1.
I also had to put this in void setup
`if (!SD1.begin(10)) {
    //Serial.println("initialization failed!");
 return;
  }`

and then I replaced all keywords in my code where "SD" was previously used to "SD1".
Now it compiles fine and I can see images pulled from SD by guislice and also I can read/write to text files.

[project file.zip](https://github.com/ImpulseAdventure/GUIslice/files/5750859/project.file.zip)

ImpulseAdventure, 2021-01-01:
Thank you @rvxfahim for the detailed testing -- it's much appreciated!

As noted, I have not spent much time testing the software SdFat mode... and the recent breaking changes to the SdFat API have added to the complexity.

Given your feedback with SdFat v2, it sounds like others users may also hold off on updating to v2. Therefore, it would be wise to ensure GUIslice still runs well with SdFat v1.

I'll take a look at your suggested changes to v1 and will also ensure that I have a test setup locally configured so that I can confirm everything works properly. Please note that I may not have an opportunity to do this for the next couple weeks due to other work. In the interim, it sounds like you have identified a viable workaround, which is good to hear.

thanks!