ImpulseAdventure/GUIslice #323: Button Glow after longpress

mcarosh, 2020-12-26:
Hi guys :)
First of all a big merry X-mas to everybody. Hope you´re doing fine during these strange days!

I have a little problem with the glow function :/
If I perform a longpress on a button and its corresponding function is switching page, everything works fine but if I switch back to the previous page, the "longpressed" button is still glowing. I´ve already tried the redraw and set glow functions, but both failed.
If I press another button the glow disappears. Is there a solution to "reset" the status of the button immediately when switching pages? 

regards
mcarosh

Pconti31, 2020-12-26:
@mcarosh Once again your best chance of getting help is to post a zip file of a smaller sample project with the minimum number of UI elements that reproduces the issue with no external dependences.
Paul--

mcarosh, 2020-12-26:
OK
Here we go
In the example glow.ino is the effect I encounter.
But I made the code exactly like the example glow2.ino.
I cannot find any difference despite my program is really big. I have 23 buttons, 52 text elements and more stuff.
In my opinion it must be the size of the program. So far everything but the glow function is working???
Maybe someone has a suggestion for this strange problem
[glow button.zip](https://github.com/ImpulseAdventure/GUIslice/files/5743880/glow.button.zip)
[glow2 button.zip](https://github.com/ImpulseAdventure/GUIslice/files/5743885/glow2.button.zip)

Pconti31, 2020-12-27:
@mcarosh You didn't include a project file so i can't fix the demo for you.
You need to go into the builder and assign a Element reference to your E_ELEM_BTN1.
say m_pElemButton then you can add this code:
```
      case E_ELEM_BTN2:
      gslc_ElemSetGlow(&m_gui,m_pElemButton,false);
      gslc_SetPageCur(&m_gui, E_PG_MAIN);
        break;
```
which worked for me using your glow button example.
Paul--

mcarosh, 2021-01-05:
Thx for your help
I managed to get it working due to your help. I have put the "gslc_ElemSetGlow" on the wrong place :/
Now I have another problem but I will open a new issue.
best regards

closed 2021-01-05 by mcarosh