ImpulseAdventure/GUIslice #324: Having Issues Updating Images after original creation...

bostonmacosx, 2020-12-27:
### Describe the bug
I don't belive this to be a bug but a lack of knowledge on my part.

### Device hardware
- *MCU*: ?ESP32
- *Display with link*: Generic ILI9341 2.8"
- *GUIslice config selected*: ??? `esp-tftespi-default-notouch.h`

### Checklist to try first

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:*  YES
- *Touch works correctly with standalone touch library examples without GUIslice:* Not a touch issue
### Expected behavior
After setting a new image reference I would be able to replace the image. 

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info

Hello there. 

This is more of a help than a bug request. 

**in my GUISLICE_GSLC.H:**

```
#define IMG_SUNNY_SM              "/Sunny_sm.jpg"

// Create E_ELEM_IMAGE2 using Image 
  pElemRef = gslc_ElemCreateImg(&m_gui,E_ELEM_IMAGE2,E_PG_MAIN,(gslc_tsRect){270,0,50,50},
    gslc_GetImageFromFile(IMG_SUNNY_SM,GSLC_IMGREF_FMT_JPG));
  gslc_ElemSetFillEn(&m_gui,pElemRef,false);
  p_Elem_weaimg = pElemRef;
```
this draw the image fine....
I'm trying to change the image though some conditions  condition

Defined in the top of the GUISLICE FILE
```
gslc_tsElemRef* p_Elem_weaimg     = NULL;

```
**I have a function:**
```
static gslc_tsImgRef setweaimg(int code,const char icon[]){
       static gslc_tsImgRef imgref = gslc_GetImageFromFile("/cloudy_sm.jpg",GSLC_IMGREF_FMT_JPG);
       return imgref;
}
```
and later in the loop

```
gslc_tsImgRef imgref2 = setweaimg(obj["weather"][0]["id"],obj["weather"][0]["icon"]);
gslc_ElemSetImage(&m_gui,p_Elem_weaimg,imgref2,imgref2);

```

bostonmacosx, 2020-12-27:
I should also mention there are a lot of keys and passwords iin the code so it wouuld be a pain to post the whole thing....Ii'm using openweathermap and I'm hopiing to submit this as an example when done.

Pconti31, 2020-12-27:
@bostonmacosx The image UI element is meant to be static.  Maybe there is a way to change it at runtime but
I have done things differently myself.  A few years back I wrote a Photo album app that just displayed phots in a loop from an SD card.  I just hacked tjhis app to the bear minimum and changed it to load JPG images from SPIFFS for you to use as an example.  
The idea is to create a BOX UI Element with a callback for drawing.  Then you simply draw in the box using your desired image.
Hope it helps.
Paul--
[slideshow.zip](https://github.com/ImpulseAdventure/GUIslice/files/5745133/slideshow.zip)

bostonmacosx, 2020-12-28:
Thanks. I was ablle to tweak the codde and my gslc_ElemCreateImg and just use gslc_ElemSetImage which successfully changes the iimage on the next update to the interface. 

The top right iimage changes once everh 30 minutes as needed and the bottom row twice a day.

[image]

closed 2020-12-31 by bostonmacosx