ImpulseAdventure/GUIslice #325: Unable to toggle imagebutton (SetGlow not working)

MickTheMechanic, 2020-12-27:
### Describe the bug
*I am unable to toggle an image button between its glowing and not glowing/active inactive states. The workaround should be quite simple (calling SetGlow each time button is pressed). However, I have been able to implement a workaround that works*

### Device hardware
- *MCU*: ??? ESP32.
- *Display with link*: ??? MCUFriends copy
- *GUIslice config selected*: ??? `Custom`

### Expected behavior
*I expect to be able use SetGlow to toggle an image buttons glow status.*

### Additional info
*A basic example where bug is present. (BMP images in res folder) It should work, and the debugging serial.prints tells me it is working, but for some reason it doesn't*
[imgagebuttontog.zip](https://github.com/ImpulseAdventure/GUIslice/files/5744557/imgagebuttontog.zip)

Pconti31, 2020-12-28:
@MickTheMechanic The problem here is that the button is doing what its designed to do.  Touch and it blinks glow.  What you want is a toggle button that keeps it state.  Well, even more than that a custom toggle button that when your state is greater than 0 to show pause.   
Ok, forget about standard buttons and implement your own using a Box element with both a draw callback to paint your desired image and a touch callback to change state by the user I guess. 
Attached is my version doing just that.

 I don't fully understand your logic so i make no promises about my implementation but it should give you the idea since it now toggles correctly.  
Paul--
[imgagebuttontog.zip](https://github.com/ImpulseAdventure/GUIslice/files/5745452/imgagebuttontog.zip)

MickTheMechanic, 2020-12-28:
Please forgive my sloppy implementations, I am still very new to this, and only a hobbyist.
The goal here is simply to simply have an image button that toggles. 

Greater than or less than 0 is irrelevant, I was just using this to try and toggle between glowing and not glowing states.

Basically all I want is an on off button that uses one image for on and another image for off. I just assumed it would be possible to do using setGlow, as the standard image-buttons have a second glow image already.

I will take a look at your code and see if I can implement it.

Thanks again.

P.S I followed your advice regarding interpolation, and got my thermal camera project working. Thought you‘d like to know.

MickTheMechanic, 2020-12-28:
Your example functions flawlessly, but after looking at the code it seems quite complicated. 
I see it seems somewhat hardcoded around greater than/less than 0, which isnt necessary, the toggle event could be triggered by the button press rather than an integer. 

I am also worried about implementation, I am building up a rather large GUI with _alot_ of pages, and such a button is needed on 3 different pages. Is this the most "bare bones" way of doing this?  If you say this is the easiest way then I will persist, but perhaps there is a simpler way. Just a very simple toggle between glowing and not glowing, triggered by the button press itsself.

Also, could I request image button toggle as a future enhancement for GUISlice?

Pconti31, 2020-12-28:
@MickTheMechanic I had hoped you would have noticed that most of the code was simply to read a BMP file. You can remove that code my simply exposing the code inside GUIslice_drv_adagfx.h.
Edit GUIslice/src/GUIslice_drv_adagfx.h file and at about line 443 add
```
void gslc_DrvDrawBmp24FromSD(gslc_tsGui* pGui,const char *filename, uint16_t x, uint16_t y);
```
and you can now call this routine directly as I have done in this two button example l've attached.
As for "I see it seems somewhat hardcoded around greater than/less than 0, " it's what you posted so that is what you get back.  At this point anything else and you are on your own.
If you want a XImgTogglebtn you can use my XTogglebtn that I summitted earlier this year as a model and when you get it working you can do as I did and ask Calvin in a pull request to add it.
Paul--
[two-imagetoggle.zip](https://github.com/ImpulseAdventure/GUIslice/files/5747696/two-imagetoggle.zip)

MickTheMechanic, 2020-12-28:
I appreciate the help. As with many things, another closer look and a bit of patience and it looks managable.

After examining your code and doing some testing I have a question about clipping, I understand it veers away from this topic somewhat, but I hope you might answer it.

I notice that normal image buttons dont seem to make use of clipping like your above example does, and as a result when a 
normal image button is pressed  part of the screen, much larger than the button itsself, is redrawn. 
Is there some kind of provision in GUISlice to enable clipping for image buttons? Or is this something that will require some kind of workaround?

Pconti31, 2020-12-28:
@MickTheMechanic Well, again this isn't my code base so the truth is that I don't know and can only guess.
I suspect what you notice with "Normal" buttons is region invalidation not clipping.

While I don't know where @ImpulseAdventure Calvin found his algorithm or if he invented, its much like Apple's old QuickDraw algorithm for screen refreshes.   However, his is much simpler and only handles rectangular regions.  Basically, inside the update loop he looks at each UI element that marked itself as needing redraw and he builds up a larger and larger rectangle that will encompass all of the UI bits that need redrawing.
Once completed he draws the rectangle (I guess using background color).   Which is what I expect you are seeing.

As for work around I'm unclear from your question if you want that or not.
You should open a new issue or use discussions part of this repro to explore further.

As a side note.  
Hopefully, Calvin will notice this issue with Toggle Image buttons and mark it as a request for enhancement.
Not having write access I can't do that.
Paul--

MickTheMechanic, 2020-12-28:
Ok, thankyou for the info. I will open a new issue soon.

Regarding the image toggle, I am unable to implement this if the button is on a page other than E_PG_MAIN. I have found no reference to page ID, so I do not understand why this is happening.

I have included an example.
[twopage-imagetoggle.zip](https://github.com/ImpulseAdventure/GUIslice/files/5748543/twopage-imagetoggle.zip)

Pconti31, 2020-12-28:
@MickTheMechanic 
First you are missing these lines for Play button in your InitGUIslice_gen() routine.
```
  // Create E_PLAYBTN  box
  pElemRef = gslc_ElemCreateBox(&m_gui,E_PLAYBTN ,E_PG2,(gslc_tsRect){130,80,40,40});
  gslc_ElemSetClickEn(&m_gui, pElemRef, true);  // missing
  gslc_ElemSetTouchFunc(&m_gui, pElemRef, &CbBtnCommon); // missing
  // Set the callback function to handle all drawing for the element
  gslc_ElemSetDrawFunc(&m_gui,pElemRef,&CbDrawScanner); // missing
```
Because in the Builder on page 2 for E_PLAYBTN you failed to set the properties:
```
Touch Enabled? = true
Draw Function = true
```
This is the magic that ties the UI Element to a callback otherwise its just a Box.

Second in your button callback you failed to have any code for play so I added:
```
      case E_PLAYBTN :
        Serial.print("Status:");
        if (Pause == false){
            Pause = true;
            Serial.println (" paused");
        }
        else {
            Pause = false;  
            Serial.println (" playing");
        }
        // Manually mark the elements affected as needing redraw
        gslc_ElemSetRedraw(&m_gui,m_pPlayBtn,GSLC_REDRAW_FULL);
        break;
```
Attached is the repaired example for two pages.
Paul--
[twopage-imagetoggle.zip](https://github.com/ImpulseAdventure/GUIslice/files/5748709/twopage-imagetoggle.zip)

MickTheMechanic, 2020-12-28:
Wow, cant believe I missed all that..
Everything is working and successfully implemented in my main project. 

Thanks again for the help.

I will leave the issue open so it can be listed as a feature request for future versions

ImpulseAdventure, 2021-01-01:
> While I don't know where @ImpulseAdventure Calvin found his algorithm or if he invented, its much like Apple's old QuickDraw algorithm for screen refreshes. However, his is much simpler and only [...]

I wrote this algorithm independently, with a mind to simplify the invalidation design to handle an unknown number of pending redraw requests (though reinventing the wheel is rarely a good idea!). However, as noted it has a downside when we have disparate regions hit at the same time. I have posted more details on some possible enhancements in #328 

> Hopefully, Calvin will notice this issue with Toggle Image buttons and mark it as a request for enhancement.

Just catching up on some outstanding threads here... could you clarify what enhancement was being requested?
- ie. exposing `gslc_DrvDrawBmp24FromSD` or integrating something like XImgTogglebtn ?

thanks!

MickTheMechanic, 2021-01-01:
I started this issue with the hopes of having an image button function the same way as a toggle button. 

Something like XImgTogglebtn would be amazing. Paul's workaround works very well, but something that can be easily implemented via the builder would be great.

ImpulseAdventure, 2021-02-16:
It should be much easier to create a toggle image button. No need to manually draw anything or use a custom callback. All one needs to do is create a normal image button, then in the callback, do the following:
```c++
      case E_PLAYBTN :
        m_bPause = !m_bPause;
        gslc_ElemSetImage(&m_gui,m_pPlayBtn,
          gslc_GetImageFromSD(m_bPause?IMG_FNAME_PAUSE:IMG_FNAME_PLAY,GSLC_IMGREF_FMT_BMP24),
          gslc_GetImageFromSD(m_bPause?IMG_FNAME_PAUSE:IMG_FNAME_PLAY,GSLC_IMGREF_FMT_BMP24));
        break;
```

Please see the attached sketch
[twopage-imagetoggle-210216.zip](https://github.com/ImpulseAdventure/GUIslice/files/5989751/twopage-imagetoggle-210216.zip)

Does address what you were looking for?

I can still create a XToggleImgBtn if that would be useful. The `ElemXToggleImgBtnCreate()` would accept 4 image parameters, two for the pair of normal states and two for the pair of glowing states. One could either register a callback (that is triggered upon state transition) or directly query its state.

ImpulseAdventure, 2021-03-11:
For reference, the following branch integrated Paul’s XToggleImgBtn and it can be used with his test version of the Builder:

https://github.com/ImpulseAdventure/GUIslice/tree/Int384-Input2d

Builder:
https://github.com/ImpulseAdventure/GUIslice/issues/348#issuecomment-774693329

I will be updating this widget to support the new focus navigation.

closed 2021-03-11 by ImpulseAdventure

MickTheMechanic, 2021-04-14:
@Pconti31  @ImpulseAdventure 

I was just looking through the  XToggleImgBtn code, am I correct in assuming you have also added radio-button functionality to XToggleImgBtn?

Pconti31, 2021-04-25:
@MickTheMechanic @ImpulseAdventure 
Sorry for the delay in answering.  Yes, I added code to GUIslice for handling toggle buttons as radio buttons but this is untested at the moment.   You can test it yourself using `gslc_ElemSetGroup()` call to set the same group number to each toggle button, say 1.   

You need to do this outside of the Builders tags since the Builder currently has the code commented out for this support.  I haven't gotten around to doing any testing and Image toggle wasn't in GUIslice at the time so that's why the code is commented out .  It would only take a few minutes to add support to the builder since its just uncommenting the code and doing a new build.  I'll try to do so when I get the chance.  Likely still a few weeks away.
Paul--

MickTheMechanic, 2021-04-25:
Hi Paul, I actually tested this shortly after posting the question, and seems to be working fine :) 

Thanks for the reply, I hope your feeling better, and get wel soon!