ImpulseAdventure/GUIslice #326: Box corners not rounded if draw function is selected

MickTheMechanic, 2020-12-29:
### Describe the bug
If a box with rounded corners has draw function enabled, the box will no longer have rounded corners during runtime

### Checklist to try first
Confirmed rounded corners is true in .h file

### Expected behavior
*Box corners should remain cornered even if draw function is enabled*

### Additional info
*I have not uploaded a project, to recreate the bug simply place a box with rounded corners and enable draw function. Upload to device and it will have square corners during runtime. *

Pconti31, 2020-12-30:
@MickTheMechanic When you set draw callback it just creates a canvas for you to draw inside.  Make a rounded box , a graph, or space invaders.  
The square frame is simply boiler plater code my Builder outputs inside bool CbDrawScanner(() function as an example.
If you want rounded corners change the lines:
```
  // Draw the frame
//  gslc_DrawFrameRect(pGui,pElem->rElem,pElem->colElemFrame);
  gslc_DrawFrameRoundRect(pGui,pElem->rElem,5,pElem->colElemFrame);
```
Paul--

MickTheMechanic, 2020-12-30:
Thanks alot, I managed to get it working.
I also changed  gslc_DrvDrawFillRect  to  gslc_DrvDrawFillRoundRect, otherwise the corners hang out of the frame.

Some notes: i am sure you are already aware, when doing like this all boxes with the draw callback will have rounded corners. It is not a problem for my current project, but I would have expected ticking the enable draw and enable rounded corner for a box in the builder would produce localized instances of such. Perhaps it is something to consider for future releases?

As always, I appreciate the help.

Please let me know If I should close the issue, or leave it open as a feature request.

Pconti31, 2020-12-30:
@MickTheMechanic There is only one callback so just like with button callback if you want different behavior for different UI elements you need to test for them in the callback.
Say you wanted  rounded for m_pPlaybtn but not for m_pForward inside CbDrawScanner
you just test pElemRef for m_pPlaybtn and draw rounded frame while if it equals m_pForward draw 
normal frame.  

Its not a bug.  Once you declare you want to custom draw in a canvas, just like in the Matrix movie there is no spoon. Or in this case no frame.  You can't disable or enable frame or set it to rounded.
ANYTHING you want to draw you draw or don't use draw callback.

As for the Builder "enable draw and enable rounded corner for a box in the builder would produce localized instances of such. " Sorry, but that is not going to happen; Simply not worth my time given how little the feature is used.
Most likely the next release I'll disable frame and rounded once you set draw callback. 
Paul--

MickTheMechanic, 2020-12-30:
Ok, 

> Most likely the next release I'll disable frame and rounded once you set draw callback.

I figured this would be the solution, its still better than nothing.

Thanks for the help.

closed 2020-12-30 by MickTheMechanic