ImpulseAdventure/GUIslice #329: "gslc_ElemSetCol" don´t work when Element use Flash API

kellers76, 2020-12-30:
i use a box element as a digital status led by changing the fill color from the background color to a glowing color.
I notice that unfortunately I can only change the color with gslc_ElemSetCol if the element does not use the Flash API.

Would this be possible?

I already have 80% of the variables occupied on my project with a Mega2560 and need to see where I can save and outsource....

Pconti31, 2020-12-31:
@kellers76 No.  You can't change color among other items. 
 
Why?  If you use any of GUIslice API calls you know you always need an ElementRef pointer.
What is an ElementRef?  Its the structure that defines a UI Element, type, position, style (frame, no frame, rounded not rounded, etc...).  Color of frame, fill, and glow is also part of this. 

When you ask for an Element to be stored in flash you are asking for this Element Struct to be stored in Flash not RAM.  This is why you save RAM space.  Look at the bottom of GUIslice.h and you will see that the Flash versions are just C Macros that fill out the ElementRef Structure in PROGMEM (Flash).

Once something is in Flash its write protected so it can't be modified.  
Paul--

Pconti31, 2020-12-31:
@kellers76 I created a quick demo of creating multiple Leds in one UI element. A single Box element with a draw callback.  In this case 4 Leds, controlled by booleans stored in global space. You could use int8_t instead and use values from 0 to 2 to set three colors or more for each Led.
I also placed the Led Panel on a base page so all Leds would show up on each page of the UI but you don't need to do that either.  Its just an example but maybe it will help you out.
Paul--
[led-panel.zip](https://github.com/ImpulseAdventure/GUIslice/files/5757436/led-panel.zip)

kellers76, 2021-01-01:
hello paul,
thank you very much for your help and your example!
It helped me a lot.

I took your solution in combination with a box element in flash, the box element only brings the frame and I then draw a filled circle in this frame if needed.
This saves me the RAM memory as I had hoped....
May not be the cleanest solution, but good enough for me.

closed 2021-01-01 by kellers76