ImpulseAdventure/GUIslice #330: Slider value stored in a variable

bajoP, 2020-12-31:
### Feature Request or Enhancement
Make the slider and/or the seekbar store its value in a variable.

### Additional context
Hi, i am working on a project where i would like to have either slider or a seekbar ( would prefer slider) that would store its value into a variable (int) and i would be able to print that value somewhere on the screen. I have been trying to find it but so far i have been unsuccesful, and dont know how to do it. I am posting this knowing there is someone that definetly knows how to do this, but i am not able to figure it out. Thanks for any kind of help in advance.

P.S. Loving the GUIslice Builder

Pconti31, 2020-12-31:
@bajoP Not sure why you have a problem. The code generated by the Builder automatically places the current position in a local int variable nVal inside the CbSlidePos() callback which you can than use or assign to a global for later processing.
Attach is an example that simply displays its value on screen.
Paul--
 
[test-slider.zip](https://github.com/ImpulseAdventure/GUIslice/files/5757667/test-slider.zip)

bajoP, 2020-12-31:
Thanks i will try it

ImpulseAdventure, 2021-01-05:
Closing as I believe there isn't an issue. If the retesting does show a problem, please feel free to reply and I'll reopen and investigate for you. thanks.

closed 2021-01-05 by ImpulseAdventure