ImpulseAdventure/GUIslice #336: Tick marks on radial gauge disappear..on refresh...

bostonmacosx, 2021-01-02:
### Describe the bug
The Radial dial loses the tick marks on refresh

### Device hardware
- *MCU*: ESP32
- *Display with link*:  ILI9341 3.5"
- *GUIslice config selected*: esp nottouch
### Checklist to try first

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* yes
- *Touch works correctly with standalone touch library examples without GUIslice:* no touch

### Expected behavior
Tick Marks stay on refresh.

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info
Update command
        ` gslc_ElemXRadialSetVal(&m_gui, m_pElemRadial1, obj["wind"]["deg"]);`

ImpulseAdventure, 2021-01-05:
Hi @bostonmacosx --

I am guessing that you have configured the radial indicator to be long enough to overlap the tick marks. By design, the tick marks aren't redrawn in order to enable faster redraws in this simple example of a radial gauge. However, if you wish to use a longer indicator (that overlaps with the tick marks), we can enable redraw by making the following change to the /src/XRadial.c:: (change `INC` to `FULL`).

[image]

However, the rendering will probably work best if you reduce the size of the indicator/pointer length (`nIndicLen`) via `ElemXRadialSetIndicator()`. A more advanced version of the gauge could be written that handles overlap, but I won't have an opportunity to do that at this point.

bostonmacosx, 2021-01-09:
Seems to be the correct answer I made the pointer smaller in it cleared itself up thank you

closed 2021-01-09 by bostonmacosx