ImpulseAdventure/GUIslice #338: Red Dot Test draws in opposite direction...drag pen left....dots go right...

bostonmacosx, 2021-01-03:
### Describe the bug
*Please describe what you observe on the display or any error messages*

### Device hardware
- *MCU*: ESP32
- *Display with link*: Generic ILI9341 2.8" resistive 
- *GUIslice config selected*:  esp-tftespi-default-xpt2046_int.h

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* YES
- *Touch works correctly with standalone touch library examples without GUIslice:* Did a TFT_ESPPII demo and tthte touch does rrespond.

### Expected behavior
Red dots follow the pen....

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info
So the touch is working but the drag heads in the wrong direction......drag left .....red dots go right..

ImpulseAdventure, 2021-01-03:
I recommend using the XPT2046_PS touch driver instead of the TFT_eSPI built-in driver as users find the touch response from PaulStoffregen/XPT2046_TouchScreen to be more reliable. To select this you would use the esp-tftespi-default-xpt2046.h config instead.

More importantly, have you run the GUIslice calibration test and updated your config? This is generally all it takes to get the touch mapping correct.

Please take a look at the following if you haven’t seen it yet:

https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-XPT2046

bostonmacosx, 2021-01-04:
Hello there. Thaanks for the advice. 

I did the following. Hope this helps someone. 

I changed my config to 

``   #include "../configs/esp-tftespi-default-xpt2046.h" ``

I then commented out the CS for the screen iin the TFT_ESPI user setup for the touch screen. 

I then addded the Xpt2046 library from PaulStoffregen/XPT2046_Touchscreen

I then ran the calib script. and pastted the values at hte end into the config file....all seems to be well in tthe world again.

closed 2021-01-04 by bostonmacosx