ImpulseAdventure/GUIslice #340: generic 3.5 inch tft (8bit, analog touch) on Arduino Due

mcarosh, 2021-01-05:
Hello all
Now that I´m getting more and more familiar with guislice I realized that this display stuff is really RAM hungry :)
So I switched over from an Arduino Mega to an Arduino Due. With the Arduino Mega everything has been working. But with the Arduino Due I am struggling to work with Guislice.
I managed to get the TFT (3.5 inch, 8bit, analog touch) working with the "MCUFRIEND_kbv" library on the Arduino Due. I can run every example without any problems ( and the Due is a lot faster :) but I´m not able to get it working with GUIslice. 
Arduino compiles without any problems but at startup I always get the error:  Init display handler [ADA_MCUFRIEND] FAIL
I use the same config as I have used with the Arduino Mega and I think there is the problem. Maybe someone can give me hint what i am doing wrong
best regards

mcarosh, 2021-01-05:
Hi Paul
Thanks for your help
sadly none of them works for me :(
What I do not understand is, why is it working with the  "MCUFRIEND_kbv" library??? I thought that inside the config file it is only a pointer to that library?
greez
mcarosh

ImpulseAdventure, 2021-01-05:
Hi @mcarosh --

While your config file points to the library, GUIslice still performs the library initialization according to the config file settings. My guess is that either your SD card config isn't matched, or the mcufriend ID needs to be set.

Please try the following:
1) Attach a copy of the config file that you were successfully using with the ATmega2560
2) When you ran mcufriend_kbv/graphicstest_kbv, did you need to make any changes, and what ID was reported?
3) Please start with the `ard-shld-mcufriend.h` config (ie. no touch)
3a) In the config file, set **GSLC_SD_EN to 0** and **DEBUG_ERR to 2**
3b) You may need to uncomment the **DRV_DISP_ADAGFX_MCUFRIEND_FORCE** to match the ID from the graphicstest_kbv
4) Run `ex01_ard_basic`
4a) To ensure we see the full debug output, please uncomment the `delay(1000)` line
4b) Please post the complete output from your serial monitor here

thanks

closed 2021-01-05 by mcarosh