ImpulseAdventure/GUIslice #343: Intermittent button pressing....

bostonmacosx, 2021-01-09:
### Describe the bug
No error messagess. When I click on a button it sometimes fires and sometimes doesn't. The problem seems more persistent on any other pages otther than the main page. 

### Device hardware
- *MCU*: ESP32
- *Display with link*: ??? Adafruit ILI9341 2.8" resistive (https://www.adafruit.com/product/1651)
- *GUIslice config selected*: ??? `2046 touch esp32
- 
### Checklist to try first
Calibrated twiice. 
Did the red dot test seems to draw fine all over the diisplay and changes to the next "screen rotation" when I life the styllus.

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* Yes
- *Touch works correctly with standalone touch library examples without GUIslice:* Yes

### Expected behavior
Responsive button touches to text buttons and text boxes.

### Initialization messages
No Errors reported.
### Additional info
**.h**

``` // Create E_ELEM_BOX13 box
  pElemRef = gslc_ElemCreateBox(&m_gui,E_ELEM_BOX13,E_PG_POPUP1,(gslc_tsRect){241,154,20,20});
  gslc_ElemSetClickEn(&m_gui, pElemRef, true);
  gslc_ElemSetTouchFunc(&m_gui, pElemRef, &CbBtnCommon);
  gslc_ElemSetCol(&m_gui,pElemRef,GSLC_COL_GRAY,GSLC_COL_RED_LT1,GSLC_COL_BLACK);
//<InitGUI !End!>
```

```
bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui,pElemRef);

  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    Serial.println("intouch");
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BOX10:
        Serial.println("touchbbox");
         gslc_SetPageCur(pGui,1);        
        break;
      case E_ELEM_BTN2:
        break;
      case E_ELEM_BTN3:
        break;
      case E_ELEM_BTN4:
        break;
      case E_ELEM_BTN5:
        break;
      case E_ELEM_BTN6:
        break;
      case E_ELEM_BTN7:
        break;
      case E_ELEM_BTN8:
        break;
      case E_ELEM_BTN9:
        break;
      case E_ELEM_BTN10:
        break;
      case E_ELEM_BTN13:
        break;
      case E_ELEM_BTN14:
        break;
      case E_ELEM_BTN15:
        break;
      case E_ELEM_BTN16:
        break;
      case E_ELEM_BTN19:
      Serial.println("button 19");
         gslc_SetPageCur(pGui,0);  
        break;
      case E_ELEM_BTN20:
        break;
      case E_ELEM_BOX13:
      Serial.println("button 19");
         gslc_SetPageCur(pGui,0);
        break;
//<Button Enums !End!>
      default:
        break;
    }
  }
  return true;
}

```

Pconti31, 2021-01-09:
@bostonmacosx I have Adafruit's 2.4, 2.8 and 3.5 displays.
My 2.4, 2,8 and 3.5 all use stmpe610 touch chips as does the product you pointed to
on the Adafruit site.  All my displays work fine with touch on any page at least in my projects.

Above where it asks for GUIslice config file you listed.
```
GUIslice config selected: ??? `2046 touch esp32
```
2046 would be the xpt2046 chip which would be strange for the display you listed.

You can trace touch by editing your config file and uncommenting
```
  //#define DBG_TOUCH         // Enable debugging of touch-presses
```
make sure and change Serial.begin(9600); to Serial.begin(115200) or Serial.begin(57600).
9600 baud will be much too slow.

You might need to post a zip of the GUIslice config file and if you are using TFT_eSPI
include the User_Setup.h
Paul--

bostonmacosx, 2021-01-10:
Sorry  itt was a generic

https://www.amazon.com/gp/product/B07KPD4DHD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

display. I forgot to update that line. 

Rob

bostonmacosx, 2021-01-10:
Bumping up the serial to 115200 helped....why would that change things though ??

Pconti31, 2021-01-10:
@bostonmacosx The second board you pointed to has the touch chip U2 empty.
Flying kind of blind here so I'm going to stay out of this now and leave any further 
support to @ImpulseAdventure Calvin.  I'm happy its working better for you now.

Why changing serial baud rate helps might make a good discussions topic.
Calvin can do a better job explaining the API internals than I can.  

Paul--

bostonmacosx, 2021-01-10:
@Pconti31 Touch Chip U2? sorry I'm unsure of that that means...

Pconti31, 2021-01-11:
@bostonmacosx Nothing to be sorry about.   I'm just sorry I can't help you any further.

When I viewed the display screen you said you are using it clearly shows an empty spot where a touch interface IC should be soldered to the board (U2),   Nothing further down on the Amazon page for this board shows or identifies a touch interface chip.  This means that this board does not come with touch capability enabled,  Could be an optional feature.

Yet, you clearly are using touch with something and having a response problem.   So your board does support touch but we still do not know what display and touch interface chips you have.

Again, your first post states:
GUIslice config selected: ??? `2046 touch esp32 which matches no config file names.

Any support requires knowing what exact hardware you are using and the config files you are using. 

Even with this information  any further help will have to come from GUIslice's author @ImpulseAdventure.
Paul--

bostonmacosx, 2021-01-12:
It is a XpT2046 on an ESP32 I'll consider this solved.

closed 2021-01-12 by bostonmacosx