ImpulseAdventure/GUIslice #344: Toggle not Toggling....

bostonmacosx, 2021-01-12:
### Describe the bug
*Please describe what you observe on the display or any error messages*

### Device hardware
- *MCU*: ESP32
- *Display with link*: Genericc 3.2" ILI93341 tft with XppT2046 touch
- *GUIslice config selected*: ??? `esp-tftespi-default-xpt2046.h`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* YES
- *Touch works correctly with standalone touch library examples without GUIslice:* yes

### Expected behavior
Toggle switch toggles on touch. 

### Initialization messages
None

### Additional info
**all of thte other buttonss and functiions called work normally iin the interface. The 'TOGGLE1' is seen in the serial monitor when the togggle is clicked**
```
bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui,pElemRef);

  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    Serial.println("intouch");
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BOX10:
        Serial.println("touchbbox");
         gslc_SetPageCur(pGui,1);        
        break;
      case E_ELEM_BTN2:
        fillSettings(0,pGui);
        break;
      case E_ELEM_BTN3:
      fillSettings(1,pGui);
        break;
      case E_ELEM_BTN4:
      fillSettings(2,pGui);
        break;
      case E_ELEM_BTN5:
      fillSettings(3,pGui);
        break;
      case E_ELEM_BTN6:
      fillSettings(4,pGui);
        break;
      case E_ELEM_BTN7:
      fillSettings(5,pGui);
        break;
      case E_ELEM_BTN8:
      fillSettings(6,pGui);
        break;
      case E_ELEM_BTN9:
      fillSettings(7,pGui);
        break;
      case E_ELEM_BTN10:
      fillSettings(8,pGui);
        break;
      case E_ELEM_BTN13:
        break;
      case E_ELEM_BTN14:
        break;
      case E_ELEM_BTN15:
        break;
      case E_ELEM_BTN16:
        break;
      case E_ELEM_BTN19:      
        gslc_SetPageCur(pGui,0);  
        break;
      case E_ELEM_BTN20:
         updateSettings(pGui);
        break;
      case E_ELEM_TOGGLE1:
        Serial.println("TOGGLE");
        // TODO Add code for Toggle button ON/OFF state
        gslc_ElemXTogglebtnToggleState(pGui,m_pElemToggle1);      
        break;
//<Button Enums !End!>
      default:
        break;
    }
  }
  return true;
}
```

Pconti31, 2021-01-12:
@bostonmacosx The code is doing exactly what you are telling it to do.
When you touch the toggle button it fires the callback.
Your code is telling it that if it is now 'ON" you are turning it back to its previous 'OFF' state. 
So no change on screen.  Well maybe a very quick flash.

if you want to manually turn it on or off you need to do this elsewhere.  Otherwise simply test for on/off state using get.  example, say btn1 is to manually turn toggle to the opposite state your callback would look like this:
```
//<Button Enums !Start!>
      case E_ELEM_BTN1:
       // reverse toggle state
        gslc_ElemXTogglebtnToggleState(pGui,m_pElemToggle1);
        break;
      case E_ELEM_TOGGLE1:
        if (gslc_ElemXTogglebtnGetState(pGui, m_pElemToggle1))
          Serial.println("on");
        else
          Serial.println("off");
        break;
//<Button Enums !End!>
      default:
        break;
    }
```
Paul--

bostonmacosx, 2021-01-13:
@Pconti31  so the definition in the DOCS is 

"Toggle a Togglebtn element's current state."

To me this means that a touch on the E_ELEM_TOGGLE should change the sttate of the button thus what I see on the screen....I'm justt confused by the verbiage..

Pconti31, 2021-01-13:
@bostonmacosx It does change state when a you touch it.  

You were putting it back to the original state with your call to gslc_ElemXTogglebtnToggleState().  This call toggles it yet again returning it to the state it was before you touched it on screen.  Thus no change ever happens.

When the toggle button is touched you will see it color change on screen and its state will toggle.  You need to test the new state in the callback with a call to gslc_ElemXTogglebtnGetState() as I show above to determine its new state since the callback is entered on each state change.  

I'm not sure what is confusing to you.
Paul--

bostonmacosx, 2021-01-13:
@pconti  I think what was confusing me is that the toggle would justt toggle on touch without needing tthe fuunction call to make it toggle...I set up the toggle with code right off the bat.....and just didn't testt aa "raw" toggle button without the function I was using...thus it "looked" like it wasn't toggling.

Pconti31, 2021-01-13:
@bostonmacosx I see, so the fact that the callback is entered whenever you touch the button so it could be either on or off was not considered.  

Next time you might want to examine the samples provided as ex46_ard_togglebtn does demonstrate the correct way to use this Element.

You might also want to be aware that checkbox and radio button callbacks work the same way.
You need to test the state to see if they are on or off.  ex04_ard_ctrls is a reasonable example here.

Hopefully, you can close this issue now.
Paul--

closed 2021-01-16 by bostonmacosx