ImpulseAdventure/GUIslice #347: No CallBack from Ckeckbox in Flash

kellers76, 2021-01-15:
hi,
I am using several checkboxes that are located in Flash and I notice that the CallBack function is not executed.
If I set theCheckbox in the builder to "use flash API" = false, then the CallBack function works.

I have tested this with a debug output like shown in example 04...
case E_ELEM_CHECK4:
    GSLC_DEBUG_PRINT("Callback: Check[ID=%d] state=%u\n", pElem->nId,bChecked);
      break;

do I need to call the CallBack function differently for a checkbox in flash?

I want to use this CallBack to write the state to the EEprom to be able to reload it after a power failure or reset.

Translated with www.DeepL.com/Translator (free version)

Pconti31, 2021-01-15:
@kellers76 My bad.   The Builder's code gen is broken.  I'll fix it in the next release.
As a workaround you will see something like this generated for your checkboxes in your *_GSLC.h file.
```
  // create checkbox E_ELEM_CHECK1 in flash 
  gslc_ElemXCheckboxCreate_P(&m_gui,E_ELEM_CHECK1,E_PG_MAIN,150,110,20,20,GSLC_COL_BLACK,true,
    GSLC_GROUP_ID_NONE,false,GSLCX_CHECKBOX_STYLE_X,GSLC_COL_ORANGE,false);
  // gslc_ElemXCheckboxSetStateFunc(); currently not supported by the FLASH _P calls.
  m_pElemCB1 = gslc_PageFindElemById(&m_gui,E_PG_MAIN,E_ELEM_CHECK1);
```
Make note of the Element reference name,  **m_pElemCB1** in this sample case.

You will need to hand edit the code to using your *.ino file support checkbox callbacks.  If you do this in setup() after InitGUIslice_gen() the builder won't delete the code on new code generations.

So you need a gslc_ElemXCheckboxSetStateFunc() call for each callback and remember to use the element reference name in each call like so:
```
 // ------------------------------------------------
  // Create graphic elements
  // ------------------------------------------------
  InitGUIslice_gen();

  gslc_ElemXCheckboxSetStateFunc(&m_gui, m_pElemCB1, &CbCheckbox);
```
Paul--

kellers76, 2021-01-15:
hello Paul,
thank you for your quick reply!
I just tested it in my code and it works.
I can live with it very well for now, thank you.

closed 2021-01-15 by kellers76