ImpulseAdventure/GUIslice #348: Rotary encoder functionality/example

MickTheMechanic, 2021-01-16:
### Feature Request or Enhancement
*Please describe what you would like to see GUIslice do*: I would like to see some kind of rotary encoder functionality. I may already be possible using existing input mapping functions, but there is little information available (documentation says these features are still experimental)

I have found a barebones encoder library named [SimpleRotary](https://github.com/mprograms/SimpleRotary), which when .rotation() is called gives a value of 0,1, or 2 depending on weather the encoder is still, turning cw, or ccw. Is there any chance someone could throw together a small example where  this is somehow mapped to next and prev buttons?

Pconti31, 2021-01-16:
@MickTheMechanic Did you look at the example in the wiki under Tips and Tricks?
[Navigating pages with an Encoder](https://github.com/ImpulseAdventure/GUIslice/wiki/Tips)
Paul--

MickTheMechanic, 2021-01-16:
Aha, I knew I had seen it somewhere, I thought it was in an issue, and I wasn’t able to find it again. I didn’t even think to look there.

If I understand correctly this snippet deals with navigating between pages. What do you think would be the most straightforward approach to navigate elements on the displayed page? Is there some kind of way to navigate between the elements that have touch enabled, when you are on an element it glows, and can then be „activated“ with a push button? Or is something like that overly complicated?

Pconti31, 2021-01-16:
@MickTheMechanic I don't know what to tell you.  Generally I have used rotary encoders to control motors, including stepper motors speed and direction.  The only projects I did for navigation with encoders was when I was forced to use LCD displays 16x2x80 or 20x4x80 that don't support touch interfaces.  I really can't see why it would be useful for navigation otherwise.  But each to their own.  I don't judge.  

Ignoring rotary encoders for the moment, you can enable elements, even hide or make visual under program control.  You can enable touch or disable touch whenever.   I did that with my custom XCombobox.  On occasions I have draw GLOW frames around elements so that it was easy to turn off without repainting the element. Maybe I could help you if you gave a more specific example. 
Paul--

ImpulseAdventure, 2021-01-17:
Hi @MickTheMechanic --

Yes, GUIslice does have some support for navigating the GUI with external inputs.
If you take a look at [ex21_ard_input_pin](https://github.com/ImpulseAdventure/GUIslice/blob/master/examples/arduino/ex21_ard_input_pin/ex21_ard_input_pin.ino), you will see an example where external digital inputs are used to drive the GUI using the CbPinPoll() callback. 

So, you should be able to combine the example that @Pconti31 linked to and the `ex21_ard_input_pin` example to have the rotary encoder advance through the GUI, and optionally have a pushbutton trigger the selection.

So, it might be something along these lines: (note: haven't tested the following combination yet)

in setup(), we associate 3 arbitrary values (1,2,3) with the GUI navigation actions "previous", "select" and "next".
```c++
  // Set the pin poll callback function
  gslc_SetPinPollFunc(&m_gui, CbPinPoll);

  // Create the GUI input mapping (pin event to GUI action)
  gslc_InitInputMap(&m_gui,m_asInputMap,MAX_INPUT_MAP);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 1, GSLC_ACTION_FOCUS_PREV, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 2,  GSLC_ACTION_SELECT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 3, GSLC_ACTION_FOCUS_NEXT, 0);
```

Then, we define the encoder/pin callback function to read the inputs and set a parameter return value corresponding to the action we'd like to take (that we assigned in InputMapAdd():
```c++
bool CbPinPoll(void* pvGui, int16_t* pnPinInd, int16_t* pnPinVal)
{
  // Sample the rotary encoder
  bool bInc = false;
  bool bDec = false;  
  bool bButton = false;
  int32_t newPosition = myEnc.read();
  if (newPosition != oldPosition) {
    if (newPosition > oldPosition) { bInc = true; }
    else { bDec = true; }
    oldPosition = newPosition;
  }
  // Read the pushbutton
  btnSel.read();

  // Advance the GUI navigation from the rotary encoder or pushbutton
  // The action to take is returned in pnPinInd (ie. 1,2,3 assigned earlier)
  if (bInc) { // Rotary Increment -> GUI next
    *pnPinInd = 3; *pnPinVal = 1;
  } else if (bDec) { // Rotary Decrement -> GUI previous
    *pnPinInd = 1; *pnPinVal = 1;
  } else if (btn_sel.wasPressed()) { // Button -> GUI select
    *pnPinInd = 2; *pnPinVal = 1;
  } else {
    return false; // No action taken
  }

  // If we reach here, then an input event was detected
  return true;
}
```

Note that many GUI elements can be navigated this way. However, I still have some open decisions to make about how to show that an element is in "focus" (ie. in focus due to a GUI navigation from a pin/encoder input) versus being selected (ie. the "glow" state). Paul mentioned above the possibility of drawing a rect around the focused element in the "glow" color; this might be one way that I could do it -- I'll have to take a look at it.

I am actually in the process of adding support for the WIO Terminal, which has a 5-way navigation joystick but no touchscreen, so this GUI navigation mechanism is something that I intend to address soon.

Hope that helps!

MickTheMechanic, 2021-01-17:
First of all thanks for the help, this is exactly the kind of thing I was looking for. I had attempted this several times, but I was seriously stuck, and the doxygen info did not provide any real answers in this situation.

I think since in my situation no elements have a visable frame I would simply use the frame as a visual marker. The real challange will be scrolling list boxes and scroll bars in my opinion, or do you already have a solution for that?

@Pconti31 I am only adding the encoder as a redundancy. My touch screen is a resistive type and it appears to be very sensitive to temperature. If I want to use my device say early mornings when its 10°c cooler then you can completely forget using touch. I figure next, prev, and select would be the bare minimum needed for such a redundancy, so an encoder should be ideal. I am open to ideas though, if you have any tips or solutions.

Pconti31, 2021-01-17:
@MickTheMechanic The datasheet should show the operating temp range.   Adafruit displays I own claim -20 to +80 C.  I haven't tested myself so...
  
Here is an interesting article providing engineering tips for cold weather.  
[5 TIPS TO OPERATE LCD DISPLAYS IN COLD ENVIRONMENTS](https://blog.epectec.com/5-tips-to-operate-lcd-displays-in-cold-environments) 

Paul--

MickTheMechanic, 2021-01-17:
I appreciate the tips, the display itsself is fine, its only the touch that is affected. Touch still functions, but the calibration is thrown off with changes in temperature. My guess is the resistive touch screen operates with such minute changes that the difference in resistance caused by the ambiant temperature is enough to affect readings. 

One idea I had was to use the temperature sensor in my project to alter the calibration values for the touch screen during runtime. I havent looked any further into this as of yet, I have no idea if it is possible to change these values during runtime or not (im sure you could tell me if its a viable option or not)

Another idea I had was a calibration routine that could be accessed during runtime via some kind of settings menu. Once again, not sure if its worth the trouble, or even possible.

I am not sure if you are aware of my project (i posted a copy of it in discussion) but it is something that will be used in motorsports, and generally speaking some kind of reduncency would be a good idea, even if there were no problems with touch.

Back to the topic of rotary encoders, I still havent got it up and running yet. I am having some problems.
- The teensy encoder library doesnt work with ESP32
- I have tried SimpleRotary.h with the following code without success :
```

 // Sample the rotary encoder
  bool bInc = false;
  bool bDec = false;
  bool bSel = false;
  uint8_t encoderPosition = encoder.rotate();
  uint8_t encoderButton = encoder.push();
  if (encoderPosition == 1) {
    bInc = true;
  }
  else if (encoderPosition == 2) {
    bDec = true;
  }
  else {
    bInc = false;
    bDec = false;
  }
  if (encoderButton == 1) {
    bSel = true;
  }
  else {
    bSel = false;
  }
  Serial.println(bInc);

  // Advance the GUI navigation from the rotary encoder rotation or pushbutton
  // The action to take is returned in pnPinInd (ie. 1,2,3 assigned in void setup)
  if     (bInc)  {
    *pnPinInd = 1; * pnPinVal = 1;
  }
  else if  (bDec)  {
    *pnPinInd = 2;  * pnPinVal = 1;
  }
  else if  (bSel) {
    *pnPinInd = 3; * pnPinVal = 1;
  }
  else return false; // No pin event detected

  // If we reach here, then an pin event was detected
  return true;
}

```

-NewEncoder.h looks promising as it has provisions for ESP32 specific interrupt, but I cant get it working.

Pconti31, 2021-01-17:
@MickTheMechanic  The article is for touch  and tips on building in heat so touch works as expected.

Changing calibration at runtime  should be possible but would require changing #defines to global integers and you filling them in at runtime with whatever values you wanted.   You most likely would create your own touch routine using ex16_ard_touch_hnd as a guide.  I suspect more than you can do.

Back to your encoder.  Can you post the data sheet for your encoder. 
Also what pins did you use to hook it up?
Up and down counting with a encoder is easy.  Although you need to track the state changes.
Button push not so much due to debounce needing to be supported.   Given your app I'm not sure you need interrupt support.  Simple polling should work for you since its only for navigation and not motor control.   
Paul--

Pconti31, 2021-01-17:
@MickTheMechanic This gives a basic overview of encoders.
[How Rotary Encoder Works](https://lastminuteengineers.com/rotary-encoder-arduino-tutorial/)
Paul--

MickTheMechanic, 2021-01-17:
@Pconti31 The articel discusses using heating to better contrast, sluggyness, and prevent damage to lcd crystals, no mention of touch that I could see. Although weather or not its mentioned is irrellevent as keeping the lcd at a good operating temperature will ensure the touch works.

I would like to avoid using heating, Im not talking about using the device in freezing temperatures, just minimising decalibration from changes in temperature greater than 5°c from the original calibration temperature. 

> Changing calibration at runtime should be possible but would require changing #defines to global integers and you filling them in at runtime with whatever values you wanted

I suspected it would be something like that, Im sure I could figure it out with some trial and error. If I know the calibration values at 15°, 25° and 35° for example i could just make a LUT and pull these values according to ambient temperature. In any case this is a problem for another day.

Unfortunately I dont have a data sheet on hand, but the encoder is one that produces one complete quadrature pulse for every two detents. Similar to [this](https://eu.mouser.com/ProductDetail/alps/ec11e15244g1/?qs=YMSFtX0bdJDiV4LBO61anw==&countrycode=US&currencycode=USD)
I am using an ESP32 and I am using pin 0 for the push button,  34 and 39 for the encoder A and B pins, and the middle pin and 2nd push button pin are grounded. Pins A and B are pulled up with 10k resistors and connected to ground via 0,1uf capacitors.
it works fine with every ESP32 compatible encoder library I have tried so far.

But we may have a much larger problem on our hands...
I cannot get the input example working at all, even with ordinary buttons!
I can Serial.print from void loop and confirm buttons are working, but bool CbPinPoll() is not working at all!
Perhaps my code is fine and the problem lies elswhere?

[ard_rotary_input.zip](https://github.com/ImpulseAdventure/GUIslice/files/5826362/ard_rotary_input.zip)

MickTheMechanic, 2021-01-17:
Here is the [exact encoder I have:](https://www.ebay.de/itm/Mechanical-Encoder-Taster-Switch-12-PPR-Rotary-Incremental-2-Channel-1-St%C3%BCck/363140981886?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) 

And here is how I have it set up:

[image]

Pconti31, 2021-01-17:
@MickTheMechanic Ah... Did you remember to edit your config file?
```
  #define GSLC_FEATURE_INPUT          1  // Keyboard / GPIO input control
```
Paul--

MickTheMechanic, 2021-01-17:
Ahhh, no, ofcourse not! Haha it is workiing now. Thanks!

In the input pin example there is a slider included. Is this intended to be interatced with in some way with the 3 buttons included in the example? I can focus on it (it glows) but I cannot interact with it.

Any tips how interaction could be achieved?

Pconti31, 2021-01-17:
@MickTheMechanic The sample slider, as far as I can see, doesn't interact with the pins. I would define a callback for the slider and check current position 

  int nPos = gslc_ElemXSliderGetPos(&m_gui, m_pElemSlider);

check your  inc or dec modifying nPos accordingly followed with a gslc_ElemXSliderSetPos()
Paul--

Pconti31, 2021-01-17:
@MickTheMechanic Now that I think further that won't work either.  The callback isn't a bad idea but only gets called when pos changes.  So you need to use your encoder to directly call gslc_ElemXSliderSetPos() somehow.
How you know when to do that I can't say.  Have to leave this for Calvin's input.
Paul--

ImpulseAdventure, 2021-01-17:
Hi guys --

We actually already have support for navigating increment/decrement widgets as well :)

When assigning inputs, we can use two other "actions", one for increment and another for decrement.

For example:
```c++
  gslc_InitInputMap(&m_gui,m_asInputMap,MAX_INPUT_MAP);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 1, GSLC_ACTION_FOCUS_PREV, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 2, GSLC_ACTION_SELECT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 3, GSLC_ACTION_FOCUS_NEXT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 4, GSLC_ACTION_SET_REL, -1); // Decrement value by -1
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 5, GSLC_ACTION_SET_REL, +1); // Increment value by +1
```
(reference: example ex41_ard_ctrls_small_input)

You then just need to associate the appropriate button / rotary inputs to the action values above (ie. 4 or 5). ACTION_SET_REL changes the widget value by a relative amount in CbPinPoll(). We also have the ability to change a widget to an absolute value with ACTION_SET_ABS. The value after the ACTION_SET_REL/ABS defines the amount to change or the new value.

I haven't thought through the ideal interface flow yet, but I was thinking that a rotary encoder could switch between changing the focused widgets, and then (perhaps after a select on a "slider"), the rotary encoder then is used to increment/decrement the widget. After using the select button, the rotary encoder again returns to widget focus navigation.

Hope that helps!

MickTheMechanic, 2021-01-17:
I think the easiest solution would be long press changes encoder rotation from next/prev to set +/ set-, and vice versa. 

Thoughts?

MickTheMechanic, 2021-01-17:
Is there some way to check if any sliding element or listbox is in focus?

Pconti31, 2021-01-17:
@MickTheMechanic I think gslc_CollectGetFocus() is what you want.
You then need to compare to the enum of the listbox or slider.
Paul--

MickTheMechanic, 2021-01-17:
Thanks a bunch!

Is there some kind of unique identifier for different types of elements? (I assume each element is identified by an intiger and there is no way to tell them apart, but I thought I'd ask anyway just in case)

MickTheMechanic, 2021-01-17:
Also, I have added some code to map encoder rotation to Rel+/Rel- afer a long press. It appears to be working, however it does not do anything to the focused element (Slider in this case)

Am I missing something? Do I need to manually set the slider position using Res+/Res- or should it happen automatically?

[ard_rotary_input_slider_control.zip](https://github.com/ImpulseAdventure/GUIslice/files/5826936/ard_rotary_input_slider_control.zip)

Pconti31, 2021-01-18:
@MickTheMechanic 
_Is there some kind of unique identifier for different types of elements? (I assume each element is identified by an integer and there is no way to tell them apart, but I thought I'd ask anyway just in case)_
As I told you above the gslc_CollectGetFocus() returns the enum of the element in focus.
An enum would be like "E_ELEM_SLIDER1".  They have to be unique.

Sorry, I can't help you with anything else since I'm not the author of the API and I have no idea what Calvin has implemented for button support.  
Paul--

MickTheMechanic, 2021-01-18:
> As I told you above the gslc_CollectGetFocus() returns the enum of the element in focus.

I understood that, and it is very helpful

> An enum would be like "E_ELEM_SLIDER1". They have to be unique.

I understand this too. What I was trying to ask is if there is a way to identify element types, not the elements themselves. I assume its not possible, but Im just trying to be thorough as something like that would be perfect. Otherwise I need to include the enums of all sliding elements into a function with gslc_CollectGetFocus(). I know its not hard, but if there is some way to identify element types then why not take advantage of it. 
Perhaps the solution is to use bool CbSlidePos(void* pvGui,void* pvElemRef,int16_t nPos) to set a boolean true that can be used to remap the encoder rotation, that way it works any time _any_ slider is in focus. 

I also figured out what the problem was with my sketch, I needed to update #define MAX_INPUT_MAP. It is working now. 

Also, thankyou for your help, as always, much appreciated!

Pconti31, 2021-01-18:
@MickTheMechanic Yes, you can determine type.  The interesting basic types are outlined in the gslc_teTypeCore struct.
Ie: GSLC_TYPE_BTN, GSLC_TYPE_TXT
while the extended type like XSlider would be listed in their header file XSlider.h 
ie: GSLC_TYPEX_SLIDER
The nType is available in the gslc_tsElem struct.
if you have an element ref in hand you can access type it by
  gslc_tsElem*    pElem = gslc_GetElemFromRef(pGui,pElemRef);
  if (pElem->nType == ???)

if you only have an ENUM you can lookup the Element Ref using
gslc_PageFindElemById()

Paul--

Pconti31, 2021-01-18:
@MickTheMechanic 
@ImpulseAdventure I do think its odd the gslc_CollectGetFocus() returns Element's ENUM instead of the Element Ref pointer.  Might be something to consider for an enhancement.
Paul--

ImpulseAdventure, 2021-01-30:
Hi guys -- sorry for the lengthy delay. Was very busy with work for the past while.

@MickTheMechanic --
Regarding the calibration, yes, it is possible to perform calibration at run-time. This is how the `diag_ard_touch_calib` sketch works. However, calibration does take a little bit of code to make reliable (ie. filtering out bad touches, dealing with orientation, etc.).

In your situation, I think you would actually be better off using the ambient temperature reading and then interpolating. Might not even really need a LUT, could potentially get away with linear interpolation from a couple measured touch readings (basically the resistance of the overlay) vs temperature. Then you just call the following to update your calibration dynamically:
```c++
    gslc_SetTouchRemapCal(&m_gui, m_nTouchCalXMin, m_nTouchCalXMax, m_nTouchCalYMin, m_nTouchCalYMax);
```

It sounds like it might be useful for me to create a new function called `FocusElemGet()`.
- It would return the Element Ref pointer if an element is focused, otherwise NULL
- Note that the focus detection would be done within the top-most page/layer that is enabled

However, I should take another look at your main goal here (in having the encoder switch between modes) and provide a more seamless solution for you (maybe even make this a built-in mode). I'll take another look at it and will followup once I've had a chance to review it.

Note that `CollectGetFocus()` is basically a private/internal function and wasn't intended to be called from user code.
In fact, CollectGetFocus() doesn't actually return the element index -- it actually returns the index into the current "element collection" (which is often a page). It may turn out that the index is the same as the element enum, but that is not guaranteed. If you look at how `CollectSetFocus()` works, it will make more sense why it only stores a 16-bit index rather than a pointer to an element (we are creating a simple iterator to walk the page).

thanks

MickTheMechanic, 2021-01-30:
Thanks for the reply,  gslc_SetTouchRemapCal() provides a solid base to work from, so thanks for that!
I have since decided against rotary encoder and will go with 3 buttons instead, allthough the working principles will remain the same, the select button changes the functionality of the prev and next buttons if a scrollable is in focus.
Another minor aspect that will require attention for a seamless example is moving the highlighted element inside a listbox. It will also need to be mapped to the scrollbar in some way.
I have also been busy the last few weeks, I need to take another look at my files, but I believe there was some problems when trying to implement button control to an example generated by the builder. I will take a look and attach an example.

ImpulseAdventure, 2021-01-31:
Hi @MickTheMechanic —

This is good feedback. I plan to start working on a more complete handling of the external control over GUI elements now. I will start with the ListBox scrolling.

I think I will define a new attribute for controls to indicate whether they have an “edit” capability.

My tentative thought was for the following, but I would be most interested in suggestions to make the experience more natural.

I would like to see if we can make the GUI fully usable with 3 inputs, let’s call them next, previous and select. For people with more inputs, it is much easier to make a more direct GUI navigation.

One idea: the GUI would operate in two modes when working with external inputs (like buttons, switches or even commands from external controllers): navigate mode (default) or edit mode.

A select on an element with an edit attribute switches between navigate and edit mode. Directional inputs (buttons, encoders, etc) in navigate mode change focus, whereas in edit mode will modify the control in some way. Pressing select in edit mode would generally switch back to navigate mode.

I would also like this to support controls like keypads too. But with Keypads we might operate differently in edit mode as there is also a need to “select” in edit mode. My initial thought there is to have select just press buttons as before, but the Done / Cancel buttons would be used to get back to navigate mode. If you had more inputs (eg. 5), then a keypad could also provide left/right cursor navigation in edit mode, for example.

I’d be interested in any feedback from anyone on this point in case there are better ideas out there :)

thanks

MickTheMechanic, 2021-01-31:
I think the select button changing the next/prev functions if a scrollable is in focus covers pretty much all situations except for num input and text input, in which case I would suggest prev and next are used to navigate the keypad and select is used to enter a key,  enter or exit on the virtual keypad are selected to return to normal navigation. This would require that keys on the virtual keypad are focusable (not sure if this is already there, if not it would be an area that requires attention) I can’t see any other situations that would require any special attention? If a GUI is more complicated than that I think more than 3 buttons would be needed anyway.

MickTheMechanic, 2021-01-31:
As described earlier there is some kind of problem when adding button input to a sketch produced by the builder. Not exactly sure whats going on, I probably just forgot to add some code somewhere or something like that, but ive gone through the sketch a few times and couldnt find anything.
ard_button_input is based off one of your library examples, everything works fine. 
ard_3_button_test was built using the builder and uses the same button code but doesnt seem to regiter any button presses.

 
[ard_3_button_test.zip](https://github.com/ImpulseAdventure/GUIslice/files/5901049/ard_3_button_test.zip)
[ard_button_input.zip](https://github.com/ImpulseAdventure/GUIslice/files/5901050/ard_button_input.zip)

MickTheMechanic, 2021-02-02:
I know you guys are probably super busy, but I just wanted to give the thread here a bump and ask if either of you noticed why button input isnt working with the example from the builder?

Pconti31, 2021-02-02:
@MickTheMechanic 
@ImpulseAdventure One thing I did notice doing a desk check (just like we used to do when we needed to debug our old Fortran iv punched card programs) is that you are doing initialization in wrong order.
should be 
'''
  // ------------------------------------------------
  // Create graphic elements
  // ------------------------------------------------
  InitGUIslice_gen();

  // Set the pin poll callback function
  gslc_SetPinPollFunc(&m_gui, CbPinPoll);

  // Create the GUI input mapping (encoder event to GUI action)
  gslc_InitInputMap(&m_gui, m_asInputMap, MAX_INPUT_MAP);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 1, GSLC_ACTION_FOCUS_NEXT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 2, GSLC_ACTION_FOCUS_PREV, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 3,  GSLC_ACTION_SELECT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 4, GSLC_ACTION_SET_REL, -1); // Decrement value by -1
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, 5, GSLC_ACTION_SET_REL, +1); // Increment value by +1
```
I can't actually run the program but if its still broken I'll  do another desk check.
Paull--

MickTheMechanic, 2021-02-02:
That did the trick. Thanks! 

Some observations from the above sketch;
* Text input fields are not focusable
* Keys on the keypad are not focusable

Otherwise all other aspects seem manageable.
I will continue playing around and see if I notice anything else that may need addressing regarding external inputs.

I was snooping around the documentation and stumbled across this:
```
void* gslc_GetXDataFromRef	(	gslc_tsGui * 	pGui,
gslc_tsElemRef * 	pElemRef,
int16_t 	nType,
int16_t 	nLineNum 
)		
Returns a pointer to the data structure associated with an extended element.

Example usage: gslc_tsXListbox* pListbox = (gslc_tsXListbox*)gslc_GetXDataFromRef(pGui, pElemRef, GSLC_TYPEX_LISTBOX, LINE);
Parameters
[in]	pGui	Pointer to GUI
[in]	pElemRef	Pointer to Element Reference
[in]	nType	Expected type indicator (ie. GSLC_TYPEX_*)
[in]	nLineNum	Line number from calling function (ie. LINE)
Returns
Void pointer to extended data (pXData), or NULL if error. Needs to be typecasted accordingly.
```
What sparked my interest was nType, got me thinking about if there is some way to check the element type? Would be very useful for 3 button input.

ImpulseAdventure, 2021-02-03:
Hi @MickTheMechanic --

I have been working on incorporating a more integrated support for external input-based navigation.
My intention is to extend this properly to support listboxes, keypads, etc. This should be useful for applications like your own, and other development boards that integrate an input device (like the **M5stack**, **WIO terminal** and **Adafruit seesaw**).

The hope is that, as a library user, you would not need to do anything specifically to support this, other than configuring the functions that you want the buttons to perform.

Regarding `nType`: I had provided this capability in the API so that people who create their own custom widgets could also detect specific widget types. However, I don't plan on using that attribute to differentiate functionality for the external-input navigation. Instead, I will be updating the ElemXCreate APIs to add new attributes that dictate how external-input should operate, and keep the core GUIslice library functionality agnostic to the individual widget types.

MickTheMechanic, 2021-02-03:
Sounds very nice, looking forward very much. 
I guess it’s quite a large undertaking and will probably take a bit of time?
I would love to help but I think it’s probably far too complicated for me to be any real help, but I can definitely get involved with testing!

ImpulseAdventure, 2021-02-04:
Hi @MickTheMechanic -- thank you very much for the offer. This particular enhancement wasn't trivial as it needed integration into some of the event handling for the GUIslice core.

Thankfully, I have now completed an initial attempt at adding support for a navigate/edit mode toggle in the GUI from external pin/button inputs. To start with, I have added support for XListbox and XSlider. If things look reasonable, I will explore adding support to XKeyPad as well.

Please try the following branch and let me know what you think: https://github.com/ImpulseAdventure/GUIslice/pull/359

If you don't have a sample to work with, I can post an example of ex04/ex30/ex31 modified to add external input support. But all it takes is usually adding the following to the existing example sketches:

Add alongside the existing global definitions:
```c++
#define MAX_INPUT_MAP       3
gslc_tsInputMap             m_asInputMap[MAX_INPUT_MAP];
```

Add after gslc_Init():
```c++
  // Create the GUI input mapping (pin event to GUI action)
  gslc_InitInputMap(&m_gui, m_asInputMap, MAX_INPUT_MAP);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, GSLC_PIN_BTN_A,      GSLC_ACTION_FOCUS_PREV, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, GSLC_PIN_BTN_B,      GSLC_ACTION_SELECT, 0);
  gslc_InputMapAdd(&m_gui, GSLC_INPUT_PIN_ASSERT, GSLC_PIN_BTN_C,      GSLC_ACTION_FOCUS_NEXT, 0);
```

MickTheMechanic, 2021-02-04:
Looks good, I will test it out this evening.

I have a question about focus, what defines wether or not something is focusable? Can focus on an element be enabled/disabled? 
Or is is simply active on elements that have touch enabled.

You also mentioned getFocus was experimental, but I think something like this would be very useful for triggering other events when an element is on focus.

anyhow I really appreciate your work!

MickTheMechanic, 2021-02-04:
Its working very well! Awesome work. I just added a few lines to my tire tool sketch and i can navigate with the buttons pretty smoothly, but there are some things that need to be adressed.

Some observations:
* base page buttons cannot be selected 
* Pauls toggle image button workaround is a bit tricky now. You cant see if its in focus. This brings up another problem:
* If glow is used as a focus indicator, this means when you get around to creating xImageToggle you might have to consider that such a button contains four images, the two to be toggled and another two to show focus for two states.
* Some kind of getFocus might be a good Idea, it could be used to enable/disable a frame or similar to show that certain elements are in focus (an image with touch enabled is a perfect example of where this would be useful)
* Regarding listbox, if the scrollbar is scrolled it should pull the text glow with it, and if the text glow is scrolled it should pull the slider (in my opinion)

MickTheMechanic, 2021-02-05:
@Pconti31 
@ImpulseAdventure 

Could one of you guys post a code snippet showing how gslc_CollectGetFocus() works, with structs ect. 
I understand its experimental, but I would like to use it in my project to show a border on clipping rect's if they are in focus.
Thanks.

ImpulseAdventure, 2021-02-05:
Hi --

I should be able to add the ability to cycle through the various page layers (ie. pages in the "page stack"). This would enable you to navigate the widgets on the base page as well.

Regarding `ColletGetFocus()`, given the discussion in https://github.com/ImpulseAdventure/GUIslice/issues/348#issuecomment-770158495 would it be more useful if I just provided the proposed `FocusElemGet()` for you? If you can explain a little more about how you planned to use it with the clipping rects, that also might help me ensure I'm adding functionality that addresses the need.

Regarding your listbox comment ("pull the text glow"), I think you mean that if you manipulate the scrollbar (eg. with the external input controls) and scroll to the point where the currently-selected line item is out of view -- that you want the top/bottom visible line item to be selected instead. Is that right?

As for the ToggleImage widget, yes, this is a bit problematic. Right now we are relying on using a "focus" state to depict a navigation selection. I need to think about it further to explore the options. Clearly, I'd rather not add another "state" (focus) to every widget just for this. Adding 4 image states could be an option but perhaps a bit heavy. Ideally, we would use something like a frame highlight to indicate the focus for this particular control. I'll give it some more thought, but I'm certainly open to ideas!

thanks

MickTheMechanic, 2021-02-05:
I will try my best to explain the dificulties that are beginning to show up. 

Because of the regional invalidation problem described in #328 I am using images with touch enabled rather than image buttons for most of my gui buttons.
The reamaing buttons are toggling image buttons, which were built using Pauls workaround shown in #325 . The workaround uses a box with touch enabled to limit the refresh area to the clipping region inside the box. The images to be toggled are then shown inside the box.

The problem with this is neither have the ability to show focus. 
If its not too much to ask I think providing FocusElemGet()  would be a really great tool. It would certainly provide a solution for these problems, The user could for example enable the frame if the element is in focus, or trigger any other feedback event (Haptics for example).

I guess the next question to consider is if the 2nd image in an image button should depict focus or depression. It might be better to use a frame to show focus for image buttons. This would also make a future imagetoggle button easier to implement.

> Regarding your listbox comment ("pull the text glow"), I think you mean that if you manipulate the scrollbar (eg. with the external input controls) and scroll to the point where the [...]

Its tricky to explain, its kinda like if you scroll the listbox the slider should move with the list, and vice versa. 
At the moment if you scroll using the slider the selected line goes out of view, and if you scroll the list box directly the slider doesnt move. It just feels kinda clunky, if you scroll using the slider, when you select the list box afterwards it kind of skips back to the selected element. 

Im not sure if this is something to take care of in the source code or if if its just something the user should setup themselves. Another possible solution could be to only make the slider selectable if the listbox is generated with one. Maybe some more brainstorming is needed, im interested to hear what @Pconti31 has to say about it, maybe he has some interesting ideas about it.

ImpulseAdventure, 2021-02-05:
Thanks for the clarifications...

Yes, I will look at adding the FocusElemGet(). Alternately, we could consider registering a callback function that returns the element ref any time the focus changes.

Regarding the listbox: yes, the scrollbar is not currently associated with the Listbox, so that is why navigating inside the list doesn’t get reflected in the nearby “slider” widget. The Builder just generates these two widgets alongside each other. I could look at providing a means to add a link, making a two-way sync more automatic.

I am more inclined to leverage an image frame to reflect focus, rather than the “glow” state... I am thinking that we may want to extend widgets to support separate focus & glow states. That way widgets can decide how to render these states differently.

Thanks for the great feedback! External GUI navigation was an “alpha” feature, so I expected that the requirements would become clearer as I heard how users might want to leverage the feature.

Pconti31, 2021-02-06:
@MickTheMechanic 
@ImpulseAdventure Since I finally finished the next release of the Builder 0.16.b007 with a brand new Office style ribbon I thought I would take a moment and implement a Toggle Image Button.
It's beta so point out any issues.   I have no idea what Calvin is doing with focus so don't ask me about that.
Don't forget that the builder doesn't support this yet since its not in the GUIslice API and once it is I will add it to the Builder. A sample program is supplied.  I haven't yet tested the flash versions so beware...

[ex29_bld_toggleimgbtn.zip(https://github.com/ImpulseAdventure/GUIslice/files/5937656/ex29_bld_toggleimgbtn.zip)
 
Paul--

MickTheMechanic, 2021-02-06:
Works great!

>Don't forget that the builder doesn't support this yet since its not in the GUIslice API and once it is I will add it to the Builder.
Its a shame as soon as the sketch is modified using the builder the togglebutton no longer works. 
I was hoping to throw together a larger sketch do some tests to see how it affects the refresh/regional invalidation, and possibly implement it in my project.
I also downloaded the latest builder, noticed a few problems with it. I have opened some issues. I must say im a bit skeptical about the new layout, the older style felt more natural for me, maybe its just something to get used to.
This issue is starting to cover mutliple issues, perhaps we should move to discussions or continue this thread further in #325, or even start a new issue.

> I have no idea what Calvin is doing with focus so don't ask me about that.
If you were referring to this:

> im interested to hear what @Pconti31 has to say about it, maybe he has some interesting ideas about it.

It was more or less just asking your opinion in a kind of brainstorming way, if you had anything to input as to how elements should be interacted with via external input. Thought you might be able to provide some nice ideas

Pconti31, 2021-02-07:
@MickTheMechanic While the builder doesn't currently support this new toggle image button that doesn't mean I can't build a test version.  I'll post link when one is available.
as for:
If you were referring to this:
im interested to hear what @Pconti31 has to say about it, maybe he has some interesting ideas about it.

Nope. I simply meant I have no idea how Calvin's focus code would or will affect the new toggle button.
I haven't had much free time lately to look this discussion and now that I have yet more Builder bugs to investigate it could be longer still.  The delete key was easy fix but the image copy/paste is a real nasty problem.
That could take a while.
Paul--

Pconti31, 2021-02-07:
@MickTheMechanic 
@ImpulseAdventure So here is a snapshot builder with support for Toggle Image Buttons for each platform.

[builder-win-0.16.b009-IMGTOGGLEBTN.zip](https://drive.google.com/file/d/1Ff7ZxW1nGWAZJQ0GF_fhCfKVfDELcVjj/view?usp=sharing)
[builder-linux-0.16.b009-IMGTOGGLEBTN.tar](https://drive.google.com/file/d/1ZLFXTLa8LpYWbWSf4uGCYfC1dE76ryQb/view?usp=sharing)
[builder-osx-0.16.b009-IMGTOGGLEBTN](https://drive.google.com/file/d/1ljWZRCpxSvk25mcp06kdZ1MBGR_y86-a/view?usp=sharing)

Attached is a new version of the test project generated by this Builder.
[ex29_bld_toggleimgbtn.zip](https://github.com/ImpulseAdventure/GUIslice/files/5939252/ex29_bld_toggleimgbtn.zip)

You need to copy this version of the XToggleImgbtn to your GUIslice/src/elem folder to compile.
[elem.zip](https://github.com/ImpulseAdventure/GUIslice/files/5939255/elem.zip)

Finally, to create your own image toggle buttons simply create a normal image button and look down the property view and find the new property "Toggle?" and set it to true.   

Paul--

MickTheMechanic, 2021-02-07:
@Pconti31  Builder is working great!
Some observations regarding toggle button:
* Frame is shown regardless if enabled or not
* I have tested with external input, If the element is in focus a white frame is shown. This is fantastic, works well. But it appears to be there to show if the button has been selected. It would be awesome if this was limited to focus (the toggle is already visual confirmation of the button press). I understand we have to wait on calvin regarding focus, its just an observation.
* If possible the select/focus frame color should be selectable in the builder, or at least adjustable in code (could be already? I havnt had a detailed look yet). Rounded corners would also be awesome, but maybe not possible?
* Unfortunately It seems like the images are not centred in the frame (example provided). Possibly due to image size, possibly something else.
* Refresh/invalidation looks good, appears to be limited to within the frame (will test further)

[ex29_bld_toggleimgbtn_test.zip](https://github.com/ImpulseAdventure/GUIslice/files/5939584/ex29_bld_toggleimgbtn_test.zip)

Pconti31, 2021-02-07:
@MickTheMechanic 
@ImpulseAdventure Ah! I had copied my XTogglebtn to creater XToggleImbtn and forgot that the frame is forced on during creation.  This version should let you turn it on/off.
[elem2.zip](https://github.com/ImpulseAdventure/GUIslice/files/5939672/elem2.zip)

You can change the frames focus color simply.  You can look at the first example 29 I sent you.
```
 // Set frame glow color during initialiaztion
  gslc_ElemSetGlowCol(&m_gui,pElemRef,GSLC_COL_WHITE,GSLC_COL_WHITE,GSLC_COL_WHITE);

 // turn glow on/off whenever
 gslc_ElemSetGlow(&m_gui,m_pElemToggleImg1,true);
```
Just be sure to do so outside any Builder's tags so it doesn't get deleted.   If you want an enhancement to the builder post the request in my repository not here.

The centering issue is due to the fact that I forgot to inform you that this test builder isn't doing any new sizing. That's left up to you.   If you look closely at the example icon for exit you should notice its 32x32 but in the builder I defined it as 33x33 to allow for the frame which was enabled in the version I sent you.  If you are going to turn frames on/off you need to allow space for them.   The Release builder will likely just increase all image buttons by one pixel to allow for this.

I'm going to watch the Superbowl pregame now so I won't be checking back today.
Paul--

MickTheMechanic, 2021-02-07:
Unfortunately the toggle button does force a partial page redraw, which in turn crashes my display driver when using it in my tire tool project on the same page that is running the thermal camera. I wont be able to implement it in my project until a solution for #328 is found. But it does work perfectly for less demanding GUI's.
I am waiting on a new display which should be supported by the tft-espi library, which should provide an optimized driver for for the ESP32. Perhaps the partial redraws are then much faster/less noticable/less problematic.
It seems the toggle button is being redrawn twice, it is redrawn once with the current image, then again with the toggled image. Both times a partial page redraw is forced.
So it is at this stage still not as good as your makeshift toggle button using clipping (shown in #325 ). 

Enjoy the superbowl!!

ImpulseAdventure, 2021-02-11:
Hi @MickTheMechanic --

Just wanted to give you a quick update on what I'm currently working on:
- Adding ability for external input navigation to step through multiple pages in the page stack
- Optimizing redraw for #328 . For this enhancement, I decided that it made more sense to split the button redraw and the callback function triggers into separate update cycles.

Also, you should generally find the performance on the ESP32 with a TFT_eSPI supported-display should be much better.

thx

MickTheMechanic, 2021-02-11:
Sounds great, looking forward to testing it out!
Progress has slowed here, I want to continue testing with a proper screen/driver combo, just waiting on the screen to arrive. Im using the time to focus on other parts of the project.
I kinda gave up on my current set up, I cant use spiffs because the tft_espi library is needed for the tft_efx library to work, and the tft_efx library is needed for spiffs to work. Id much rather have everything working how it should rather than searching for workarounds all the time just because a $10 screen isnt supported. I am also hoping performance will be generally better when using the tft_espi library. A few FPS more on my thermal cam would be awesome.

Will you still add a getFocus() function?

ImpulseAdventure, 2021-02-12:
Hi @MickTheMechanic --

I have now rewritten the external input navigation logic to enable proper navigation across all "pages" in the page stack. So this should let you navigate between the base pages, popups, etc. I have tested it with ex24_ard_tabs.

This update also adds FocusElemGet(), which you can call get identify the element that is currently in focus.

Please see the following branch (if you happen to have a chance to try it out, that would be great, but otherwise I'll keep advancing my updates):
https://github.com/ImpulseAdventure/GUIslice/tree/WIP348-Input2

Will be good to have a faster MCU + display setup on your end given the dynamic updates you have, and yes, finding a display with better support should save a lot of effort :)

I will now proceed to add the separate focus stage on widgets and then the optimized redraw.

cal

MickTheMechanic, 2021-02-13:
I tested it out and noticed a few things.

* FocusElemGet() works well. Nice one!
* I can navigate elements on the current page and base page, also very nice.
* Regarding the base page, I am using it for my home button, and for some headers like time, date, ambiant temperature and so on. One thing that may need some attention is the home button. I am using the base page to place a home button on every page, and for the home page and screen saver I just cover it with a box. This could become problematic once the focus frame is added. I am not exactly sure how the base page was originally intended, If it wasn't intended for such things like a home button then I will just manually add a home button on each page. If however, it was intended for things like a home button, it may require some attention, or possibly a home button element as a future feature. (or maybe there is some trick to build a home button that i'm unaware of?)
* There are some bugs with text buttons and scrollables. If they are focused glow becomes gray and doesnt change back again, and inside listboxes all items glow gray and you cant tell which item is currently focused. Onced a slider has been focused once the slider dissapears and only the glow remains.

ImpulseAdventure, 2021-02-13:
Thanks for testing out the development branch... there are likely a couple loose ends I still need to resolve, so any feedback is appreciated.

Regarding the home button: I think you should be able to do what you are asking for with the following:
```c++
// Jump to home page
ElemSetClickEn(pGui, m_pBtnHome, false);
ElemSetVisible(pGui, m_pBtnHome, false);
SetPageCur(pGui, E_PG_HOME);

// Jump to other page
ElemSetClickEn(pGui, m_pBtnHome, true);
ElemSetVisible(pGui, m_pBtnHome, true);
SetPageCur(pGui, E_PG_OTHER);
```

As for your last bullet point regarding observations in text buttons and scrollables: I saw your comment to ignore the “last statement”. I assume you mean that I should ignore the point about the slider but that the text buttons and list boxes still exhibit the glow issue.

Thanks!

ImpulseAdventure, 2021-02-15:
I have now made a significant number of updates to provide full support for external input navigation of the GUI.

Please have a look at the test branch:
- [Int348-Input2b](https://github.com/ImpulseAdventure/GUIslice/tree/Int384-Input2b)

Major updates include:
- Added ability to fully navigate the XKeyPad.
- All major widgets now support navigate / edit. Tested with XSlider, XListbox, XTogglebtn, XCheckbox, XKeyPad and standard controls.
- Widgets now control their own entry/exit from navigate/edit mode. This enables complex widgets like XKeyPad.
- Note that "editable" widgets (like XKeyPad and XListbox) require a "select" action in order to enter edit mode. We don't currently have a way to differentiate (visually) between an element in focus in navigate mode vs edit mode.

This is still a test branch, so there are a significant number of debug messages enabled. If you get a chance to test out the basic operations, I'll proceed to clean up for a proper release.

thanks!

MickTheMechanic, 2021-02-17:
I have tested out Input 2b, (I havent tested 2a, are those changes also inlcluded in 2b?)

Anyway here is what I found:
* Multi-page navigation is working
* Popup dialog navigation is working
* XKeyPad navigation is working, but there are some minor issues. 
    * Navigation/glow only becomes active after pressing select button after the keypad is already open (not such a big deal). 
    * Glow color is quite hard to see. Is the color arbitrary or fixed? 
    * During navigation I noticed the keys on the numpad dont navigate in the correct order. The current order is -,BS, 9-1,>,<,ENT,ESC.  However due to layout of the numpad it should be <,>,ESC,0-4,-,BS,5-9, . , ENT   _or_   <,>, ESC, -, BS, . ,ENT, 0-9.   I havent tested alphaPad yet, the navigation order may also be wrong there. 
* I have tested the following widgets: Slider, listbox and togglebutton. They seem to be working fine.
* I have been giving it alot of thought and I think the following would be the best solution for glow/focus; Slider should use frame for focus and glow for editmode. Listbox should have frame for focus, as for edit mode im not sure, maybe a frame on the text item, or a different color glow on the text item, but I think nothing would also be ok. Toggle button seems fine how it is with the frame color/glow, nothing more is needed. Image-buttons, buttons, images, and boxes just need a frame for focus, num input and alpha input boxes too. And the keypad glow needs more contrast, or should be user defineable. 

I also wanted to point out, further up in the comments Paul has posted some code for an xImageToggleBtn, and has also added it to the builder. I say this because of your comment in #325, just incase you overlooked it. An example of ElemXToggleImgBtnCreate() as you described in 325 would be perfect, it would work the same as a normal button with press/release "glow". This together with navigation, focus, and the revised refresh your working on in #328 should result in a very smooth and professional gui experience!

ImpulseAdventure, 2021-02-18:
Great feedback, @MickTheMechanic ! This is very helpful.

A few responses:
- The colors (for the XKeyPad) are fully customizable in the `XKeyPad_Alpha-setup.h`, for example. In this particular update I did increase the contrast a bit (by darkening the normal fill colors, but I can certainly increase the contrast further). Look at the **XKEYPAD_COL** section of the file -- you can use one of the built-in GUIslice color defines or use your own RGB triplet, eg:
  - `#define XKEYPAD_COL_BASIC_GLOW GSLC_COL_BLUE_LT4`
  - `#define XKEYPAD_COL_BASIC_GLOW (gslc_tsColor) { 128, 128, 255 }`
- Yes, we should rearrange the XKeyPad focus order. Currently, the focus navigation uses the sequence of keys defined in the **KEYPAD_LAYOUT** in `XKeyPad_*-setup.h`. So we should be able to reorder the keys in this file -- I'll make some mods to change this up to make it more intuitive.
- Good feedback on the glow vs focus indicators. Now that I have separated out the glow & focus states, I should be able to differentiate these along the lines you describe.
- Thank you for pointing out Paul's XImageToggleBtn from this thread! I was looking at the other thread and had started to create a new XImageToggleBtn, but as you point out Paul had attached a reference example earlier here. I may merge a few changes between the two (eg. consider supporting 4 ImgRef, normal/glow for both states), but I intend to keep the API largely intact.

Thanks again for taking the time to contribute your ideas here!

ImpulseAdventure, 2021-03-01:
I have now created a new update to incorporate some of the recommendations you have noted above:
- XKeyPad: Add support for arbitrary focus sequencing. You can now edit this in the XKEYPAD_LAYOUT section of `XKeyPad_*-setup.h`.
- XKeyPad: Increased contrast, but you can define your own color scheme in the XKeyPad_*-setup.h file (see `#define XKEYPAD_COL_*`)
- XKeyPad_Num: Changed focus sequencing.
- XKeyPad_Alpha: Changed focus sequencing.
- XSlider: Support for focus/edit mode indicators as suggested.

Please try out the following branch: (also includes all changes from Input2a & Input2b):
- https://github.com/ImpulseAdventure/GUIslice/tree/Int384-Input2c

My next related plans are to: (once we've confirmed the above work for you)
- Wrap the new GUI navigation code in the GSLC_FEATURE_INPUT config mode
- Cleanup / remove debug messaging
- Integrate the changes for Paul's Toggle Image Button widget

thanks!

ImpulseAdventure, 2021-03-08:
I have now completed the following changes:
- Add **XToggleImgBtn** widget
- Add support for [WIO Terminal](https://www.seeedstudio.com/Wio-Terminal-p-4509.html) (integrated button controls can now navigate the GUI)
- Wrapped GUI navigation code with GSLC_FEATURE_INPUT
- Removed debug messaging

Please see the branch:
- https://github.com/ImpulseAdventure/GUIslice/tree/Int384-Input2d

If you get a chance to test this out and it looks like things are working OK, I may proceed to merge everything and then work on the redraw optimizations for #328

thanks!

MickTheMechanic, 2021-03-08:
Sorry I havnt been active for a while, I will test it out during the week and let you know what I find.

[12 more replies]

closed 2021-07-05 by ImpulseAdventure