ImpulseAdventure/GUIslice #351: More of some C  help than an issue....

bostonmacosx, 2021-01-17:
My Main file has become untennable due to a lot of functions. Decided to move them out to their own files.  1 other. 

The new .h file is 

```
void updateSettings(gslc_tsGui * w_Gui);
void fillSettings(int which,  gslc_tsGui * w_Gui);
const char* windgust(double wind, double gust);
const char* humiddw(double temp, double humid);
void setweaimg(int code,const char icon[],gslc_tsElemRef* whichref);
char * direction(int degrees);
```

In the Main file I'v added the new .h
```
#include "GUISLICE_GSLC.h"
#include "TFT_eFEX.h"
#include <JPEGDecoder.h>
#include <WiFi.h>
#include "time.h"
#include <ArduinoJson.h>
#include <string.h>
#include <HTTPClient.h>
#include <EEPROM.h>
#include "changedisplay.h"
```
and now I end up with 

> In file included from sketch/changedisplay.c:1:0:
>changedisplay.h:1:21: error: unknown type name 'gslc_tsGui'
>void updateSettings(gslc_tsGui * w_Gui);
  >                   ^
>changedisplay.h:2:31: error: unknown type name 'gslc_tsGui'
 >void fillSettings(int which,  gslc_tsGui * w_Gui);

Pconti31, 2021-01-17:
@bostonmacosx The problem seems to be in your changedisplay.c not the main ino.
you can rename changedisplay.c to changedisplay.ino and that might help or you can
make sure you include #include "GUISLICE_GSLC.h"
If you still have a problem post the full list of includes for changedisplay.c
Paul--

bostonmacosx, 2021-01-17:
@Pconti31 I thouut thaatt the #include "changedisplay.h" in the main .ino file would handle that? Why do I have too include GUISLICE_GSLC.h in the other changediisplay.c?

bostonmacosx, 2021-01-17:
@pConti ok so I got around this error by including 

#include "GUIslice.h"
#include "GUIslice_drv.h"
in the changedisplay.h file...
but that leads to not being able to use Serial.println in any of the functions iin the .c files.... arrrgggghhh!!!

Pconti31, 2021-01-17:
@bostonmacosx did you try the rename changedisplay.c to changedisplay.ino?
Paul--

Pconti31, 2021-01-17:
@bostonmacosx The #include "GUISLICE_GSLC.h" is useful since it includes all config files including arduino.h which is where Serial is defined.  Also, C files can have an issue with calling C++ so thats the reason for either rename of .c to .ino or even .cpp.
Also, make sure you have a guard inside your changedisplay.h
as in:
```
#ifndef _changedisplay_h
#define _changedisplay_h

....
#endif //_changedisplay_h
```
Paul--

Pconti31, 2021-01-17:
@bostonmacosx I just broke up one of samples and it worked fine with Serial without adding 
#include "GUIslice.h"
#include "GUIslice_drv.h"
or the _GSLC.h file simply by renaming the extra file using ino extension.
Paul--

ImpulseAdventure, 2021-01-30:
@bostonmacosx -- I'll assume that Paul's proposed correction above enabled your code to compile. I will close the issue, but if you are still encountering an issue, please post an update here and I'll reopen it. thanks!

closed 2021-01-30 by ImpulseAdventure