ImpulseAdventure/GUIslice #352: Graph with second value

mcarosh, 2021-01-22:
Hello 
I am not sure if this is a feature request. But I was asking myself if it is possible to draw a second line in the graph without loosing resolution? In my case I display a temperature and the given Set-temperature. It is working but it started really "flickering" when drawing both lines. My (silly?) question is if it is possible to handle the Settemp value as a static variable which only changes when it changes with a pressed button. This should maybe save some processor performance and keep its resolution high. I hope I could explain this request/question understandable enough.
But so far really thanks for this great library. it saves a lot of work and time ;)
regards
mcarosh

Pconti31, 2021-01-22:
@mcarosh I'm sorry but your idea doesn't help.  I gave it a go and hacked the graph to support the set point. It does save memory but the flicker is just as bad.  I attached it so you can see for yourself.

The only way I can see to improve matters is to either use hardware scrolling which is not directly supported or if you used double buffering.   If you use a ESP32 with TFT_eSPI this would be easy to do with its sprite support.
The Adafruit GFX does have something called canvas for a backstore but I have never used it and I doubt Arduino would have enough memory to store a graph.
Paul--
[graph_setpoint.zip](https://github.com/ImpulseAdventure/GUIslice/files/5857141/graph_setpoint.zip)

mcarosh, 2021-01-22:
Hi Paul
First thx for your fast reply and your efforts. I´ve now cheated in a way where I simply draw a line overlapped to the graph and made y-values the SetTemp variables. (Also recalculated the y-position according to the display) This works a bit better because now only the line is flickering. But this is better than both (graph and line :). Again big thanks for your help and maybe someone else has a better idea?
Do you wish to close this comment?
regards
mcarosh

Pconti31, 2021-01-22:
@mcarosh Happy you found a solution.  
Closing this is your choice but you should add this to issue:
Proposal for new Plot control `#288`
Paul--

mcarosh, 2021-01-22:
OK
thanks

closed 2021-01-22 by mcarosh